zephyrproject-rtos/zephyr #45362: sample/net/sockets/dumb_http_server not working with enc28j60

mauricionh, 2022-05-05:
**Describe the bug**
I'm trying to build the dumb_http_server sample application for a CC1352 launch pad, but it doesn't seem to work with the enc28j60 ethernet adapter board. I get a lot of build errors.

I'm using zephyr sdk 0.14.1, on Windows 10, following the instructions from the getting started [page](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) v 3.0.99

Please also mention any information which could help others to understand
the problem you're facing:
The target board I'm using is a Texas Instruments CC1352r1 launch pad xl. Hw: rev B.
- What have you tried to diagnose or workaround this issue?
I have roamed around several posts here in the repository, but they seem to include an outdated set of configuration flags. I was able to add the following to create my own overlay-enc28j60.conf file:

```
CONFIG_GPIO=y
CONFIG_SPI=y
CONFIG_ETH_ENC28J60=y
CONFIG_ETH_ENC28J60_0=y
CONFIG_ETH_ENC28J60_0_FULL_DUPLEX=y

CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_ARP=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.1.250"

CONFIG_NET_HOSTNAME_ENABLE=y
CONFIG_NET_HOSTNAME="zephyr-device"

CONFIG_INIT_STACKS=y

CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

# Networking tweaks
# Required to handle large number of consecutive connections,
# e.g. when testing with ApacheBench.
CONFIG_NET_TCP_TIME_WAIT_DELAY=0

# Network debug config
CONFIG_LOG=y
CONFIG_NET_SHELL=y
```

I also used this to create a cc1352r1_launchxl.overlay file:

```
&spi0 {
	compatible = "ti,launchxl-cc1352r1";
	status = "okay";
	pinctrl-0 = <&spi0_sck_default &spi0_mosi_default
				 &spi0_miso_default &spi0_cs_default>;
	pinctrl-names = "default";
	cs-gpios = <&gpio0 11 GPIO_ACTIVE_LOW>;

	eth_enc28j60: eth_enc28j60@0 {
		compatible = "microchip,enc28j60";
		label = "ENC28J60";
		reg = <0x0>;
		local-mac-address = [01 02 03 04 05 06];
		spi-max-frequency = <4000000>;
		int-gpios = <&gpio0 20 GPIO_ACTIVE_LOW>;
	};
};
```

**To Reproduce**
Steps to reproduce the behavior:
I'm using west build on a folder containing all elements from the dumb_http_server sample.
west build --pristine --build-dir <path-to-project>\cc1352_enc28j60\build <path-to-project>\cc1352_enc28j60 -b cc1352r1_launchxl -DCONF_FILE="prj.conf overlay-enc28j60.conf"

**Expected behavior**
Having no build errors.
Having a proper overlay for the board configuration 
Having a proper enc28j60 configuration overlay

**Impact**
Showstopper

**Logs and console output**
Upon building the project I see the following:
* My custom board overlay file is found 
* ```warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but got the value 'n'. Check these unsatisfied dependencies: (!ENTROPY_HAS_DRIVER) (=n). See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_TEST_RANDOM_GENERATOR and/or look up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.```
* The configuration overlay file for the enc28j60 adapter is found as well and merged with the projects prj.conf file.

I then start getting some warnings and errors:
*```[152/292] Building C object zephyr/drivers/spi/CMakeFiles/drivers__spi.dir/spi_cc13xx_cc26xx.c.obj
<path-to-sdk>/zephyrproject/zephyr/drivers/spi/spi_cc13xx_cc26xx.c:237:36: warning: 'spi_cc13xx_cc26xx_driver_api' defined but not used [-Wunused-const-variable=]
  237 | static const struct spi_driver_api spi_cc13xx_cc26xx_driver_api = {
      |```
*```In file included from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/arch/arm/aarch32/arch.h:20,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/arch/cpu.h:19,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/kernel_includes.h:33,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/zephyr.h:18,
                 from <path-to-sdk>/zephyrproject/zephyr/drivers/ethernet/eth_enc28j60.c:16:
<path-to-sdk>/cc1352_enc28j60/build/zephyr/include/generated/devicetree_unfixed.h:3865:40: error:
DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_local_mac_address' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_label'?
 3865 | #define DT_N_INST_0_microchip_enc28j60 DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0
      |```
*```In file included from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/toolchain/gcc.h:88,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/toolchain.h:50,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/sys/atomic.h:12,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/logging/log_msg.h:9,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/logging/log_core.h:9,
                 from <path-to-sdk>/zephyrproject/zephyr/include/zephyr/logging/log.h:11,
                 from <path-to-sdk>/zephyrproject/zephyr/drivers/ethernet/eth_enc28j60.c:13:
<path-to-sdk>/zephyrproject/zephyr/include/zephyr/device.h:96:39: error: '__device_dts_ord_DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_BUS_ORD' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_ORD'?
   96 | #define DEVICE_NAME_GET(name) _CONCAT(__device_, name)
      |```

And several others...

**Environment (please complete the following information):**
 - OS: Windows 10 Pro
 - Toolchain: Zephyr SDK
 - 0.14.1

Thanks!

mauricionh, 2022-05-05:
Upon further inspection of the issues involving the enc28j60, I reviewed this [one](https://github.com/zephyrproject-rtos/zephyr/pull/23730#issue-587211033). I followed the changes and replicated in my setup, and I still get the same issues. The basic understanding I have at the moment is pointing me towards the board overlay that I've added to the project. The build error messages involve the properties of the **eth_enc28j60: eth_enc28j60@0** node.

These are the errors I get:
* < path-to-project >/build/zephyr/include/generated/devicetree_unfixed.h:3865:40: error: 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_local_mac_address' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_label'?
 3865 | #define DT_N_INST_0_microchip_enc28j60 **DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0**

* < path-to-sdk >/zephyrproject/zephyr/include/zephyr/device.h:96:39: error: '__device_dts_ord_DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_BUS_ORD' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_ORD'?
   96 | #define DEVICE_NAME_GET(name) _CONCAT(**__device_**, name)

* < path-to-project >/build/zephyr/include/generated/devicetree_unfixed.h:3865:40: error: 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_spi_max_frequency' undeclared here (not in a function); 
did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_reg'?
 3865 | #define DT_N_INST_0_microchip_enc28j60 **DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0**
 
 * < path-to-project >/build/zephyr/include/generated/devicetree_unfixed.h:3865:40: error: 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_duplex' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_reg'?
 3865 | #define DT_N_INST_0_microchip_enc28j60 **DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0**

* < path-to-project >/build/zephyr/include/generated/devicetree_unfixed.h:3865:40: error: 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_frame_format' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_reg'?
 3865 | #define DT_N_INST_0_microchip_enc28j60 **DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0**

* < path-to-sdk >/zephyrproject/zephyr/include/zephyr/device.h:96:39: error: '__device_dts_ord_DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_int_gpios_IDX_0_PH_ORD' undeclared here (not in a function)  
   96 | #define DEVICE_NAME_GET(name) _CONCAT(**__device_**, name)
   
* < path-to-project >/build/zephyr/include/generated/devicetree_unfixed.h:3865:40: error: 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_int_gpios_IDX_0_VAL_pin' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0_P_reg_IDX_0_EXISTS'?
 3865 | #define DT_N_INST_0_microchip_enc28j60 **DT_N_S_soc_S_spi_40000000_S_eth_enc28j60_0**

This is the board overlay for the cc1352:
```
&spi0 {
	compatible = "ti,launchxl-cc1352r1";
	status = "okay";
	/*pinctrl-0 = <&spi0_sck_default &spi0_mosi_default
				 &spi0_miso_default &spi0_cs_default>;*/
	pinctrl-names = "default";
	cs-gpios = <&gpio0 11 GPIO_ACTIVE_LOW>;
	clock-frequency = <4000000>;

	eth_enc28j60: eth_enc28j60@0 {
		compatible = "microchip,enc28j60";
		reg = <0x0>;
		spi-max-frequency = <2000000>;
		label = "ETH_0";
		duplex = <2048>;
		frame-format = <32768>;
		local-mac-address = [01 02 03 04 05 06];
		int-gpios = <&gpio0 20 GPIO_ACTIVE_LOW>;
	};
};
```
I saw that the _.yaml_ file in the binding folders specified to use the macros and not the ENUM values. I'm using visual studio code and it didn't recognize the values as SPI_HALF_DUPLEX, and SPI_FRAME_FORMAT_TI. 

Any insights will be extremely appreciated at this point. Thank you

cfriedt, 2022-05-14:
@mauricionh - I was able to build your setup with this:
```
cfriedt@Christophers-MacBook-Air zephyr % git add samples/net
cfriedt@Christophers-MacBook-Air zephyr % git diff --cached  
diff --git a/samples/net/sockets/dumb_http_server/boards/cc1352r1_launchxl.overlay b/samples/net/sockets/dumb_http_server/boards/cc1352r1_launchxl.overlay
new file mode 100644
index 0000000000..27551ef7fe
--- /dev/null
+++ b/samples/net/sockets/dumb_http_server/boards/cc1352r1_launchxl.overlay
@@ -0,0 +1,11 @@
+&spi0 {
+       eth_enc28j60: eth_enc28j60@0 {
+               compatible = "microchip,enc28j60";
+               label = "ENC28J60";
+               reg = <0x0>;
+               local-mac-address = [01 02 03 04 05 06];
+               spi-max-frequency = <4000000>;
+               int-gpios = <&gpio0 20 GPIO_ACTIVE_LOW>;
+       };
+};
+
diff --git a/samples/net/sockets/dumb_http_server/overlay-enc28j60.conf b/samples/net/sockets/dumb_http_server/overlay-enc28j60.conf
new file mode 100644
index 0000000000..5c3a57aa03
--- /dev/null
+++ b/samples/net/sockets/dumb_http_server/overlay-enc28j60.conf
@@ -0,0 +1,7 @@
+CONFIG_NET_L2_ETHERNET=y
+
+CONFIG_GPIO=y
+CONFIG_SPI=y
+CONFIG_ETH_ENC28J60=y
+CONFIG_ETH_ENC28J60_0=y
+CONFIG_ETH_ENC28J60_0_FULL_DUPLEX=y
```
and
`west build -p auto -b cc1352r1_launchxl samples/net/sockets/dumb_http_server -- -DCONF_FILE="prj.conf overlay-enc28j60.conf"`

mauricionh, 2022-05-14:
Thank you @cfriedt, were you able to reach the server through a browser or ping the device? I can't find it.

cfriedt, 2022-05-14:
> Thank you @cfriedt, were you able to reach the server through a browser or ping the device? I can't find it.
> 
> I'm in the same (local)network, checking with different tools to see what is connected to the local network.

I don't have the same Ethernet adapter board. Likely you will also need to set some network addresses (just additional Kconfig settings). The `echo_server` sample would be a good reference.

mauricionh, 2022-05-16:
Thank you so much! I will let you know how it goes. :)

mbolivar-nordic, 2022-05-24:
Closing for now. @mauricionh please reopen if you believe there is a bug here.

mauricionh, 2022-05-28:
Hi @cfriedt, the application builds correctly, but I don't see anything in the network. 

I've attached the files that are involved.. prj.conf, overlay-enc28j60.conf, cc1352r1_launchxoverlay
[prj.zip](https://github.com/zephyrproject-rtos/zephyr/files/8790151/prj.zip)

Also tried removing the network addresses declared in prj.conf and only using the ones in the overlay.

Does the dumb_http_server require some commands to start, maybe something needs to happen over the CC1352 serial interface?

mauricionh, 2022-05-28:
@cfriedt do you know how to invert the polarity on the MOSI and MISO lines? I'm comparing the signals from a sample app from arduino. MOSI is Active low, while MISO is Active high. This is the opposite for the current state of the dumb http server app for the cc1352. 

I think I need to change this in the board overlay, but I'm not sure how, and I haven't found the documentation. In the cc1352r1_launchxl-pinctrl.dtsi I see that the spi0_mosi_default pin is defined with the properties pinmux, and bias-disable, and spi0_miso_default is defined with the properties pinmux, bias-disable, and input-enable.

cfriedt, 2022-05-29:
@mauricionh - I would suggest asking on Discord maybe in the #spi channel (or if there is a #pinctrl channel, there too)