zephyrproject-rtos/zephyr #45365: Zephyr IP Stack Leak Diagnostic Help

bob2oneil, 2022-05-05:
**Describe the bug**
I have a networking based application that provides 2 interfaces under the Native-POSIX board target, a simulated wired and simulated wireless interface.

The wired interface provides support for 2 UDP receive sockets, which can optionally (conditional compilation) be replaced with a singular raw layer 2 socket.

A Telnet server is active on this interface to provide status and control.

In my current Zephyr configuration, when data is ingress to either one of the registered UDP socket ports, or if the raw socket is used, the Zephyr stack appears to be leaking, resulting in an application failure.

This is confirmed by viewing the results of the shell output to "net allocs" command showing network buffer allocation and corresponding free (where present).

So it would seem that the stack allocations are not being properly de-registered and freed.

I am looking for some suggestions on where to dig into determine the problem and solution.

I can delay the amount of ingress traffic before the exception/failure by increasing the number of network buffers, but eventually it will happen.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?  Native-POSIX
 - What have you tried to diagnose or workaround this issue? Yes
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b native_posix_64 ${workspaceFolder} 

**Expected behavior**
Network allocations are properly freed.

**Impact**
Showstopper

**Logs and console output**

Here is a capture of the callstack at the point of the exception occurence.

```
net_pkt_cursor_init(struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_pkt.c:1516)
net_if_recv_data(struct net_if * iface, struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_if.c:3929)
process_data(_Bool is_loopback, struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/include/zephyr/net/net_pkt.h:286)
processing_data(struct net_pkt * pkt, _Bool is_loopback) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_core.c:141)
net_rx(struct net_if * iface, struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_core.c:365)
net_process_rx_packet(struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/include/zephyr/net/net_pkt.h:286)
tc_rx_handler(struct k_fifo * fifo) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_tc.c:250)
z_thread_entry(k_thread_entry_t entry, void * p1, void * p2, void * p3) (/home/robocom/zephyrproject/zephyr/lib/os/thread_entry.c:36)
posix_thread_starter(void * arg) (/home/robocom/zephyrproject/zephyr/arch/posix/core/posix_core.c:305)
libpthread.so.0!start_thread(void * arg) (/build/glibc-sMfBJT/glibc-2.31/nptl/pthread_create.c:477)
libc.so.6!clone() (/build/glibc-sMfBJT/glibc-2.31/sysdeps/unix/sysv/linux/x86_64/clone.S:95)
```

Here is a dump of "net allocs" from the Telnet interface menu with a simple 1 pulse per second PING traffic ingress to the primary Native-POSIX Ethernet interface.  Note that there is no corresponding free to the packet clone.

```
0x4f06d0/1       used      RX   net_pkt_clone_internal():1823
0x4f0658/1       used      RX   net_pkt_clone_internal():1823
0x4f05e0/1       used      RX   net_pkt_clone_internal():1823
0x4f0568/1       used      RX   prepare_non_vlan_pkt():305
0x4f04f0/1       used      RX   net_pkt_clone_internal():1823
0x4f0478/1       used      RX   net_pkt_clone_internal():1823
0x4f0400/1       used      RX   net_pkt_clone_internal():1823
0x4f0388/1       used      RX   net_pkt_clone_internal():1823
0x4f0310/1       used      RX   net_pkt_clone_internal():1823
0x4f0298/1       used      RX   net_pkt_clone_internal():1823
0x4f0220/1       used      RX   net_pkt_clone_internal():1823
0x4f01a8/1       used      RX   net_pkt_clone_internal():1823
0x4f0130/1       used      RX   net_pkt_clone_internal():1823
0x4f00b8/1       used      RX   net_pkt_clone_internal():1823
0x4f0040/1       used      RX   net_pkt_clone_internal():1823
0x4eff50/1       used      RX   net_pkt_clone_internal():1823
[... 211 lines cut ...]
# File System Shell
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_SHELL=y

# Gather network statistics
CONFIG_NET_STATISTICS_ETHERNET=y

# Name threads
CONFIG_THREAD_NAME=y

# Max number of IPv4 network interfaces in the system
# Will be a value of 1 for real radio
CONFIG_NET_IF_MAX_IPV4_COUNT=2

# How many Tx traffic classes to have for each network device
# 1 = all the network traffic is handled equally.
# 0 = all the network traffic is pushed to the driver directly without any queues
CONFIG_NET_TC_TX_COUNT=1

# How many network connections is supported (default is 4)
CONFIG_NET_MAX_CONN=8

# Total number of sockets to poll in select()
CONFIG_NET_SOCKETS_POLL_MAX=5
```

jukkar, 2022-05-11:
Are you familiar with the promiscuous mode sample app at `samples/net/promiscuous_mode`, it shows how to use the promiscuous mode API. In short, if you enable promiscuous mode by calling ` net_promisc_mode_on()`, then you need to have a handler that will need to unref the received packet, otherwise there is a leak.

bob2oneil, 2022-05-11:
I understand, but this then implies that using standard POSIX APIs at the network layer is incompatible with promiscuous mode for the network interface, that is Zephyr is not compatible with that standard.  Is there any way that the underlying collection pointer to be freed could be exposed to say the zsock_recv() API so that the application could free it?, or would it just make sense that Zephyr perform the free itself as a result of that call?, meaning as soon as the content is copied to the user space buffer, there is no reason to keep the allocation around anymore.  I think this approach would be the most elegant, keeping Zephyr compatible to the POSIX standard and freeing the developer adhering to POSIX standard APIs out of the business of knowing deallocations.  Using the POSIX APIs provides efficiencies to folks porting existing applications, and adding promiscuous support would make Zephyr compatible with the Linux APIs.

rlubos, 2022-05-11:
I think we've discussed this already in https://github.com/zephyrproject-rtos/zephyr/issues/36748.

TLDR the conclusion was that the module enabled with `CONFIG_NET_PROMISCUOUS_MODE` is a standalone API, not integrated with sockets, therefore I think it's safe to say that Zephyr does not support promiscuous RAW sockets. Threfore I think the issue should rather be cosidered as an enhancement, not a bug.

bob2oneil, 2022-05-11:
Yes, the discussion from the past is the same one, over time I have found that I really needed that raw socket, which is not uncommon for any Layer 2 networking device, and as the Zephyr code base as it has evolved since my original posting, changes/updates have not addressed that issue.  

Is the level of effort described to have the **zsock_recv** API simply free up the local allocation a complex issue to implement?  

I really don't know the scope of what I am asking, and you guys being the subject matter experts, know the full scope of the design

jukkar, 2022-05-11:
You could have a somewhat hackish workaround by having a "dummy" promiscuous mode handler that just discards the received packets, and then your raw socket should also receive all the same packets. This is not optimal of course but should work until we get support for proper promiscuous mode sockets.

jukkar, 2022-05-12:
Yeah, I was thinking a thread that would set the interface to promiscuous mode and have a while loop like you have above.
After some code digging there seems to be a simpler way. The promiscuous packets are processed in https://github.com/zephyrproject-rtos/zephyr/blob/bb8575962da291e3b75dfb3abe2fcb9341f76be2/subsys/net/ip/net_if.c#L3900 and if the interface is set to promiscuous mode but there is no handler i.e., the `net_promisc_mode_on()` is **not** called, then the packet should be unreffed properly and no leaks should happen.
So if you want to receive packets in promiscuous mode, but do not want to use the promiscuous mode API, you can enable promiscuous mode in config file, and then call `net_if_set_promisc(iface);` to set the interface to promiscuous mode. You should then be able to receive all the network traffic in your device without any buffer leaks and without any dummy handlers etc. Not tested but should work ok.

carlescufi, 2022-05-18:
> Yes, the discussion from the past is the same one, over time I have found that I really needed that raw socket, which is not uncommon for any Layer 2 networking device, and as the Zephyr code base [...]

Given that you are really asking for a feature request here, I will convert this to a discussion and then ask you to open  a new issue (marked as enhancement), along the lines of: "Implement promiscuous mode for `AF_PACKET` sockets"