zephyrproject-rtos/zephyr #45367: net: tcp: Scheduling dependent throughput

ssharks, 2022-05-05:
**Describe the bug**
The throughput with the TCP stack for large transfers is even in ideal situations dependent on scheduling / timing. Even till the point where it can fail entirely.

The situation is happening on an proprietary embedded system in combination with an Azure cloud application, when sending data. Receiving data does not prose challenges so far.
It can also be reproduced using a modified testcase in https://github.com/ssharks/zephyr/tree/net/tcp_throughput using the Qemu x86 emulator.

In this modified testcase the debug logging has been enabled, which shows various unexpected messages like:
`[00000400] <err> net_pkt: Data buffer (68) allocation failed.`
Also rather big time leaps where within almost 100 ms, no action takes place without clear explanation:
```
[00000910] <dbg> net_tcp: tcp_unsent_len: (rx_q[0]): unsent_len=0
[00001000] <dbg> net_tcp: tcp_window_full: (test_v4_send_recv_large): conn: 0x1703f8 window_full=0

```
**To Reproduce**
Steps to reproduce the behavior:
Run ./scripts/twister -T tests/net/socket/tcp -p qemu_x86
And look in the generated handler.log

**Expected behavior**
The transfer should complete in ~1 second instead of > 20 seconds without a clear difference between platforms and cooperative or preemptive scheduling.

**Impact**
The current TCP implemenation is not workable for uploading larger payloads

**Logs and console output**
Log files of both the test in cooperative and preemptive scheduling have been attached. The preemptive case failed with a timeout.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...
[handler_coop.log](https://github.com/zephyrproject-rtos/zephyr/files/8629030/handler_coop.log)
[handler_preempt.log](https://github.com/zephyrproject-rtos/zephyr/files/8629032/handler_preempt.log)

AndreyDodonov-EH, 2022-05-05:
@ssharks I'm about to open an issue and probably PR dedicated specifically to
`Data buffer ... allocation failed`
(or, rather, underlying problems, it's just a symptom)

I'll create it and link it here within maximum next week.
If I forget to do so, feel free to ping me.

ssharks, 2022-05-05:
Thanks to a comments from @rlubos the pending PR https://github.com/zephyrproject-rtos/zephyr/pull/45018 fixed the hanging. Secondly the time leaps are explained by https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets.c#L708. Halving the WAIT_BUFS significantly reduces the test duration.

AndreyDodonov-EH, 2022-05-06:
@ssharks
Here is the issue dedicated to one of the reasons for `Data buffer allocation` errors
https://github.com/zephyrproject-rtos/zephyr/issues/45426

Correspondent PR:
https://github.com/zephyrproject-rtos/zephyr/pull/45437

AndreyDodonov-EH, 2022-05-06:
@ssharks 
Another relevant issue would be https://github.com/zephyrproject-rtos/zephyr/issues/5842
because if we have select/poll working for writing sockets, application would be able to determine _exactly_ when we can send new data.

ssharks, 2022-05-09:
@AndreyDodonov-EH 
As indicated by @rlubos, most of the time is spend in the polling loop in the zsock_sendto_ctx. An alternative implementation has been made by using exponentially increasing delays in:
https://github.com/ssharks/zephyr/commit/00b744d67d2c07210b614565bf402616c14f4d29

Using the test: test_v4_send_recv_large

With exponential delay:
Preempt: 10.3
Coop: 6.2

With conventional 100 ms delay:
Preempt: 39.3
Coop: 12.3

A similar exponential back off might be usefull in tcp_resend_data and other places where -ENOBUFS can occur.

ssharks, 2022-05-09:
Other significant contributors to the low throughput are the enormous amount of context switches as well as the lack of a real congestion avoidance algorithm using fast re-transmit or something similar. Implementation of Nagle's algorithm has to potential to significantly reduce the amount of context switches when sending small chunks of data.

Implementing a fast re-transmit would make the TCP stack recover much faster from a packet loss. But implementing this without a proper congestion avoidance algorithm has the potential to make connection collapse on itself.

PR #45626 (2022-05-18): net: tcp: Notify TX window avaialbility 
This PR adds a mechanism which allows the TCP layer to notify the TX window state to the upper layer through a semaphore. This allows to achieve better throughput in low buffer scenarios, as the socket layer does not need to wait blindly, as well as to implement POLLOUT functionality.

Additionally, improve ZWP transmission logic, by dedicating a separate timer for ZWP sending. This is essantial for POLLOUT to work correctly, as we're not triggering send in this case, which was the place ZWP was sent before.

Fixes #45367
Fixes #45426