zephyrproject-rtos/zephyr #45371: frdm_k64f: failed to run test case tests/drivers/can/timing/drivers.can.timing

shaoanxli, 2022-05-05:
**Describe the bug**
The test case failed after commit id: 4896a23aeabf937293aa3f231d1e5c0cf85ce8cb.And before this commit, there is a build error. The root cause is not clear yet.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. west build -p always -b tests/net/socket/offload_dispatcher/ 
3. Open another terminal and execute: picocom -b 115200 /dev/ttyACM0
4. See error
 or
 2.twister --device-testing --hardware-map $HOME/sam_e70b_xplained.file tests/net/socket/offload_dispatcher/  -vv
 3.See error

**Expected behavior**
Test case should pass as expected.

**Impact**
The test case drivers.can.timing failed(timeout).

**Logs and console output**
```
-- runners.pyocd: Flashing file: /home/ztest/zephyrproject/zephyr/twister-out/frdm_k64f/tests/net/socket/offload_dispatcher/net.socket.offload.dispatcher/zephyr/zephyr.bin
[---|---|---|---|---|---|---|---|---|----]
[========================================]

DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sendto_not_bound
DEBUG   - DEVICE: PASS - test_sendto_not_bound in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_recvfrom_not_bound
DEBUG   - DEVICE: PASS - test_recvfrom_not_bound in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_getsockopt_not_bound
DEBUG   - DEVICE: PASS - test_getsockopt_not_bound in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_setsockopt_not_bound
DEBUG   - DEVICE: PASS - test_setsockopt_not_bound in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sendmsg_not_bound
DEBUG   - DEVICE: PASS - test_sendmsg_not_bound in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_getpeername_not_bound
DEBUG   - DEVICE: PASS - test_getpeername_not_bound in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_getsockname_not_bound
DEBUG   - DEVICE: PASS - test_getsockname_not_bound in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_so_bindtodevice_iface_offloaded
DEBUG   - DEVICE: PASS - test_so_bindtodevice_iface_offloaded in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_so_bindtodevice_iface_native
DEBUG   - DEVICE: PASS - test_so_bindtodevice_iface_native in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_tls_native_iface_offloaded
DEBUG   - DEVICE: PASS - test_tls_native_iface_offloaded in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_tls_native_iface_native
DEBUG   - DEVICE: E: ***** BUS FAULT *****
DEBUG   - DEVICE: E:   Stacking error
DEBUG   - DEVICE: E:   Imprecise data bus error
DEBUG   - DEVICE: E:   NXP MPU error, port 3
DEBUG   - DEVICE: E:     Mode: Supervisor, Data Address: 0x20008fe8
DEBUG   - DEVICE: E:     Type: Write, Master: 0, Regions: 0x8100
DEBUG   - DEVICE: E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
DEBUG   - DEVICE: E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
DEBUG   - DEVICE: E:  xpsr:  0x00000000
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x00000000
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x200014a0 (test_tls_native_iface_native)
DEBUG   - DEVICE: E: Halting system
```

**Environment (please complete the following information):**
OS: Linux,
Toolchain Zephyr SDK 0.13.2
Commit SHA or Version: 4896a23aeabf937293aa3f231d1e5c0cf85ce8cb

henrikbrixandersen, 2022-05-05:
The title says tests/drivers/can/timing/drivers.can.timing is failing on frdm_k64f, but the body talks about sam_e70b_xplained and tests/net/socket/offload_dispatcher/? Which is it?

shaoanxli, 2022-05-06:
> The title says tests/drivers/can/timing/drivers.can.timing is failing on frdm_k64f, but the body talks about sam_e70b_xplained and tests/net/socket/offload_dispatcher/? Which is it?

Sorry ,I got the command wrong, I have fixed it , thanks for reminding. The bug occurred on the frdm_k64f platform.

shaoanxli, 2022-05-11:
Before this bug , the test had a build error after the commit: 7ad2e604bb3155c09a484717e1252c8f404640c0

console output:
```
INFO    - Adding tasks to the queue...
DEBUG   - adding frdm_k64f/tests/net/socket/offload_dispatcher/net.socket.offload.dispatcher
INFO    - Added initial list of jobs to queue
DEBUG   - Launch process 0
DEBUG   - Launch process 1
DEBUG   - Launch process 2
DEBUG   - Launch process 3
DEBUG   - Running cmake on /home/ztest/zephyrproject/zephyr/tests/net/socket/offload_dispatcher for frdm_k64f
DEBUG   - Calling cmake with arguments: ['-B/home/ztest/zephyrproject/zephyr/twister-out/frdm_k64f/tests/net/socket/offload_dispatcher/net.socket.offload.dispatcher', '-S/home/ztest/zephyrproject/zephyr/tests/net/socket/offload_dispatcher', '-DTC_RUNID=7e72602e572111bc363165e5867092e7', '-DEXTRA_CFLAGS=-Werror', '-DEXTRA_AFLAGS=-Werror -Wa,--fatal-warnings', '-DEXTRA_LDFLAGS=-Wl,--fatal-warnings', '-DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror', '-GUnix Makefiles', '-DBOARD=frdm_k64f']
DEBUG   - Launch process 4
DEBUG   - Launch process 5
DEBUG   - Launch process 6
DEBUG   - Launch process 7
DEBUG   - Finished building /home/ztest/zephyrproject/zephyr/tests/net/socket/offload_dispatcher for frdm_k64f
DEBUG   - build test: frdm_k64f/tests/net/socket/offload_dispatcher/net.socket.offload.dispatcher
DEBUG   - Building /home/ztest/zephyrproject/zephyr/tests/net/socket/offload_dispatcher for frdm_k64f
INFO    - 1/1 frdm_k64f                 tests/net/socket/offload_dispatcher/net.socket.offload.dispatcher FAILED Build failure (device)
ERROR   - see: /home/ztest/zephyrproject/zephyr/twister-out/frdm_k64f/tests/net/socket/offload_dispatcher/net.socket.offload.dispatcher/build.log
```
I think two bugs are related.

PerMac, 2022-05-11:
I also see the same test failing on nrf52840dk_nrf52840 also with stack overflow:
```
START - test_tls_native_iface_native
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x20006830
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x20000a58 (test_tls_native_iface_native)
E: Halting system
```

dleach02, 2022-05-20:
@rlubos, can you check to see if you can reproduce what @PerMac is reporting?

dleach02, 2022-05-20:
Okay, this was fixed by: #45070

closed 2022-05-20 by dleach02