zephyrproject-rtos/zephyr #45372: PWM not working

Ventsislav-Trayanov-GHD, 2022-05-05:
**Describe the bug**
In commit 3650664875a645c184cfe79e6f7f369b3af23653
a bug was introduced when sending 0 to period_cycles its expected to stop the PWM that was the case in the previous version now this does not work. 

Please also mention any information which could help others to understand
the problem you're facing:
I can stop the PWM by sending 0XFFFF but that shudnt be the case. 
 - ...

**To Reproduce**
Steps to reproduce the behaviour:
1.STM32 any dev board 
2. Send PWM sysganl to a buzzer and try terminating the tone by sending 0 for cycles 

**Expected behaviour**
It is expected when you send 0 cycles  to terminate the PWM

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (Zephyr SDK)
 - Commit 3650664875a645c184cfe79e6f7f369b3af23653

gmarull, 2022-05-05:
@GeorgeCGV

GeorgeCGV, 2022-05-05:
The check is still present.

https://github.com/zephyrproject-rtos/zephyr/blob/3650664875a645c184cfe79e6f7f369b3af23653/drivers/pwm/pwm_stm32.c#L298

However, it shall be moved higher up.. before subtraction happens.

PR #45376 (2022-05-05): drivers: pwm: stm32: fix PWM channel disabling
Move period and pulse computation to right before
the channel enable code.

That fixes the inability to disable the channel by
providing the period of 0.

Fixes #45372
closed 2022-05-05 by MaureenHelm