zephyrproject-rtos/zephyr #45379: net: TCP/IP Maxwell Pro test results

rlubos, 2022-05-05:
Thanks to the courtesy of NXP, we've collected a series of TCP/IP stack test results. The test report can be found in a separate repository (https://github.com/zephyrproject-rtos/test_results/issues), each test failure received a separate issue.

Now, before we start moving the issues into Zephyr as an actual bug reports, we need to triage them first. This means, that every issue needs to be examined whether it's an actual problem (i. e. not a false positive report) and whether it refers to a feature that is actually implemented (for instance there is a series of reports regarding IPv4 fragmentation, which is not supported by Zephyr).

If you have a certain knowledge of Zephyr TCP/IP stack and are willing to help with triaging, feel free to assign an issue in the "test_results" repository to yourself. The idea is to label the issues with either "Bug", "Won't fix" or "Enhancement" labels once triaged - the actual bug reports would be then moved to Zephyr. Note, that you're not expected to fix the issue on your own - of course, you can propose a fix, but simply triaging the issue is already a great help.

Currently, there are 121 open issues for triaging. I propose to keep this thread as an umbrella issue for tracking the progress of the triaging and bug fixing process.

carlescufi, 2025-04-02:
@dleach02 @hakehuang should we start tracking the failures a bit closer? If so, we need detailed reports on the failures and not just a list of  failed tests.

hakehuang, 2025-04-03:
> [@dleach02](https://github.com/dleach02) [@hakehuang](https://github.com/hakehuang) should we start tracking the failures a bit closer? If so, we need detailed reports on the failures and not just a list of failed tests.

may I have your mail address, so that I can send you the test report

rlubos, 2025-04-03:
@hakehuang I've sent you an email, thanks!