zephyrproject-rtos/zephyr #45394: Bug when sending a BLE proxy mesh msg of length exactly 2x the MTU size

aralex1, 2022-05-05:
Hi all, I am sorry if this is not the correct format for reporting something. I am new to open source code, so beforehand, I apologize if this is not the correct procedure. 

I am currently using the [bt_mesh_proxy_msg_send ](https://github.com/zephyrproject-rtos/zephyr/blob/d8b45e38c130b6c83fc8eba6f9e031c78cb2b276/subsys/bluetooth/mesh/proxy_msg.c#L140) function call in my implementation and I noticed that if the message is exactly double or triple of the MTU size minus the msg size/MTU, the while loop in this function will never send the SAR_LAST message. 

Let's consider this scenario that it is how I noticed this. If the MTU is equal to 250 and the msg length is equal to 498, the function will add the type to the original msg increasing the msg length to 499, and then, it will send 250 bytes. Now, we have the rest of the msg whose length is now 249. Then, once we enter into the while loop, it checks if we can send the complete msg only if the msg length +1 (249 + 1) is less than 250. Since it is not less than the MTU, we add the type to the message which increases the message length to 250 and we send a SAR_CONT msg. This leaves us with a msg of size 0 and we leave the loop without ever sending the SAR_LAST msg.

Shouldn't the while loop check if the msg length + 1 is [less or equal](https://github.com/zephyrproject-rtos/zephyr/blob/d8b45e38c130b6c83fc8eba6f9e031c78cb2b276/subsys/bluetooth/mesh/proxy_msg.c#L167) than the MTU size? in this case, we will send the SAR_LAST msg when the msg is exactly the MTU size.

mbolivar-nordic, 2022-05-10:
@PavelVPV could you please take a look and decide if it's a bug or not?

PavelVPV, 2022-05-11:
Hi @aralex1 , yes, you are right, this is a bug. I'll fix this.

aralex1, 2022-05-11:
Hi @PavelVPV, thanks for checking this bug.

cvinayak, 2022-05-12:
@PavelVPV Adding the `Fixes #45394` in the PR description would link it with this issue, adding as a comment does not.

PR #45551 (2022-05-13): Bluetooth: Mesh: Fix segmentation when sending proxy message
Previous check in the if-statement would never allow to send last
segment if msg->len + 2 == MTU * x.

Fixes #45394
closed 2022-05-13 by jhedberg