zephyrproject-rtos/zephyr #45405: up_squared: most of the test case timeout

shaoanxli, 2022-05-06:
**Describe the bug**
The test case failed after commit id: d5e7732e0d0f66bc9a3684f9597d9fd6a2d351b2 .

Many tests failed , we take hello_world as an example.
**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. west build -p always -b up_squared samples/hello_world
3. Open another terminal and execute: picocom -b 115200 /dev/ttyUP2
4. west flash
5. See error
 or
1.twister -p up_squared --device-testing --device-serial /dev/ttyUP2 -T samples/hello_world/ --west-flash=/home/chenp1/up_squared_efi.sh --no-skipped-report -vv
2.See error

**Expected behavior**
Test case should pass as expected.

**Impact**
Most the test case failed(timeout).

**Logs and console output**
```
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/chenp1/work/zephyrproject/zephyr/twister-out/up_squared/samples/hello_world/sample.basic.helloworld', '--', '/home/chenp1/up_squared_efi.sh']
DEBUG   - -- west flash: using runner misc-flasher
<HTML><HEAD>

<META HTTP-EQUIV="refresh" content="0; URL=/script.htm">

</HEAD><BODY></BODY>
</HTML>/home/chenp1/work/zephyrproject/zephyr/twister-out/up_squared/samples/hello_world/sample.basic.helloworld/zephyr/zephyr.efi
<HTML><HEAD>

<META HTTP-EQUIV="refresh" content="0; URL=/script.htm">

</HEAD><BODY></BODY>
</HTML>
DEBUG   - Timed out while monitoring serial output on up_squared
DEBUG   - run status: up_squared/samples/hello_world/sample.basic.helloworld timeout
INFO    - 1/1 up_squared                samples/hello_world/sample.basic.helloworld        FAILED Timeout (device 62.632s)
ERROR   - see: /home/chenp1/work/zephyrproject/zephyr/twister-out/up_squared/samples/hello_world/sample.basic.helloworld/device.log

```

**Environment (please complete the following information):**
OS: Linux,
Toolchain Zephyr SDK 0.13.2
Commit SHA or Version: d5e7732e0d0f66bc9a3684f9597d9fd6a2d351b2

enjiamai, 2022-05-06:
Hi @andyross, @spoorthik  we found the up2 daily testing blocked by this change https://github.com/zephyrproject-rtos/zephyr/commit/d5e7732e0d0f66bc9a3684f9597d9fd6a2d351b2 . Do we need to set any extra BIOS configuration for supporting it? Thanks!

yerabolu, 2022-05-06:
@shaoanxli: Do you see any BIOS output ? If not, could you check BIOS settings if EFI is to display-only ?

shaoanxli, 2022-05-07:
> @shaoanxli: Do you see any BIOS output ? If not, could you check BIOS settings if EFI is to display-only ?

Currently ，the daily up_squared  board cannot enter the bios , is trying to find another board or wait for chenpeng to enter the lab to see if it can be solved .

chen-png, 2022-05-12:
this is mainly caused by the bios console redirection setting, not related to zephyr code.
because now we could still get test results from monitor screen via HDMI, the only problem was that we couldn't get results from device serial COM0, because that after d5e7732, we got results from bios efi console directly, if we want to access efi via serial, we need to enable this console redirection for COM0, like below:
[image]

 

tested to enable this on QA daily up_squared, now it worked well, so closed this.

closed 2022-05-12 by chen-png