zephyrproject-rtos/zephyr #45407: Support for flashing the Zephyr based application on i.MX RT 1160 EVK through SDP Mode(USB-HID/ UART) & PyOCD runner

vishnusudhan, 2022-05-06:
Hi,

I am very new to this platform, I am using i.MX RT 1166 EVK board for my development purpose. I could able to build & flash the zephyr based application using west flash command using JTAG (JLink segger) alone.

The i.MX RT1166 EVK has the SDP mode flash support through USB-HID / UART, Is there any support / reference available to flash the zephyr based application through SDP mode (USB-HID / UART). 

Also west support PyOCD runner , I could not able to flash the zephyr application for i.MX RT1160 EVK platform, seems like PyOCD runner support for i.MX RT1160 not yet added. Suggest any steps / updated zephyr source code to flash the application using PyOCD runner for i.MX RT1160EVK.

vishnusudhan, 2022-05-19:
is there any update for the pyocd support for i.MX RT1160 EVK?

dleach02, 2022-05-20:
The board page for this board highlights:

> however the [pyOCD Debug Host Tools](https://docs.zephyrproject.org/latest/develop/flash_debug/host-tools.html#pyocd-debug-host-tools) do not yet support programming the external flashes on this board so you must reconfigure the board for one of the following debug probes instead.

This request should be moved over to https://github.com/pyocd/pyOCD project

closed 2022-05-20 by dleach02