zephyrproject-rtos/zephyr #45441: SPI NOR driver assume all SPI controller HW is implemnted in an identical way

scottwcpg, 2022-05-06:
**Describe the bug**
The Zephyr SPI NOR driver in drivers/flash/spi_nor.c calls the SoC's SPI driver to perform the transaction.
SPI NOR assumes the SPI controller hardware implementation always clocks data in when transmitting to the flash.
For example, in a simple flash read the SPI NOR driver will pass the the same array of two struct spi_buf's to the SPI driver spi_transcieve API.  The first spi_buf will contain the opcode and formatted SPI address.  The second spi_buf points to the read buffer and is set to the requested read length.  When spi_transceive is called the code sets the TX and RX spi_buf_set counts to 2.
This implies SPI controller HW that clocks in data for every byte received and this unneeded data must be read from the controller HW and either thrown away or stored somewhere.  There are old SPI controllers that work this way.  There are also modern SPI controllers that don't need this.  The Microchip QMSPI controller does not need to clock in data while transmitting.  The MCHP QMSPI driver processing each TX buffer and then moves to process each RX buffer.  I have not found any Zephyr documentation that states the SPI driver must read in a data object for each data object it transmits.

Please also mention any information which could help others to understand
the problem you're facing:
Reading data from SPI flash using the SPI driver directly works.
Using the Zephyr SPI NOR flash driver to read results in incorrect data.
For example: Read 16 bytes using flash_read()
SPI NOR driver creates an array of two struct spi_buf's
{ .buf = local buffer on stack holding 4 bytes: opcode + correct 24-bit SPI offset MSBF, .len = 4 },
{ .buf = data buffer passed in by caller, .len = 16 }

SPI NOR creates two struct spi_buf_sets: tx_set and rx_set
tx_set { .buffers = spi_buf, .count = 2 } Code logic sets count = 2 if caller supplied buffer size > 0.  Code is not taking into account read vs write
rx_set { .buffers = spi_buf, .count = 2 } Code hard code count to 2. Why?
MCHP driver will read TX size + RX size bytes instead of what caller wants: RX size.

 - What have you tried to diagnose or workaround this issue?
We have hacked the Flash SPI NOR driver with MCHP IFDEF logic.

**To Reproduce**
Steps to reproduce the behavior:
1. Microchip MEC172x EVB (board: mec172xevb_assy6906).
2. CONFIG_SPI=y, CONFIG_FLASH=y, CONFIG_SPI_NOR=y
3. Hack samples/hello_world to use flash_read() to read a know pattern from an external SPI NOR flash chip.
2. Observe data buffer has incorrect data.

**Expected behavior**
Correct usage of spi_transceive by the SPI NOR driver or clarification on what is "exact" expected behavior of SPI driver API spi_transceive.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

anangl, 2022-05-11:
The problem is not in the SPI NOR flash driver but in the Microchip XEC QMSPI driver. SPI is a full-duplex interface, so when both TX and RX buffers are passed to the `spi_transceive` function, they should be processed in parallel, not first TX then RX, like in the QMSPI driver: https://github.com/zephyrproject-rtos/zephyr/blob/4966de201407e255542362941000f3824abf974d/drivers/spi/spi_xec_qmspi.c#L534-L556
In the current form the driver will not even pass the basic [SPI loopback](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/spi/spi_loopback) test.

scottwcpg, 2022-05-12:
I can accept the bug.  I wish the SPI driver documentation was more clear in stating the buffers must be processed in parallel. 
NOTE: We have a new user requiring SPI NOR and multiple old users no using SPI NOR (using the QMSPI as is to talk to SPI flash).  We can't force all the current users to immediately update their application code.  I am proposing a QMSPI Kconfig defaulting to OFF for this fix.  What is your opinions for these alternate proposals?
1. Add temporary Kconfig for QMSPI.  Default to 'N'.  MEC172x and MEC152x QMSPI drivers behave as is. SPI NOR driver can't be used.  User's current code keeps working.  Any user wanting SPI NOR enables sets the new Kconfig to 'Y' enabling code changes the fix the bug.  This will break old code using QMSPI.
2. Create new QMSPI driver that implements correct driver behavior. New YAML and node name.  New Kconfig item(s) for the build. This driver will implement correct SPI behavior and work with SPI NOR.  One issue are HW differences between MEC152x and MEC172x: some IFDEF logic will be required.  User's of this driver will be required to override the chip QMSPI node until all users have stopped using the old drivers.  Then the old drivers can be deleted from the code tree.

closed 2022-07-26 by github-actions[bot]

greuljoSICKAG, 2023-03-08:
Hi, 
I have the exact same behavior working with nRF52833 controller and MX25R8035F or AT25SF081B flash memory. I see the correct SPI write on the oscilloscope but the SPI read gives back incorrect data.

I understand your report and where the bug comes from. However, I do not understand how I can implement a workaround to fix this problem. Could you please explain in more detail what you did so that I can transfer it to my controller?

scottwcpg, 2023-03-08:
Hi @greuljoSICKAG 
You can look at PR #55129 for the changes to drivers/spi/spi_xec_qmspi_ldma.c
It uses the helper functions from drivers/spi/spi_context.h
I believe this PR has a commit for samples and test code demonstrating how to setup the buffers when calling spi_transceive API.