zephyrproject-rtos/zephyr #45448: Request for backporting to 3.0.0

povergoing, 2022-05-07:
This is a request for backporting the following PRs/commits:

- https://github.com/zephyrproject-rtos/zephyr/pull/43309:
> on main branch:
>  fd231e32e92a5b78a0d5d18450ef1384b52a70f8 arm64: Fix booting issue with FVP V8R >= 11.16.16
>  3c1f3197e2f02f24fd1d06aeab700096a97d639d cmake: armfvp: Add FVP min version check
>  3875a1e7870804c44c9ca03fb4ac6e2f4ec38de0 board: arm64: fvp_baser_aemv8r: Update the version requirement
>  a483828d46a37630f1819f62be689a3d7c5409a7 board: arm64: fvp_baser_aemv8r_smp: Increase CONFIG_MAX_THREAD_BYTES

- https://github.com/zephyrproject-rtos/zephyr/pull/43576:
> on main branch:
>  3122b9ed107d877444d34f34111331400d45bc58 arm64: smp: Fix broadcast_ipi issue
>  65d7e64e06b298b0ee86ae373c0d1f662e0ce0e7 board: arm64: fvp_baser_aemv8r: Fix misc SMP issues
>  48cce661d5d1085f1cb0fa121f866ecdaac400c0 board: arm64: qemu_cortex_a53: Increase CONFIG_MAX_THREAD_BYTES

- https://github.com/zephyrproject-rtos/zephyr/pull/43845:
> on main branch:
> 1a682680a72d0b6788ead6bffbc60cab27caa7f8 cmake: armfvp: Improve the FVP version check
> a072a801e768ffca24a0710b20e84d9e7141edca cmake: armfvp: Remove the unnecessary parameters
> 7ea0591d30fac3c9880f1e16409ce351c59e0719 arm64: v8r: Enable AARCH64_IMAGE_HEADER by default
> 357aacab6b8a10d54e1fa4f180e3da212cc23d1d boards: fvp v8r: Fix the SRAM/FLASH to the right address

Generally, most of them fixed arm64 fvp v8r platform, one commit fixed qemu_cortex_a53 compile error

stephanosio, 2022-05-07:
@povergoing There is no need to create an issue requesting backport, you can add the `backport v3.0-branch` label to a PR and a backport PR will be automatically created for you (if there is a merge conflict, you may need to manually create one).

povergoing, 2022-05-07:
> @povergoing There is no need to create an issue requesting backport, you can add the `backport v3.0-branch` label to a PR and a backport PR will be automatically created for you (if there is a merge conflict, you may need to manually create one).

@stephanosio Thanks for reminding me. Sounds very convenient!

povergoing, 2022-05-07:
@dkalowsk @stephanosio  I have added the backport tag into the PRs that I need. So should I close this issue?

dkalowsk, 2022-05-09:
@povergoing linking the backport to this issue would help give a reason for why it was needed/done.

povergoing, 2022-05-10:
> @povergoing linking the backport to this issue would help give a reason for why it was needed/done.

The followings are the backport link:

https://github.com/zephyrproject-rtos/zephyr/pull/45452
https://github.com/zephyrproject-rtos/zephyr/pull/45450 depends on #45452
https://github.com/zephyrproject-rtos/zephyr/issues/45451 also needs #45452, I will manually fix the auto backport failure when #45452 merged

closed 2022-05-10 by mbolivar-nordic