zephyrproject-rtos/zephyr #45451: [backport v3.0-branch] backport of #43845 failed

zephyrbot, 2022-05-07:
This issue was created automatically because the backport of #43845 to `v3.0-branch` failed.

The backport to `v3.0-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v3.0-branch v3.0-branch
# Navigate to the new working tree
cd .worktrees/backport-v3.0-branch
# Create a new branch
git switch --create backport-43845-to-v3.0-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x 357aacab6b8a10d54e1fa4f180e3da212cc23d1d~4..357aacab6b8a10d54e1fa4f180e3da212cc23d1d
# Push it to GitHub
git push --set-upstream origin backport-43845-to-v3.0-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v3.0-branch
```
Then, create a pull request where the `base` branch is `v3.0-branch` and the `compare`/`head` branch is `backport-43845-to-v3.0-branch`.

povergoing, 2022-05-07:
waiting for https://github.com/zephyrproject-rtos/zephyr/pull/45452

povergoing, 2022-05-18:
https://github.com/zephyrproject-rtos/zephyr/pull/45534 meged

closed 2022-05-18 by povergoing