zephyrproject-rtos/zephyr #45458: it8xxx2_evb: tests/drivers/pwm/pwm_api assertion fail

hunterhu3000, 2022-05-07:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using -> it8xxx2_evb
 - What have you tried to diagnose or workaround this issue? -> see the comments

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b it8xxx2_evb tests/kernel/sched/schedule_api/
2. sudo -S ~/itedlb4-linux-v104/ite -f build/zephyr/zephyr.bin -s check

**Expected behavior**
The test can pass

**Impact**
Daily test will fail

**Logs and console output**
```
Running test suite pwm_basic_test
===================================================================
START - test_pwm_usec
[PWM]: 0, [period]: 2000, [pulse]: 1000
E: PWM prescaler CxCPRS only support 2 bytes !
Fail to set the period and pulse width

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/pwm/pwm_api/src/test_pwm.c:137: test_pwm_usec: test_task(DEFAULT_PWM_PORT, DEFAULT_PERIOD_USEC, DEFAULT_PULSE_USEC, UNIT_USECS) == TC_PASS is false

 FAIL - test_pwm_usec in 0.30 seconds
```
**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: zephyr-sdk-0.13.2
 - Commit SHA or Version used: 6388778fb17dfef75b5ae26abdc98d09171351dc

**Additional context**

hunterhu3000, 2022-05-07:
The commit below changed the interface of pwm, but does not remove the mircosecond test in the test case
```
commit ae91933c4add21552985dd9af50e49cad80da326
Author: Gerard Marull-Paretas <gerard.marull@nordicsemi.no>
Date:   Wed Apr 27 13:02:27 2022 +0200

    drivers: pwm: always use nanoseconds for set

    In order to be consistent with what is possible in Devicetree, always
    take a period in nanoseconds. Other scales or units may be specified by
    using, e.g., the PWM_MSEC() macros (all of them converting down to
    nanoseconds). This change then deletes the "_nsec" and "_usec" versions
    of the pwm_set call.

    Note that this change limits the period to UINT32_MAX nanoseconds,
    ~4.3s. PWM is, in generali, used with periods below the second so it
    should not be a problem.

    Signed-off-by: Gerard Marull-Paretas <gerard.marull@nordicsemi.no>
```
I'll provide a PR later

hunterhu3000, 2022-05-07:
In the file ./tests/drivers/pwm/pwm_api/src/test_pwm.c function:
``` 
static int test_task(uint32_t port, uint32_t period, uint32_t pulse, uint8_t unit)
```
The code of
```
        if (unit == UNIT_CYCLES) {
                /* Verify pwm_set_cycles() */
                if (pwm_set_cycles(pwm_dev, port, period, pulse, 0)) {
                        TC_PRINT("Fail to set the period and pulse width\n");
                        return TC_FAIL;
                }
        } else { /* unit == UNIT_NSECS */
                /* Verify pwm_set() */
                if (pwm_set(pwm_dev, port, period, pulse, 0)) {
                        TC_PRINT("Fail to set the period and pulse width\n");
                        return TC_FAIL;
                }
        }
```
The code does not have the branch of UNIT_USECS. It takes the usec parameters such as "DEFAULT_PERIOD_USEC" the same as nsec. This will cause the period/pulse value out of range in some of the boards.

hunterhu3000, 2022-05-07:
PR https://github.com/zephyrproject-rtos/zephyr/pull/45461 is created for review. Thanks

mbolivar-nordic, 2022-05-10:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/45461

closed 2022-05-10 by mbolivar-nordic