zephyrproject-rtos/zephyr #45468: Is uart_poll_in() blocking or not?

yashi, 2022-05-08:
I don't know which one is right or wrong.  There is mismatch in documentation and implementations for `uart_poll_in()`. The documentation for UART states that both `uart_poll_in()` and `uart_poll_out()` are blocking functions:

https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html

> Polling is the most basic method to access the UART peripheral. _The functions are blocking_ and the thread waits until a character is sent or received.

But, from the first commit, `uart_poll_in()` has never be a blocking function, and most implementations are non-blocking:

- https://github.com/zephyrproject-rtos/zephyr/blob/8ddf82cf70dc6f951ab477f325dee0efde3ec589/drivers/serial/k20UartDrv.c#L118-L138
- https://github.com/zephyrproject-rtos/zephyr/blob/8ddf82cf70dc6f951ab477f325dee0efde3ec589/drivers/serial/ns16550.c#L293-L311
- https://github.com/zephyrproject-rtos/zephyr/blob/8ddf82cf70dc6f951ab477f325dee0efde3ec589/drivers/serial/stellarisUartDrv.c#L288-L308

But because we have documented as blocking functions, some of the current drivers are implemented as blocking functions:

- https://github.com/zephyrproject-rtos/zephyr/blob/6388778fb17dfef75b5ae26abdc98d09171351dc/drivers/serial/uart_msp432p4xx.c#L155-L162
- https://github.com/zephyrproject-rtos/zephyr/blob/6388778fb17dfef75b5ae26abdc98d09171351dc/drivers/serial/uart_imx.c#L105-L118

**Describe the solution you'd like**
I'd like to know our true intention. I can create a PR for documentation fix if `uart_poll_in()` should be a non-blocking function.  Or, create issues against drivers with blocking `_poll_in()` function.

I also want to know that, in case `uart_poll_in()` is a non-blocking function, what the "poll" means in Zephyr.  There are some functions with the word "poll".  Does it imply "blocking"?

mbolivar-nordic, 2022-05-10:
@yashi based on the release working group discussion:

- The functions are not meant to be blocking based on their existing documentation in Doxygen, so the bug is in the .rst documentation page and in the drivers which implement this according to the .rst instead of doxygen
- Yes, could you please raise bugs against the drivers which are busy-waiting forever?
- Do you think you can send a documentation PR to remove the incorrect information from .rst?

yashi, 2022-05-11:
> * The functions are not meant to be blocking based on their existing documentation in Doxygen, so the bug is in the .rst documentation page and in the drivers which implement this according to the .rst instead of doxygen

OK. Thanks.

> * Yes, could you please raise bugs against the drivers which are busy-waiting forever?

- #45532
- #45533

> * Do you think you can send a documentation PR to remove the incorrect information from .rst?

Will do.

commit bc35b134 (2022-05-13): doc: uart: Make sure uart_poll_in is non-blocking

The documentation has wrongly stated that the function uart_poll_in() is
also a blocking function.  The uart_poll_out() is indeed a blocking
function but uart_poll_in() has never been since day one.

Make it clear that the uart_poll_in() is a NON-blocking function, and
uart_poll_out() IS a blocking.

This fixes #45468.
closed 2022-05-13 by mbolivar-nordic