zephyrproject-rtos/zephyr #45486: MCUBootloader can't building for imxrt1160_evk_cm7 core

vishnusudhan, 2022-05-09:
I tried the following procedure to build the MCU Bootloader for imxrt1160 EVK for CM7 core.

1. In the ZEPHYRBASE/bootloader/mcuboot/zephyr/prj.conf file , I included KEY value as    CONFIG_BOOT_SIGNATURE_KEY_FILE="root_rsa_2048.pem " (for singing the image)
2.  Now using the "**west build -s boot/zephyr -b mimxrt1160_evk_cm7**" , I build the MCUBootloader.
      While building , I observed fatal error as ,
           home/vishnusudhan/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd:       zephyr/zephyr_pre0.elf section `**rodata' will not fit in region `FLASH'**
          /home/vishnusudhan/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld./ld.bfd: **region `FLASH' overflowed by 632 bytes**

[image]

Kindly provide suggest how to overcome this issue/ any reference doc / steps available to find the solution for this issue?

vishnusudhan, 2022-05-11:
I have also tried to build the unsigned image, just enabled CONFIG_BOOTLOADER_MCUBOOT=y & start to build the MCUBootloader by giving command as "**west build -s boot/zephyr -b mimxrt1160_evk_cm7**", but still getting same issue as 

**home/vishnusudhan/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section **rodata' will not fit in region FLASH'**
/home/vishnusudhan/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld./ld.bfd: region `FLASH' overflowed by 632 bytes**

danieldegrasse, 2022-05-12:
@vishnusudhan, using the changes to the partition layout for the RT1160 captured in #45602, as well as increasing the maximum number of sectors supported by MCUBoot (see the required changes here: https://github.com/mcu-tools/mcuboot/pull/1365), I was able to use MCUBoot on this platform. Can you verify that MCUBoot works for you with these changes?

PR #45602 (2022-07-07): boards: mimxrt11xxx: update partition alignment for MCUBoot
update partition alignment for iMX RT1170 and RT1160 to enable storing
MCUBoot image in boot partition.

Fixes #45486
commit 8f6fef93 (2022-07-07): boards: mimxrt11xxx: update partition alignment for MCUBoot

update partition alignment for iMX RT1170 and RT1160 to enable storing
MCUBoot image in boot partition.

Fixes #45486
closed 2022-05-13 by vishnusudhan