zephyrproject-rtos/zephyr #45505: NXP MIMXRT1050-EVKB: MCUBoot Serial Recover: mcumgr hangs when trying to upload image

kunoh, 2022-05-10:
**Describe the bug**
When entering serial recovery mode in MCUboot and trying to upload an image, the progress bar stays at 0B`
```bash
~/zephyrproject$ mcumgr --conntype serial --connstring "/dev/ttyACM0,baud=115200" image list
Images:
 image=0 slot=1
    version: 0.0.0.0
    bootable: false
    flags: 
    hash: Unavailable
Split status: N/A (0)

~/zephyrproject$ mcumgr --conntype serial --connstring "/dev/ttyACM0,baud=115200" image upload build-smp_svr/zephyr/zephyr.signed.bin 
 0 B / 74.59 KiB [--------------------------------------------------------------------------------------------------------------------------------]   0.00%
```

**To Reproduce**
Board config bootloader/mcuboot/boot/zephyr/boards/mimxrt1050_evk_qspi.conf
```bash
# Copyright 2021 NXP
# SPDX-License-Identifier: Apache-2.0

CONFIG_BOOT_MAX_IMG_SECTORS=1024

# Disable Zephyr console
CONFIG_CONSOLE=n

# Multithreading
CONFIG_MULTITHREADING=y

# MCUboot serial recovery
CONFIG_MCUBOOT_SERIAL=y
CONFIG_BOOT_SERIAL_DETECT_PORT="GPIO_5"
CONFIG_BOOT_SERIAL_DETECT_PIN=0
CONFIG_BOOT_SERIAL_DETECT_PIN_VAL=0
CONFIG_BOOT_SERIAL_DETECT_DELAY=450
CONFIG_MCUBOOT_INDICATION_LED=y
```

west build -b mimxrt1050_evk_qspi bootloader/mcuboot/boot/zephyr/ -d build-mcuboot -- -DOVERLAY_CONFIG='overlay-rsa.conf;serial_recovery.conf' -DCONFIG_BUILD_OUTPUT_HEX=y

west flash -f build-mcuboot

Entering recovery mode by pressing and holding SW8 button during MCU reset ( LED lights up )

Then try to upload any image as showed under "Describe the bug"

**Expected behavior**
Progress bar hangs at 0B during upload

**Impact**
In case of uploading a bricked fw it would be nice to have a way to recover.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain Zephyr SDK 0.14
 - Commit: Zephyr e97a2ee6d0d64a3c1711942db854e6ccf78ea8d6
                  MCUboot e86f575f68fdac2cab1898e0a893c8c6d8fd0fa1

danieldegrasse, 2022-05-11:
@kunoh Just wanted to give you a heads up- I am looking into this right now, and it seems that the issue is likely similar to the USB DFU mode problem. Decreasing the size of the MCUBoot partitions (from 3MB to 128K) fixed the issue for me. I'm looking into enabling progressive erase for MCUBoot now, to see if the full partition size can be supported.

danieldegrasse, 2022-05-11:
@kunoh Progressive erase can be supported by fixing the flash node name in the RT1050 QSPI DTS file (see PR #45569). With this change and `CONFIG_BOOT_ERASE_PROGRESSIVELY=y`, I was able to download an image using the MCUBoot serial recovery feature and mcumgr. Please let me know if these changes work for you.

closed 2022-05-18 by dleach02

kunoh, 2022-05-23:
Sorry for my late response @danieldegrasse. I have just tested and it works perfectly, thank you once again :)

PR #45569 (2022-05-18): boards: mimxrt1xxx: change flash controller to correct node
flash controller on RT boards should be the flash device, rather than
the flexspi, as the flash device's driver implements flash APIs.

Fixes #45505, when combined with https://github.com/mcu-tools/mcuboot/pull/1364 in order to enable progressive erase for these platforms.

This change is required on the RT1050 and RT1060 to enable support for progressive erase, but I have updated the remaining RT platforms because the FlexSPI flash controller itself does not implement the flash driver API, the flash chip device object does.
commit dfabb2e3 (2022-05-18): boards: mimxrt1xxx: change flash controller to correct node

flash controller on RT boards should be the flash device, rather than
the flexspi, as the flash device's driver implements flash APIs.

Fixes #45505