zephyrproject-rtos/zephyr #45514: UDP Packet socket doesn't do L2 header processing

krish2718, 2022-05-10:
**Describe the bug**
Datagram socket (https://docs.zephyrproject.org/3.0.0/reference/kconfig/CONFIG_NET_SOCKETS_PACKET_DGRAM.html) should handle L2 header processing i.e., adding L2 header from sockaddr_ll in TX and removing L2 header and populating sockaddr_ll in the RX. But this is not being done and the packet is sent as is.

**To Reproduce**
Steps to reproduce the behavior:
Run the unit test: https://github.com/zephyrproject-rtos/zephyr/pull/45513 

**Expected behavior**
The unit test should pass.
**Impact**
Applications should do the extra L2 handling

**Logs and console output**

**Environment (please complete the following information):**
 - OS:Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 523cd60a77

rlubos, 2022-05-11:
Thanks for the report @krish2718 

I've investigated the issue briefly and noticed two problems with the UDP packet socket handling:
1. The ethernet header does not seem to be filled correctly on the TX path, for example, the EtherType field in the header is empty.
2. Because of this, the packet is rejected by the Ethernet L2, when the packet is being processed on the RX path. At the same time, the `net_packet_socket_input()` function does not seem to differentiate between `SOCK_RAW` and `SOCK_DGRAM ` sockets, wrongly forwarding the unprocessed packet to the `SOCK_DGRAM` socket.

I will investigate further and look for viable solutions for the aforementioned issues.

krish2718, 2022-05-11:
Thanks, Robert. It is inline with my investigations as well, for now, I have workedaround this in the application to do L2 processing.

PR #45592 (2022-05-27): net: sockets: Fix datagram AF_PACKET socket handling
This PR fixes the datagram AF_PACKET socket handling, which improperly received packet a packet before it was processed by L2 layer.

Additionally, fix the existing unit test to work correctly after update, so that the packets used in the test are not rejected by L2. Finally add two new unit tests for DGRAM sockets to improve test coverage.

Fixes #45514
closed 2022-05-27 by mbolivar-nordic