zephyrproject-rtos/zephyr #45528: Support LittleFS and Settings subsystem on LPC55s69

danieldegrasse, 2022-05-10:
**Is your enhancement proposal related to a problem? Please describe.**
The LPC55s69 flash driver does not support using LittleFS with the default configuration or the Zephyr settings subsystem. This is due to the requirement enforced by the LPC flash driver that all writes be in multiples of 512 bytes.

**Describe the solution you'd like**
The LPC flash driver should support bytes of less than 512 bytes.

**Describe alternatives you've considered**
Subsystems could be updated to support 512 byte writes, but this will just result in reimplementations of "unaligned write" algorithms in most subsystems.

Laczen, 2022-05-11:
@danieldegrasse, I have looked at the issue and the proposed solution (#45526 ). The solution uses a read/modify/erase/write method that will increase the wear and might leave zephyr users breaking their system early. Next to this it also might result in a broken littlefs system or lost data (nvs) whenever a write is interrupted.

For nvs let's consider a flash area (page) that contains ate's, these will be 8 byte long and for the page to be completely written it will be erased 64 times. Every write of an ate will erase a page, and every write can result in loosing all other ate's in that area. Nvs does not play nice with these kind of flashes, and nvs shouldn't be used on them.

For littlefs the same holds, and it is not required/advantageous. Littlefs should be defined with a block size of 512 bytes and everything will be taken into account, the fs will try to limit flash writes. And as a bonus the fs will not be broken when a write is interrupted.

thedjnK, 2025-01-14:
@rghaddab seems this can be closed as fixed?

rghaddab, 2025-01-15:
> [@rghaddab](https://github.com/rghaddab) seems this can be closed as fixed?

Using ZMS, LPC55s69 can be supported by adding CONFIG_ZMS_CUSTOMIZE_BLOCK_SIZE=y and CONFIG_ZMS_CUSTOM_BLOCK_SIZE=512
So this issue can be closed

danieldegrasse, 2025-02-14:
I'm ok to close this, yeah- the write size limitations of the LPC55S69 make some filesystems impractical unfortunately

closed 2025-02-17 by rghaddab