zephyrproject-rtos/zephyr #45542: Implementing firmware image decompression in img_mgmt_upload()

markusarnoldsson, 2022-05-11:
Hey!

I am working on implementing decompression of a firmware image file. I believe this should be done in img_mgmt_upload() and decompressed chunks.

I further believe that It should be within the cbor_read_object, since before it, all values are at zero and after they are the compressed values. After cbor_read_object, upload_inspect and so forth are called, where I want the decompressed values to be ready, not the compressed. 

My question is, where in cbor_read_object -> cbor_internal_read_object should a decompression algorithm be placed?

de-nordic, 2022-05-11:
This is not an issue, this is topic for discussion.

closed 2022-05-11 by markusarnoldsson

de-nordic, 2022-05-11:
@markusarnoldsson You can just press the "Convert to discussion" at the right bottom of the screen (next to green Comment).

de-nordic, 2022-05-11:
And the `cbor_read_object` is no more because we have converted mcumgr to zcbor.