zephyrproject-rtos/zephyr #45543: Build samples/bluetooth/broadcast_audio_sink raises an error

CXsCode, 2022-05-11:
**Describe the bug**
Direct build samples/bluetooth/broadcast_audio_sink for nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:

cd zephyrproject/zephyr
west build -b nrf52840dk_nrf52840 -p auto -d 52840 .\samples\bluetooth\broadcast_audio_sink\

**Expected behavior**
Is this a bug or does this demo not support nrf52840dk_nrf52840?

**Logs and console output**
warning: default value 1 on BT_CTLR_ADV_SYNC_SET (defined at subsys/bluetooth\controller\Kconfig.ll_sw_split:182) clamped to 0 due to being outside the active range ([1, 0])

error: Aborting due to Kconfig warnings

**Environment (please complete the following information):**
 - OS: Windows10
 - Toolchain: arm_none_eabi/9_2020_q2_update
 - Commit: 3.0.99

Thalley, 2022-05-13:
Already fixed by https://github.com/zephyrproject-rtos/zephyr/pull/45015

CXsCode, 2022-05-16:
Building success

Thalley, 2022-05-16:
Reopened since https://github.com/zephyrproject-rtos/zephyr/pull/45015 isn't merged yet :) This issue will automatically be closed once that gets merged.

closed 2022-05-19 by carlescufi