zephyrproject-rtos/zephyr #45564: Zephyr does not boot with CONFIG_PM=y

gordonklaus, 2022-05-11:
I am running Zephyr v2.7.2 on the nucleo_l552ze_q board.

Not long after enabling CONFIG_PM, I produced a build that simply does not boot.  Where there is normally a banner announcing the Zephyr version, there is nothing, and the system has clearly halted (my normally blinking LED is dark).

Oddly, arbitrary changes to the code/configuration make the problem disappear and reappear.  For example, putting some code in a loop; adding logging; returning early from main; enabling CONFIG_ASSERT or CONFIG_DEBUG -- any of these might cause the program to boot again, or to stop booting.

Importantly, it is consistent:  With a particular code and configuration, it always either boots or does not.  Better than an intermittent bug, but still very troubling 🙂 

I normally run MCUBoot but I can reproduce the problem without it.

I did not see the problem before introducing power management, so I strongly suspect it is related.

I have produced a minimal configuration that reproduces the problem.  Remove/modify any part of it and the problem goes away:

CMakeLists.txt:
```cmake
cmake_minimum_required(VERSION 3.20.0)

set(BOARD nucleo_l552ze_q)
set(DTC_OVERLAY_FILE board.overlay)

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(ulmo)

target_sources(app PRIVATE
	src/main.c
)
```

board.overlay:
```dts
&cpu0 {
	cpu-power-states = <&stop0 &stop1 &stop2>;
};

&rcc {
	apb1-prescaler = <2>;
	apb2-prescaler = <2>;
};

&lptim1 {
	status = "okay";
};

&lpuart1 {
	pinctrl-0 = <&lpuart1_tx_pa2 &lpuart1_rx_pa3 &lpuart1_rts_pb1 &lpuart1_cts_pb13>;
	current-speed = <115200>;
	status = "okay";

	gsm: gsm-modem {
		compatible = "zephyr,gsm-ppp";
		label = "gsm_ppp";
	};
};

&usart2 {
	pinctrl-0 = <&usart2_tx_pd5 &usart2_rx_pd6>;
	current-speed = <115200>;
	status = "okay";
};

&usart3 {
	pinctrl-0 = <&usart3_tx_pd8 &usart3_rx_pd9 &usart3_rts_pd12 &usart3_cts_pd11>;
	current-speed = <19200>;
	status = "okay";
};

&uart4 {
	pinctrl-0 = <&uart4_tx_pc10 &uart4_rx_pc11>;
	current-speed = <9600>;
	status = "okay";
};

/ {
	power-states {
		stop0: state0 {
			compatible = "zephyr,power-state";
			power-state-name = "suspend-to-idle";
			substate-id = <1>;
			min-residency-us = <100>;
		};
		stop1: state1 {
			compatible = "zephyr,power-state";
			power-state-name = "suspend-to-idle";
			substate-id = <2>;
			min-residency-us = <500>;
		};
		stop2: state2 {
			compatible = "zephyr,power-state";
			power-state-name = "suspend-to-idle";
			substate-id = <3>;
			min-residency-us = <900>;
		};
	};

	chosen {
		zephyr,console = &usart2;
		zephyr,shell-uart = &usart2;
		zephyr,gsm-ppp = &lpuart1;
	};
};
```

prj.conf:
```conf
CONFIG_PM=y
CONFIG_PM_DEVICE=y

CONFIG_MODEM=y
CONFIG_MODEM_GSM_PPP=y
CONFIG_GSM_PPP_AUTOSTART=n

CONFIG_NETWORKING=y
CONFIG_NET_DRIVERS=y
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y

CONFIG_LOG=y
```

src/main.c:
```c
void main() {}
```

gordonklaus, 2022-05-12:
Update:  The behavior is not as consistent as I previously thought (perhaps I didn't do many repetitions before).  I now see the behavior changing after reset, for a particular firmware image.  Sometimes it boots, sometimes not.

I should say that the impact is considerably higher now, if we can't be sure after a single test whether the program will boot or not.

gordonklaus, 2022-05-12:
@erwango I'm able to reproduce it on v2.7.2 and v2.7.1, but not on v2.7.0.  It's hard to be totally sure about v2.7.0, as the same program doesn't always show the same behavior between versions (e.g. my minimal reproducer did not have the issue (but I didn't have any trouble finding a variant that did)).  But I did try several variants and they all booted on v2.7.0.

I'll revert to v2.7.0 for now and I'll keep you posted if the problem shows up again.

Let me know if there are any specific revisions you'd like me to test.

gordonklaus, 2022-05-16:
I spent a couple tedious hours trying to narrow down which revision could have caused this bug.  I would take these results with a small grain of salt because it has proven difficult to find code/configuration which provokes the bug.  That is, even though I wasn't able to provoke the bug in a particular revision, it may still be that it exists there.

That said, my testing indicates that the bug might have been introduced in c65645cc39e4a3513ac754733266766459b3cef0.  Does this seem with the realm of possibility, @erwango ?

Here are all of the revisions I tested (YES means I saw the bug, NO means no):
v2.7.1 - YES
5c08f183f2bcd428df3c42f76c5b130f771cf101 - YES
525c112ac3195661583f618b0e63275b64faf274 - YES
edd1612388faf0524ac0c41aac9826ec32fbd760 - NO
7e2e9d612db6045bab0a47b5d4cd225e2febdc41 - YES
7bdf17756c815e7d4d24cd2b39b5bcb9d9c854c9 - YES
053dc7572062845255aeb3551a1e904b8bf9fc09 - YES
b7954c9991d753177ad5a4ae984d2ae1a5d37a12 - YES
f5d4fb40b510d86996e2f7bf2acd682f059120e1 - YES
6a851f1f485a59b7bc50a82e8d2d1d9d6e9435ed - YES
68a6a3e5c3d44be1309dd2297326533b8b93c8fd - YES
dff331a40c507ce111501a45164542445ec283d7 - YES
c65645cc39e4a3513ac754733266766459b3cef0 - YES
cef9cbeb608117b8babfc4aea026d9c1805ab382 - NO
v2.7.0 - NO

It would be prudent to check some even earlier revisions (pre-2.7.0), but I don't have any more time for it today.

gordonklaus, 2022-05-16:
@erwango Can you think of a better way to diagnose the problem than mere trial?

erwango, 2022-05-16:
> It would be prudent to check some even earlier revisions (pre-2.7.0), but I don't have any more time for it today.

Thanks for this work. Given that the SHA1 you pointed doesn't have impact on binary and it is so close to V2.7.0, I'm pretty sure this is not a regression but rather a bug that could be silent under specific conditions.

i'll try to reproduce on my end and keep you posted.

erwango, 2022-05-17:
@gordonklaus Unfortunately, I'm not able to reproduce on my end.
The only difference with your set up is that I kept console on lpuart1. can you check if this is a real reproduction condition ?

gordonklaus, 2022-05-18:
@erwango I'm also not able to reproduce it when I move the console back to lpuart1 (I tried many configurations, but I suppose it's still possible I didn't search long enough).  Do you think this could somehow be related?

I was able to reproduce it on another identical Nucleo board, FWIW.

gordonklaus, 2022-05-18:
@erwango I'm not sure how to explain that c65645cc39e4a3513ac754733266766459b3cef0 should have no effect on the binary, yet I saw different behavior.

I `diff`ed the binaries before and after that change and they only differ in the boot banner (which include the git revision) and in two 16-bit words that are swapped at lower address:
```diff
4902c4902
< 00013250: ff7f 0600 0100 ff7f ff7f ff7f ff7f ff7f  ................
---
> 00013250: ff7f 0100 0600 ff7f ff7f ff7f ff7f ff7f  ................
5318,5319c5318,5319
< 00014c50: 322e 372e 302d 352d 6763 3635 3634 3563  2.7.0-5-gc65645c
< 00014c60: 6333 3965 3400 2a2a 2a20 426f 6f74 696e  c39e4.*** Bootin
---
> 00014c50: 322e 372e 302d 342d 6763 6566 3963 6265  2.7.0-4-gcef9cbe
> 00014c60: 6236 3038 3100 2a2a 2a20 426f 6f74 696e  b6081.*** Bootin
```

But neither of these binaries reproduces the problem (and I can't find a configuration that does again at that revision) :(

closed 2022-08-01 by github-actions[bot]