zephyrproject-rtos/zephyr #45566: Out of tree soc sample

junkbondtrader, 2022-05-11:
**Is your enhancement proposal related to a problem? Please describe.**
I have been unable to pull a riscv-privilege soc out of tree. Placing it in an application/soc/riscv/riscv-privilege folder leads to the Kconfigs seemingly not being discovered. Placing it in an application/soc/riscv folder results in unresolved dependencies.

**Describe the solution you'd like**
I would love to see a sample of an out of tree soc, particularly that is part of an soc family.

**Describe alternatives you've considered**
I've tried getting this working on my own without success. I haven't been able to find examples of others doing so successfully. I've tried this with my own custom soc that works in tree as well as with the sifive-freedom soc as a sanity check (and to attempt to show what an unsuccessful example looks like).

**Additional context**
I'm happy to provide any additional context that would be helpful. My end goal is just to get this working but I think a sample would more useful to others going forward.

henrikbrixandersen, 2022-11-17:
> I'm happy to provide any additional context that would be helpful.

A non-working example repo along with error messages would be good.

OkannShn, 2023-10-12:
Did you find a solution for out-of-tree soc porting?

kartben, 2026-10-30:
Closing since the SoC layout described here was replaced by hardware model v2, where out-of-tree SoCs are described with soc.yml (see the SoC porting guide). Please open a new issue if you hit problems with the current model.

closed 2026-10-30 by kartben