zephyrproject-rtos/zephyr #45579: Application defined Shell Command not registered

tfir1981, 2022-05-12:
**Describe the bug**
I implemented some submodules in my Zephyr-Project with some shell command impelmentations. When i was still working on zephyr version 2.6.0 everything worked fine.

Now i did an Update to Zephyr 2.7.1 and later to 2.7.2. Since the Update i faced the problem, that my shell command have disappeared. When i went back to 2.6.0 everything is fine again.

If the source files with my shell commands are moved to `/src`-directory (in project root), the shell commands are available again. Any other directorys are not working.
Another way to get back the commands is to create an `shell_cmd.c` file `/src`-directory (in project root) and include my shell-command-implementation-files like:

```c
#include "my_cmd_1.c"
#include "my_cmd_2.c"
```
I think it has something to do with build-system settings (cmake)? Registration of the commands may not work, because in map-File i cannot find any symbols concerning my commands. 
Additionally i did some tries with different gcc optimization settings but no success.

My project is based on a STM32F4 platform.

I am working with PlatformIO, but i think that is not a possible reason, because Zephyr 2.6.0 still works.

**Expected behavior**
My commands should appear in shell.

**Impact**
Key features are not available in my application. showstopper

**Logs and console output**
No logs. Shell commands not appear.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.4 LTS x86_64
 - Toolchain: Platformio Core 5.2.5·Home 3.4.1
 -  Zephyr 2.7.2

carlescufi, 2022-05-17:
@tfir1981 could you please try to see if this issue is present in the `main` branch as well?

ThomasFike, 2022-05-18:
Have you verified that those files are being compiled when not in your src directory? You could check this quick by adding #error to one of the files. Then make sure you run a pristine build and see if it errors at the error you added. It sounds like the files are not being compiled from what your saying.

tfir1981, 2022-05-19:
> Have you verified that those files are being compiled when not in your src directory? You could check this quick by adding #error to one of the files. Then make sure you run a pristine build and see [...]

Yes, when my Code is not in src-directory it is compiled. On one hand i got an error if adding #error on other hand i can see compiler log. .o-File and .a-File were created.
Another test i did is adding an error inside of makro `SHELL_CMD_REGISTER` by changing symbolnames.

tfir1981, 2022-05-24:
In my PlatoformIO i addes `-Wl,--whole-archive` to my build_flags. This is the solution.
Thanks for your help.
So this issue is solved for me.

closed 2022-05-24 by tfir1981