zephyrproject-rtos/zephyr #45580: ESP32-C3: CONFIG_ESP32_PHY_MAX_TX_POWER undeclared error when building with CONFIG_BT=y

pflonz, 2022-05-12:
**Describe the bug**
As soon as I activate the flag CONFIG_BT I get the following error during the build of the firmware:

```
In file included from /home/user/workspace/modules/hal/espressif/components/esp_wifi/src/phy_init.c:31:
/home/user/workspace/modules/hal/espressif/zephyr/esp32c3/../../components/esp_wifi/esp32c3/include/phy_init_data.h:49:15: error: 'CONFIG_ESP32_PHY_MAX_TX_POWER' undeclared here (not in a function)
   49 |         LIMIT(CONFIG_ESP32_PHY_MAX_TX_POWER * 4, 0, 0x50),
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
    - ESP32-C3 DevKitM1
 - What have you tried to diagnose or workaround this issue?
    - If I enable the following flags I am able to build the application again:
      ```
      CONFIG_NETWORKING=y
      CONFIG_NET_L2_ETHERNET=y
      
      CONFIG_WIFI=y
      CONFIG_WIFI_ESP32=y
      CONFIG_MBEDTLS=y
      CONFIG_MBEDTLS_ENTROPY_ENABLED=y
      CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
      CONFIG_MBEDTLS_ECP_ALL_ENABLED=y
      ```
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Zephyr application with an empty main() function
2. Run `west espressif update`
3. `west build -p auto -b esp32c3_devkitm -s my_project_folder`
4. See error

**Expected behavior**
A successful build without me having to enable the additional CONFIG flags. 

**Impact**
showstopper - Because the fimrware image becomes to large

**Logs and console output**
nothing

**Environment (please complete the following information):**
 - OS: Windows 10 with WSL2 (development environment runs inside a Docker container)
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: Zephyr (last main) commit: [806ee86](https://github.com/zephyrproject-rtos/zephyr/commit/806ee8601ca73b0de0767b2a22c533e307bcee8b)

**Additional context**
nothing

pflonz, 2022-05-17:
@sylvioalves Maybe you can support here or give more information where the dependency between BT and WIFI interface comes from? Many thanks in advance!

sylvioalves, 2022-05-17:
Hi, I'll submit the PR asap.  
You have to add the following in `soc/xtensa/esp32/Kconfig.soc`:  

```c
config ESP32_PHY_MAX_WIFI_TX_POWER
	int "Max WiFi/BLE TX power (dBm)"
	range 10 20
	default 20
	help
	  Set maximum transmit power for WiFi radio. Actual transmit power for high
	  data rates may be lower than this setting.

config ESP32_PHY_MAX_TX_POWER
	int
	default ESP32_PHY_MAX_WIFI_TX_POWER
```

PR #45716 (2022-05-17): soc: esp32: fix missing configuration
Closes #45580
closed 2022-05-17 by carlescufi