zephyrproject-rtos/zephyr #45591: settings: Loading settings from NVS

Damian-Nordic, 2022-05-12:
**Is your enhancement proposal related to a problem? Please describe.**
I'm working on a project that writes a few settings on each boot. It turns out it causes performance degradation, namely, each boot takes about 70ms more than the previous one:

```
Reboot, Boot time
0,      2340 ms
1,      2467 ms
2,      2559 ms
3,      2668 ms
4,      2651 ms
5,      2770 ms
6,      2946 ms
7,      2910 ms
8,      3034 ms
9,      3047 ms
10,     3145 ms
11,     3182 ms
12,     3243 ms
13,     3336 ms
14,     3362 ms
15,     3470 ms
16,     3542 ms
17,     3626 ms
18,     3689 ms
19,     3735 ms
20,     3789 ms
```

Some application characteristics in the tested scenario:
- Due to the nature of the solution - we don't know all possible keys in advance - the application must use `settings_save_one`/`settings_load_subtree_direct` APIs rather than settings handlers.
- Number of keys in the settings store is about 30.
- Number of keys that the app is trying to read on boot is about 100.
- Number of keys written on boot: 5 (small 4 byte counters).

I think the current implementation of settings over NVS is inefficient; it tries to leverage the existing NVS API that doesn't allow for fast key lookup.  Let me give you some statistics from the 20th boot of the application:
```
Number of settings loads: 98 * 32056us (average time)
Number of NVS reads (key lookup): 1692 * 943us (average time)
Number of NVS reads (value probe): 1692 * 836us (average time)
Number of value reads: 1563 * 74us (average time)
Number of NVS ATE reads: 735614
```

Thus 100 setting loads results in 735614 ATE reads. It's the consequence of the current `settings_load_subtree_direct` implementation (pseudocode):
```pseudocode
settings_load_subtree_direct(subtree, callback):
	max_key_id <- nvs_read(0x8000) // about 30 in our case
	
	for id in [0x8001, max_key_id]
		key <- nvs_read(id)
		value <- nvs_read(id + 0x4000) 

		if not key and not value then
			continue

		if not key or not value then
		    // delete orphaned key/value
		    nvs_delete(id)
		    nvs_delete(id + 0x4000)

		// call settings_loader which validates that 'key' matches 'subtree' and runs
		// another nvs_read(id + 0x4000) using the user's buffer for data.
		settings_loader(key, subtree, callback)
```

Important observations:
- The load time linearly depends on the number of keys.
- We read both key and value and even if the key doesn't match the request.
- Keys and values are stored in the same NVS and keys rarely change, so they will likely be located close to the end of the NVS history - we may expect a lot of ATE reads during the key lookup.

**Describe the solution you'd like**
We've come up with a few ideas and not all of them are mutually exclusive (we can implement a few of them):

**1. Don't read the value if the key doesn't match the request**
It's an obvious and simple optimization - move the filtering by requested key/subtree as early as possible. The integrity check that compares if there are no keys without values or the other way around could be done on the settings initialization. We may expect about 50% load time reduction (not enough)

**2. Cache/map from setting key to NVS ID**
We can maintain an array: `[hash(key) % CACHE_SIZE] -> NVS ID` to help quickly find the NVS ID that **likely** contains the requested key. Note that it's just a hint, so if the NVS ID does not contain the desired key, we may fallback to the existing loop over all possible IDs. That way we can avoid implementing a complex cache solution with cache eviction and substantial RAM overhead - for 100 keys we need to devote just 200B of RAM. Note that a cache miss allows us to quickly conclude that a given key does not exist.

**3. nvs_walk**
Instead of the cache we can also extend the NVS API by adding some nvs_walk(min_id, max_id, callback) function that would allow a user to walk through all ATEs for a given ID range and have a user-provided callback called for each entry. The callback could read data associated with a given ATE and compare it with the requested key. `nvs_walk` function could be used by settings_nvs backend to find NVS ID of a given key in a single pass.

**4. Cache from NVS ID to the most recent ATE**
Rather than optimizing settings, we may think about optimizing NVS itself. Currently, reading an NVS value takes averagely linear time with respect to the number of ATEs. We may try to make it averagely constant by implementing a similar idea to 2, but at the NVS layer. Namely, we can maintain an array: `[NVS ID % CACHE_SIZE] -> Most recent ATE position`. Like before, the value stored in the cache can be used as a hint only to avoid the need for resolving cache collisions/overruns and the implementation could fallback to the linear lookup if the hint is incorrect.

Any other ideas?

**Describe alternatives you've considered**
Change the current scheme for allocating NVS IDs for setting keys to use a hash function rather than "give me the first free ID" approach. That could help us achieve a direct hit instead of looping over all possible IDs. The solution, however, wouldn't help to existing products with already written settings.

**Additional context**
N/A

Damian-Nordic, 2022-05-12:
@Laczen fyi

Laczen, 2022-05-12:
I have no experience at all with OpenThread or Matter. I have seen that OpenThread has a method to directly use flash instead of using the settings system. Maybe that is an alternative approach.

Damian-Nordic, 2022-05-12:
I'm not sure about OpenThread, but in Matter we definitely need some sort of key-value store, so if we use flash directly it only means that we need to implement the settings-like layer ourselves or use another open source implementation, which is one option, but of course, we would prefer to invest in a common, reusable solution rather than project-specific solutions.

While we probably use settings more than a typical embedded application, it's by far not a massive use, so it's pretty likely that other users of settings may at some point encounter similar issues as we do. So that may be a good opportunity to think it through  unless we have reasons to believe that fast storage is not critical feature of Zephyr.

Laczen, 2022-05-12:
@Damian-Nordic, some history information:
a. The settings system has been designed basically as a read-once system. The addition of `settings_load_subtree(_direct)` has never been my favorite addition because the usage breaks this read-once rule. This addition comes at the disadvantage of speed.
b. The nvs backend for settings has been created to minimize flash wear, as a key will normally not change it would be disadvantageous to store the key on each write. This makes the system slower but wear is more important.
c. The settings system should be backend unaware, as it is only nvs that separates the key from the value the proposal under 1. (see above) is not really viable.

Regarding the other proposals:
* nvs_walk: this is possible but would work because the information about deleted entries is lost when walking.
* hashing: I have been thinking about this some time ago, the problem is that it is needed to handle multiple matches and this will complicate the storage solution.

IMO the most appropiate and fastest solution would be to rethink how the settings system is used and try to avoid calling `settings_load_subtree_direct()`. And if it is not possible using the settings system use nvs as a direct key-value storage (this could be combined with the settings system as the settings system is only using id's 0x8000-...).

Laczen, 2022-05-12:
Regarding proposal 1: The nvs backend does not do any comparison of keys or values, it just prepares them for the settings system. If however a entry is deleted it is not allowed to call the settings_handler, this is why it needs to see if the value is there.

Regarding the caches: Maybe I am misundestanding your proposal, suppose we have two keys that give the same hash how will this be handled (take into account that there is no way of knowing when a value was added).

Damian-Nordic, 2022-05-12:
What I meant in proposal 1 is that if I want to load a key "a/b/c",  when checking NVS ID 0x8001 we should first just load the NVS data and compare it with "a/b/c" rather than go straight to reading NVS ID 0x8001 + 0x4000 just to realize in the callback that the data doesn't match "a/b/c".

Ad caches: if I save a setting and I determine that NVS ID to store my key "a/b/c" is 0x8001 I simply put 0x8001 to the cache at position `hash("a/b/c")%CACHE_SIZE`. If I then write another key, but I'm unlucky and it results in the same cache position, the previous value is then overwritten. Then when I want to read the setting, I first use the cache do find out possible NVS ID for the key and I start from that, so instead of doing:
```
for id in [0x8001, max_key_id]:
    key <- nvs_read(id)
    ...
```
I do:
```
for id in [cached_key_id, max_key_id] + [0x8001, cached_key_id - 1]:
   key <- nvs_read(id)
```
So I consider the same set of possible IDs but start from the most likely one.

Laczen, 2022-05-12:
@Damian-Nordic, don't forget the settings backends have no `load(*key, ...)` method. The backends have no idea what key is requested. The settings system would first need to be extended with such a load method.

Regarding the caches, this could work inside the nvs_settings_backend.

Laczen, 2022-05-13:
@Damian-Nordic, could you change the title in: using the settings system as a key-value store and retrieve system is slow.

The answer to that is yes, because that is not the intention of the settings system (otherwise it would be called kv_store or similar). 

Zephyr has no key-value store and retrieve system but adding one on top of e.g. nvs should not be a problem.

Damian-Nordic, 2022-05-16:
@Laczen Sure, I can change the title.

I don't quite understand why we would want to add another key-value store instead of making settings more efficient. After all settings is a key-value store meaning that it has string keys and values, so the only thing that's missing is a little bit more focus on the dynamic save/load use case. Do you think that Zephyr doesn't need to have an efficient storage for applications that are not limited to a known and relatively small subset of keys? Or that store bigger data in settings and hence want to load them on demand to save RAM?

Laczen, 2022-05-16:
@Damian-Nordic, sure it would be possible to try and improve this. At the moment the settings backend is very simple and straightforward. For saving a key and value is passed and settings does not care how this is saved. For loading the backend goes over all stored key/values and returns them. This last step is what makes it slow. The backend does not do any comparison on keys whatsoever (except when trying to minimize storage when something is saved). Supporting the dynamic load use case would push this complexity down to the backend and will result in incompatibility with existing solutions.

Maybe the simplest solution would be to extend the settings system with a `settings_load_key()` routine that passes the key (not a subtree, as this would result in having to go over all entries again) to the backend. Any backend that does not support this (does not have this routine) could fallback to the existing `settings_load_subtree()` with the key as subtree.

As a second step to support hashes a new backend ("nvshash") could be defined. In this a hash table of e.g. 128 values could point to id's 0x800a

Laczen, 2022-05-16:
@Damian-Nordic, sorry for the close (hit the wrong button)...

nvlsianpu, 2022-05-16:
I think `settings_load_key()` API extension is good idea (obviously it gives proper place to put optimization/proper algorithm needed)

"nvshash" - Would it be something similar to addition of a has-table to current design?

Damian-Nordic, 2022-05-17:
I agree that `settings_load_key` (or `settings_load_one` for consistency with `settings_save_one`) would be a nice addition, but I made a couple of PoCs and it looks like a cache at the NVS level gives the biggest performance boost. By using a cache with 256 entries from NVS IDs to NVS ATEs I managed to improve the boot time from 4 seconds to 800ms. The same sized cache at the settings level (from keys to NVS IDs) gave around 1.5 seconds. I'll prepare a draft PR and will let you know. That way we wouldn't need to change any APIs and we would speed up bare NVS, too.

Laczen, 2022-05-17:
@Damian-Nordic, as a test could you also check what happens if you just double (or triple) the nvs sector size (just make sure there are at least 2 sectors).

Damian-Nordic, 2022-05-17:
Yes, I can but does it make a big difference? We will still need to traverse the entire NVS log, so the bigger sector size may only trigger the garbage collector a little bit earlier. Or am I missing something?

Laczen, 2022-05-17:
> Yes, I can but does it make a big difference? We will still need to traverse the entire NVS log, so the bigger sector size may only trigger the garbage collector a little bit earlier. Or am I missing something?
 
A bigger sector size also means that there are less ate's, so less travelling.

Damian-Nordic, 2022-05-18:
In a Matter example application I test we have 4 sectors * 4kB for NVS.

JPHutchins, 2024-10-07:
Is this closed by the addition of CONFIG_NVS_LOOKUP_CACHE, https://github.com/zephyrproject-rtos/zephyr/pull/45769?

Damian-Nordic, 2024-10-08:
@JPHutchins Yes, the lookup cache solved most performance issues for us and I have no plans for more improvements, so I'll close the issue.

closed 2024-10-08 by Damian-Nordic