zephyrproject-rtos/zephyr #45595: LE Audio: Add LC3 TX for Unicast Server Sample

Thalley, 2022-05-12:
**Is your enhancement proposal related to a problem? Please describe.**
The LE Audio unicast server sample had TX capabilities added by https://github.com/zephyrproject-rtos/zephyr/pull/42964, but without LC3 support. 

Besides LC3 support, we may want to refactor the code such that the code for sending over LC3 and the unencoded data is more separated (e.g. move the TX and RX handling into new files). 

**Describe the solution you'd like**
Add the same LC3 encoding from the unicast client sample to the unicast server sample. Similar update the unicast client sample to decode LC3 audio. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

kartben, 2023-08-20:
Fixed with #44225

Thalley, 2023-08-20:
> Fixed with #44225

Incorrect - That added support for RX for the Unicast Server, but not TX. Reopening

kartben, 2023-08-21:
Of course! Sorry -- should have realized that it didn't even make sense from an issue number perspective.