zephyrproject-rtos/zephyr #45604: modem APN can only be set at build time, but some use cases needs it to be configured runtime

szundi, 2022-05-12:
Modem APN can only be set at build time with CONFIG_MODEM_GSM_APN option. This is used by gsm_ppp.c:
```
static const struct setup_cmd setup_cmds[] = {
	/* no echo */
	SETUP_CMD_NOHANDLE("ATE0"),
	/* hang up */
	SETUP_CMD_NOHANDLE("ATH"),
	/* extender errors in numeric form */
	SETUP_CMD_NOHANDLE("AT+CMEE=1"),

	/* disable unsolicited network registration codes */
	SETUP_CMD_NOHANDLE("AT+CREG=0"),

	/* create PDP context */
	SETUP_CMD_NOHANDLE("AT+CGDCONT=1,\"IP\",\"" CONFIG_MODEM_GSM_APN "\""),
};
```

I would like to be able to set this at **runtime**, my use case requires APN configuration after clients gets the device and puts own SIM card in.

Tried to find a way I could insert my dynamic APN value into gsm_ppp.c's namespace, but this seems to be directly wired from CONFIG_MODEM_GSM_APN to a const array. I can not really think any way to change the APN value without changing this code.

slide333333, 2022-11-08:
This would be a very useful feature and should be actually pretty easy to implement since PPP can be started manually. We would just need some kind of public function like `gsm_ppp_set_apn` that needs to be called once for the device before `gsm_ppp_start` is run. Internally AT+CGDCONT can be run as a separate command instead of being called as a part of  `setup_cmds`. 

If anyone could confirm this would fit and could review and I can setup a PR?

szundi, 2022-11-08:
This sounds very interesing and would solve crucial problem for setting up devices by the end-user to their sim cards.