zephyrproject-rtos/zephyr #45632: ESP32   get error "undefined reference to `sprintf' "  while CONFIG_NEWLIB_LIBC=y

sam131208, 2022-05-13:
**Describe the bug**
Get a undeclared errors "undefined reference to `sprintf' " after enabling CONFIG_NEWLIB_LIBC.
Get a undeclared error "'CONFIG_ESP32_PHY_MAX_TX_POWER' undeclared" while enable CONFIG_BT.

**Logs and console output**
zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/lib/esp32/libphy.a(phy_chip_v7.o): in function `fe_reg_init':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:3382: undefined reference to `sprintf'

/home/sam/.local/zephyr-sdk-0.14.1/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/10.3.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /home/sam/zephyrproject/modules/hal/espressif/zephyr/esp32/../../components/esp_wifi/lib/esp32/libphy.a(phy_chip_v7.o): in function `get_phy_version_str':
/home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:3382: undefined reference to `sprintf'

sam131208, 2022-05-13:
While enable PWM, the file 'pwm_led_esp32.c':
    in function 'pwm_led_esp32_set_cycles' get error "redefine channel".

sylvioalves, 2022-05-13:
@sam131208 Any sample code I can test?

sam131208, 2022-05-14:
@sylvioalves You can test the 'hello_world' Project with under config, you will get error "undefined reference to `sprintf'",  and , if you comment the flash config, will get error "undefined symbol `printf'".

CONFIG_BT=y
CONFIG_BT_DEVICE_NAME="zephyr esp32"
CONFIG_BT_DEVICE_APPEARANCE=384

# FLASH
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y

CONFIG_NEWLIB_LIBC=y

sylvioalves, 2022-05-14:
@sam131208 I have already fixed those build errors. I still have not fixed an issue that prevents BT to start when NEWLIBC is enabled. WIP.

sam131208, 2022-05-15:
@sylvioalves  It looks like there are still some issues to work out。But, i can't test this problem until you update those build errors。

carlescufi, 2022-05-17:
@sam131208 @sylvioalves is this fixed by https://github.com/zephyrproject-rtos/zephyr/pull/45716 ?

sylvioalves, 2022-05-17:
@carlescufi there is a pending issue regarding ESP32 BT and newlibc. I'll PR this and tag this issue.

sam131208, 2022-05-20:
@sylvioalves  It seems to be working fine after updating to the new version df85671c5d0405c0747c2939c8dfe808b7e4cf38。

 Whether to arrange to modify the following error, the channel is repeatedly defined in the two functions.

static int pwm_led_esp32_set_cycles(const struct device *dev, uint32_t channel,
				    uint32_t period_cycles,
				    uint32_t pulse_cycles, pwm_flags_t flags)
{
	int speed_mode;
	int channel;
	int timer;
.....
}

static int pwm_led_esp32_get_cycles_per_sec(const struct device *dev,
					    uint32_t channel, uint64_t *cycles)
{
	const struct pwm_led_esp32_config *config;
	int channel;
	int timer;
	int speed_mode;
.....
}

PR #45833 (2022-05-24): west.yml: esp32: fix bluetooth adapter variable definition
When newlibc is enabled, memcpy_chk fails due to
overflow when testing destination address length.
This updates the source and destination range defintion
so that it works as expected.

This also fixes the issue related to undefined references when newlibc is enabled.

Closes #45755
Closes #45632