zephyrproject-rtos/zephyr #45658: Build failure: civetweb/http_server with target blackpill_f411ce and CONFIG_DEBUG=y

Coi-l, 2022-05-15:
**Describe the bug**
Building http_server for board blackpill_f411ce and CONFIG_DEBUG=y fails

**To Reproduce**
Steps to reproduce the behavior:
```
west build -p auto -b blackpill_f411ce zephyr/samples/net/civetweb/http_server/ -- -DCONFIG_DEBUG=y
```

**Expected behavior**
Build succeded

**Impact**
It's would be nice with debug output.

**Logs and console output**
```
/home/xxx/toolchains/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `pull_inner':
/home/xxx/Projects/zephyrproject/modules/lib/civetweb/src/civetweb.c:6426: undefined reference to `fileno'
/home/xxx/toolchains/zephyr-sdk-0.14.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `push_inner':
/home/xxx/Projects/zephyrproject/modules/lib/civetweb/src/civetweb.c:6267: undefined reference to `ferror'

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK,
 - Commit SHA or Version used:
   zephyr: 47850a301c419df5d01e2c5eef04ca05c48e33e1
 

**Additional context**
Seems related to this issue:
#36796
But the same fix does not seem to work (I think)

erwango, 2022-05-17:
@rlubos Would you mind having a look? Seems this issue is not STM32 specific but rather linked to sample configuration.

rlubos, 2022-06-27:
Civetweb is no longer maintained in Zephyr, see https://github.com/zephyrproject-rtos/zephyr/pull/46746 for more details.