zephyrproject-rtos/zephyr #45664: mqtt_publisher does not work in atsame54_xpro board

rlevano77, 2022-05-15:
**mqtt_publisher does not work in atsame54_xpro board**
After building the example using 
west build -b atsame54_xpro samples/net/mqtt_publisher

The board ethernet interface link gets up and I am able to ping the board's hardcoded ip address "192.168.1.71"
But the application fails when the mqtt clients try to connect to the mqtt broker.

Zephyr v.3.0
Zephyr_sdk version 0.14.1
Platform : atsame54_xpro board ( https://docs.zephyrproject.org/2.6.0/boards/arm/atsame54_xpro/doc/index.html )

Console log : 
uart:~$ *** Booting Zephyr OS build zephyr-v3.0.0-3319-g8566b00df3e1  ***
[00:00:00.255,000] <err> mdio_sam: transfer timedout MDIO
[00:00:00.510,000] <err> mdio_sam: transfer timedout MDIO
[00:00:00.765,000] <err> mdio_sam: transfer timedout MDIO
[00:00:01.020,000] <err> mdio_sam: transfer timedout MDIO
[00:00:01.020,000] <inf> eth_sam: MAC: 02:04:25:4c:b8:00
[00:00:01.020,000] <inf> eth_sam: Queue 0 activated
[00:00:01.026,000] <inf> net_config: Initializing network
[00:00:01.026,000] <inf> net_config: Waiting interface 1 (0x20000388) to be up...
[00:00:05.059,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
[00:00:05.059,000] <inf> eth_sam: Link up
[00:00:05.059,000] <inf> net_config: Interface 1 (0x20000388) coming up
[00:00:05.060,000] <inf> net_config: IPv4 address: 192.168.1.71
[00:00:05.060,000] <inf> net_mqtt_publisher_sample: attempting to connect:
[00:00:08.060,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:11.561,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:15.061,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:18.562,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:22.062,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:25.563,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:29.063,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:32.564,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:36.064,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:39.565,000] <inf> net_mqtt_publisher_sample: mqtt_connect: -116 <ERROR>
[00:00:40.065,000] <inf> net_mqtt_publisher_sample: try_to_connect: -22 <ERROR>
[00:00:45.065,000] <inf> net_mqtt_publisher_sample: attempting to connect:
uart:~$ uart:~$ *** Booting Zephyr OS build zephyr-v3.0.0-3319-g8566b00df3e1  **

carlescufi, 2022-05-18:
>But the application fails when the mqtt clients try to connect to the mqtt broker.

Which broker? Because Mosquito will not allow external connections by default.

rlevano77, 2022-05-23:
@carlescufi 
I am running mosquitto but I suspect I have wrong settings in my mosquitto.conf I will check.

In https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/mqtt_publisher/src/config.h#L24
```
define ZEPHYR_ADDR		"192.168.1.101"
#define SERVER_ADDR		"192.168.1.10"
```
If **SERVER_ADDR** is the IP address of the machine running the mosquitto broker?
Then what **ZEPHYR_ADDR**  is?

rlubos, 2022-05-23:
`ZEPHYR_ADDR` is the address of the device using Zephyr, but it's not really used by the sample, must've been some leftover from the past. `SERVER_ADDR` defaults to `CONFIG_NET_CONFIG_PEER_IPV4_ADDR` and that config should be set to the IP address of the host running mosquitto broker. 

As for the mosquitto configuration, it's well explained in https://mosquitto.org/documentation/authentication-methods/ what needs to be enabeld (in this particular case you’re interested in unauthenticated access).

rlevano77, 2022-05-24:
Wrong mosquitto configuration and `CONFIG_NET_CONFIG_PEER_IPV4_ADDR`  were the issue.
`mqtt_publisher ` example tested with no TLS worked ok.

Thanks @rlubos , @carlescufi

closed 2022-05-24 by rlevano77