zephyrproject-rtos/zephyr #45666: Building samples about BLE audio with nrf5340dk does not work

CXsCode, 2022-05-16:
**Describe the bug**
I built zephyr/samples/broadcast_audio_source and zephyr/samples/unicast_audio_server for nrf5340dk_nrf5340_cpuapp, but the demo prompts an error when running, I think it has something to do with the configuration of zephyr/samples/hci_rpmsg built for nrf5340dk_nrf5340_cpunet.I see in zephyr/samples/hci_rpmsg that in addition to prj.conf there is also nrf5340_cpunet_iso_broadcast.conf and nrf5340_cpunet_iso_receive.conf, but I can't get the sample to work no matter which file I use.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf5340dk_nrf5340_cpuapp -d 5340_cpuapp -p auto .\samples\bluetooth\broadcast_audio_source\
2. west flash -d 5340_cpuapp
3. west build -p auto -b nrf5340dk_nrf5340_cpunet -d 5340_cpunet -- -DCONF_FILE="E:/zephyr/zephyrproject/zephyr/samples/bluetooth/hci_rpmsg/nrf5340_cpunet_iso_broadcast.conf" .\samples\bluetooth\hci_rpmsg\
4. west flash -d 5340_cpunet
or use default configuration for hci_rpmsg 
3. west build -p auto -b nrf5340dk_nrf5340_cpunet -d 5340_cpunet .\samples\bluetooth\hci_rpmsg\
4. west flash -d 5340_cpunet

And:
1. west build -b nrf5340dk_nrf5340_cpuapp -d 5340_cpuapp -p auto .\samples\bluetooth\unicast_audio_server\
2. west flash -d 5340_cpuapp
3. west build -p auto -b nrf5340dk_nrf5340_cpunet -d 5340_cpunet -- -DCONF_FILE="E:/zephyr/zephyrproject/zephyr/samples/bluetooth/hci_rpmsg/nrf5340_cpunet_iso_broadcast.conf" .\samples\bluetooth\hci_rpmsg\
4. west flash -d 5340_cpunet
or same as above

**Expected behavior**
Samples running successfully

**Logs and console output**
broadcast_audio_source:(use nrf5340_cpunet_iso_broadcast.conf)
*** Booting Zephyr OS build zephyr-v3.0.0-3960-g782480313edc  ***
Bluetooth initialized
Creating broadcast source
Unable to create broadcast source: -5
[00:00:00.282,165] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.282,196] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
[00:00:00.282,226] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.0 Build 99
[00:00:00.283,874] <inf> bt_hci_core: Identity: FD:57:DF:FB:69:05 (random)
[00:00:00.283,905] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
[00:00:00.283,935] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0xffff
[00:00:00.292,724] <wrn> bt_hci_core: opcode 0x2037 status 0x07

unicast_audio_server:(use nrf5340_cpunet_iso_broadcast.conf)
*** Booting Zephyr OS build zephyr-v3.0.0-3960-g782480313edc  ***
Bluetooth initialized
Failed to set advertising data (err -5)
[00:00:00.279,632] <wrn> bt_hci_core: Controller to host flow control not supported
[00:00:00.282,012] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.282,043] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
[00:00:00.282,043] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.0 Build 99
[00:00:00.284,057] <inf> bt_hci_core: Identity: FD:57:DF:FB:69:05 (random)
[00:00:00.284,088] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
[00:00:00.284,088] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0xffff
[00:00:00.287,811] <wrn> bt_hci_core: opcode 0x2037 status 0x07
[00:00:00.296,783] <wrn> bt_pacs: PACS notify failed: -2

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: arm-none-eabi
 - Commit: 3.0.99

**Additional context**
I want to know the configuration of the nrf5340dk_nrf5340_cpunet for each BLE audio sample,I can't find any relevant descriptions

cvinayak, 2022-05-18:
@CXsCode Please add `CONFIG_BT_CTLR_ADV_DATA_LEN_MAX=191` in the  nrf5340_cpunet_iso_broadcast.conf file.
Relates to #45015

CXsCode, 2022-05-19:
> @CXsCode Please add `CONFIG_BT_CTLR_ADV_DATA_LEN_MAX=191` in the nrf5340_cpunet_iso_broadcast.conf file. Relates to #45015

It's useful.But why was it not added to the nrf5340_cpunet_iso_broadcast.conf in 45015, but only to the samples/bluetooth/broadcast_audio_source/boards/nrf52840dk_nrf52840.conf?
Although the samples running,but the features of this sample does not seem to be realized.
I have two other questions:
1.Does the samples of LE Audio support nrf52840dk?
I built a sample(broadcast_audio_sink) for nrf52840dk, but it shows that the broadcast source cannot be searched.
Log:
------------------------------------------------------------------------------------------------------------------------------
*** Booting Zephyr OS build v3.1.0-rc1-56-g4c504d3f5adf  ***
Bluetooth initialized
Scanning for broadcast sources
[00:00:00.257,781] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.257,781] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.257,812] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.1 Build 0
[00:00:00.258,422] <inf> bt_hci_core: Identity: D0:D3:07:CE:DF:F5 (random)
[00:00:00.258,453] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
[00:00:00.258,483] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0xffff
sem_broadcaster_found timed out, resetting
Scanning for broadcast sources
Unable to start scan for broadcast sources: -120
------------------------------------------------------------------------------------------------------------------------------

2.I built a sample(broadcast_audio_source) for nrf5340dk, after running for a period of time, start printing some errors,and printing stopped after a while.
Log:
------------------------------------------------------------------------------------------------------------------------------
*** Booting Zephyr OS build v3.1.0-rc1-56-g4c504d3f5adf  ***
Bluetooth initialized
Creating broadcast source
Starting broadcast source
[00:00:00.282,623] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.282,653] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
[00:00:00.282,684] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.1 Build 0
[00:00:00.284,332] <inf> bt_hci_core: Identity: FD:57:DF:FB:69:05 (random)
[00:00:00.284,362] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
[00:00:00.284,362] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0xffff
Broadcast source started
Waiting 30 seconds before stopped
Sent 1000 total ISO packets
Sent 2000 total ISO packets
Stopping broadcast source
Broadcast source stopped
Deleting broadcast source
Broadcast source deleted
Creating broadcast source
Starting broadcast source
[00:00:31.589,813] <err> bt_hci_core: packets count mismatch
[00:00:31.589,874] <err> bt_hci_core: packets count mismatch
Broadcast source started
Waiting 30 seconds before stopped
[00:00:32.820,190] <err> bt_hci_core: No connection for handle 0
[00:00:32.820,281] <err> bt_hci_core: No connection for handle 0
Sent 3000 total ISO packets
Sent 4000 total ISO packets
Sent 5000 total ISO packets
Stopping broadcast source
Broadcast source stopped
Deleting broadcast source
Broadcast source deleted
Creating broadcast source
Starting broadcast source
[00:01:02.901,550] <err> bt_hci_core: packets count mismatch
Broadcast source started
Waiting 30 seconds before stopped
[00:01:04.131,896] <err> bt_hci_core: No connection for handle 0
[00:01:04.131,958] <err> bt_hci_core: No connection for handle 0
Sent 6000 total ISO packets
Sent 7000 total ISO packets
Stopping broadcast source
Broadcast source stopped
[... 73 lines cut ...]
Creating broadcast source
Starting broadcast source
[00:04:10.811,553] <err> bt_hci_core: packets count mismatch
[00:04:10.811,645] <err> bt_hci_core: packets count mismatch
Broadcast source started
Unable to broadcast data on 0x20001250: -105
Waiting 30 seconds before stopped
[00:04:12.041,931] <err> bt_hci_core: No connection for handle 0
[00:04:12.042,022] <err> bt_hci_core: No connection for handle 0
[00:04:12.061,950] <err> bt_conn: Unable to allocate TX context
Sent 21000 total ISO packets
Sent 22000 total ISO packets
Stopping broadcast source
Broadcast source stopped
Deleting broadcast source
Broadcast source deleted
Creating broadcast source
Starting broadcast source
[00:04:42.133,239] <err> bt_hci_core: packets count mismatch
[00:04:42.133,300] <err> bt_hci_core: packets count mismatch
Broadcast source started
[00:04:43.363,677] <err> bt_hci_core: No connection for handle 0
[00:04:43.363,769] <err> bt_hci_core: No connection for handle 0

------------------------------------------------------------------------------------------------------------------------------

Thalley, 2022-05-19:
> ## It's useful.But why was it not added to the nrf5340_cpunet_iso_broadcast.conf in 45015, but only to the samples/bluetooth/broadcast_audio_source/boards/nrf52840dk_nrf52840.conf?

I think the dual-CPU setup may be confusing for this case. 
When you build the `cpunet` image, that is unrelated to the BT host samples (I think perhaps documentation can improved for this). 

For the cpunet you should build something like the `/samples/bluetooth/hci_rpmsg/` with `CONFIG_BT_CTLR_ADV_DATA_LEN_MAX=191`. 

For the nr5340 any of the `CONFIG_BT_CTLR_*` configs will not be present in the BT host samples due to the split architecture

Thalley, 2022-05-19:
So perhaps https://github.com/zephyrproject-rtos/zephyr/pull/45015 doesn't solve this issue completely, but may require some additional documentation for the nRF5340, as that isn't obvious.

Thalley, 2022-05-19:
> 1.Does the samples of LE Audio support nrf52840dk?
I built a sample(broadcast_audio_sink) for nrf52840dk, but it shows that the broadcast source cannot be searched.

The samples work with https://github.com/zephyrproject-rtos/zephyr/pull/45015

Thalley, 2022-05-19:
> 2.I built a sample(broadcast_audio_source) for nrf5340dk, after running for a period of time, start printing some errors,and printing stopped after a while.
Log:

I'll setup my board to advertise and see if I see the same. Could simply be that the connection between the board and PC broke (have you tried re-attaching it?).

CXsCode, 2022-05-19:
> I'll setup my board to advertise and see if I see the same. Could simply be that the connection between the board and PC broke (have you tried re-attaching it?).

I don't think this caused the problem, but I'll try it again.

CXsCode, 2022-05-19:
> > 1.Does the samples of LE Audio support nrf52840dk?
> > I built a sample(broadcast_audio_sink) for nrf52840dk, but it shows that the broadcast source cannot be searched.
> 
> The samples work with #45015

I have pulled the latest submission

Thalley, 2022-05-19:
@cvinayak using https://github.com/zephyrproject-rtos/zephyr/pull/45015 and compiling hci_rpmsg on the nRF5340 cpunet with
```
CONFIG_BT_EXT_ADV=y
CONFIG_BT_CTLR_ADV_ISO=y
CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE=155
CONFIG_BT_CTLR_ADV_ISO_PDU_LEN_MAX=155
CONFIG_BT_CTLR_ADV_DATA_LEN_MAX=191
```

never sends any data, and with https://github.com/zephyrproject-rtos/zephyr/pull/45015 on the nRF52840DK I actually now get
```
*** Booting Zephyr OS build v3.1.0-rc1-26-g3b6d876bedf8  ***
Bluetooth initialized
Creating broadcast source
Starting broadcast source
[00:00:00.262,390] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.262,420] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.262,451] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.1 Bu0
[00:00:00.263,061] <inf> bt_hci_core: Identity: C4:E7:C8:4B:98:F5 (random)
[00:00:00.263,061] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
[00:00:00.263,092] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0xffff
Broadcast source started
Waiting 30 seconds before stopped
Sent 1000 total ISO packets
Sent 2000 total ISO packets
Stopping broadcast source
Broadcast source stopped
Deleting broadcast source
Broadcast source deleted
Creating broadcast source
Starting broadcast source
[00:00:31.564,208] <err> bt_hci_core: packets count mismatch
[00:00:31.564,483] <err> bt_hci_core: packets count mismatch
Broadcast source started
Waiting 30 seconds before stopped
[00:00:32.789,611] <err> bt_hci_core: No connection for handle 0
[00:00:32.789,703] <err> bt_hci_core: No connection for handle 0
[00:00:32.840,454] <err> os: ***** BUS FAULT *****
[00:00:32.840,484] <err> os:   Precise data bus error
[00:00:32.840,484] <err> os:   BFAR Address: 0x6d8cd0
[00:00:32.840,515] <err> os: r0/a1:  0x006d8ccc  r1/a2:  0x20006ccc  r2/a3:  0x20009f2c
[00:00:32.840,515] <err> os: r3/a4:  0x20006704 r12/ip:  0x000005e3 r14/lr:  0x0000e643
[00:00:32.840,545] <err> os:  xpsr:  0x81000028
[00:00:32.840,545] <err> os: Faulting instruction address (r15/pc): 0x000144a6
[00:00:32.840,576] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:32.840,576] <err> os: Fault during interrupt handling

[00:00:32.840,606] <err> os: Current thread: 0x20005b30 (unknown)
[00:00:33.405,090] <err> os: Halting system
```

consistently, so it does look like something broke somewhere.

cvinayak, 2022-05-19:
@Thalley  what does `arm-none-eabi-addr2line -e zephyr/zephyr.elf 0x000144a6` give the file and line number as?

Thalley, 2022-05-19:
> @Thalley what does `arm-none-eabi-addr2line -e zephyr/zephyr.elf 0x000144a6` give the file and line number as?

`zephyr/lib/libc/minimal/source/string/string.c:337`
It's a `memcpy` failing somewhere.

cvinayak, 2022-08-23:
@Thalley is this still a valid issue and reproducible on latest upstream/main?

Thalley, 2022-08-29:
Will need to check once I get the time :)

Thalley, 2022-08-29:
I cannot reproduce this at the moment, but that might be likely due the host not getting any num_complete_packets back from the controller, thus stalling the TX'ing:

```c
*** Booting Zephyr OS build zephyr-v3.1.0-4333-g4e039fac03ca  ***
Bluetooth initialized
Creating broadcast source
Starting broadcast source
Broadcast source started
sent_cnt: 0, stopping: 0
sent_cnt: 1, stopping: 0
Waiting 30 seconds before stopped
[00:00:00.467,895] <inf> bt_hci_core: Identity: FE:E2:2C:9F:F9:F6 (random)
[00:00:00.467,926] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x0000, manufacturer 0x05f1
[00:00:00.467,956] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0xffff
[00:00:00.472,290] <dbg> bt_audio_stream: bt_audio_stream_attach: conn (nil) stream 0x2000f8c4 ep 0x2000fa04 codec 0x20000104
[00:00:00.482,208] <dbg> bt_iso: bt_iso_chan_add: iso 0x2000f620 chan 0x2000fae0
[00:00:00.482,818] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x2000fae0 iso 0x2000f620 disconnected -> connecting
[00:00:00.502,410] <dbg> bt_iso: hci_le_big_complete: BIG[0] 0x2000f9b0 completed, status 0
[00:00:00.502,441] <dbg> bt_iso: bt_iso_connected: 0x2000f620
[00:00:00.502,868] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x2000fae0 iso 0x2000f620 connecting -> connected
Stopping broadcast source
Broadcast source stopped
Deleting broadcast source
Broadcast source deleted
Creating broadcast source
Starting broadcast source
Broadcast source started
sent_cnt: 2, stopping: 0
[00:00:30.515,441] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x2000fae0 iso 0x2000f620 connected -> disconnecting
[00:00:30.593,444] <dbg> bt_iso: hci_le_big_terminate: BIG[0] 0x2000f9b0 terminated
[00:00:30.593,444] <dbg> bt_iso: bt_iso_disconnected: 0x2000f620
[00:00:30.593,475] <dbg> bt_iso: bt_iso_chan_disconnected: 0x2000fae0, reason 0x16
[00:00:30.593,475] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x2000fae0 iso 0x2000f620 disconnecting -> disconnected
[00:00:30.603,820] <dbg> bt_audio_stream: bt_audio_stream_attach: conn (nil) stream 0x2000f8c4 ep 0x2000fa04 codec 0x20000104
[00:00:30.613,739] <dbg> bt_iso: bt_iso_chan_add: iso 0x2000f620 chan 0x2000fae0
[00:00:30.614,318] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x2000fae0 iso 0x2000f620 disconnected -> connecting
[00:00:30.633,911] <dbg> bt_iso: hci_le_big_complete: BIG[0] 0x2000f9b0 completed, status 0
[00:00:30.633,941] <dbg> bt_iso: bt_iso_connected: 0x2000f620
[00:00:30.634,368] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x2000fae0 iso 0x2000f620 connecting -> connected
```

cvinayak, 2022-08-30:
Thanks @Thalley, I will follow up as part of https://github.com/zephyrproject-rtos/zephyr/pull/45989 rework.

closed 2022-09-06 by carlescufi