zephyrproject-rtos/zephyr #45675: testing.ztest.customized_output: mismatch twister results in json/xml file

LixinGuoX, 2022-05-16:
**Describe the bug**
When running tests/ztest/custom_output/ with twister on multiple hardware platforms, we can get testing.ztest.customized_output passed as below:
```
Running TESTSUITE framework_tests
1: Test [test_assert_pass] reported (PASS) no. 1
2: Test [test_assert_pass] reported (PASS) no. 2
3: Test [test_assert_pass] reported (PASS) no. 3
4: Test [test_assert_skip] reported {SKIPPED} no. 1
5: Test [test_assert_skip] reported {SKIPPED} no. 2
TESTSUITE framework_tests succeeded
RunID: 123511537b52fde13c95f439ebab65be
PROJECT EXECUTION SUCCESSFUL
```
But in twister.json or twister_report.xml, we got different results, both testcases in testing.ztest.customized_output showed a status of `blocked`. Also there has an extra `identifier` called `testing.ztest.customized_output` with a status of `passed` which is misleading.
```
{
    "environment":{
        "os":"posix",
        "zephyr_version":"zephyr-v3.0.0-3960-g782480313edc",
        "toolchain":"zephyr"
    },
    "testsuites":[
        {
            "name":"tests/ztest/custom_output/testing.ztest.regular_output",
            "arch":"arm",
            "platform":"reel_board",
            "run_id":"e508c6df6f07caaca6f69bd22461f0bb",
            "runnable":true,
            "status":"passed",
            "execution_time":"0.00",
            "testcases":[
                {
                    "identifier":"testing.ztest.regular_output.assert_pass",
                    "execution_time":"0.10",
                    "status":"passed"
                },
                {
                    "identifier":"testing.ztest.regular_output.assert_skip",
                    "execution_time":"0.10",
                    "status":"skipped",
                    "reason":"ztest skip"
                }
            ]
        },
        {
            "name":"tests/ztest/custom_output/testing.ztest.customized_output",
            "arch":"arm",
            "platform":"reel_board",
            "run_id":"d3029f0f7c8c7877e7b2c1a9af5b237a",
            "runnable":true,
            "status":"passed",
            "execution_time":"0.00",
            "testcases":[
                {
                    "identifier":"testing.ztest.customized_output.assert_pass",
                    "execution_time":"0.00",
                    "status":"blocked",
                    "reason":"Unknown"
                },
                {
                    "identifier":"testing.ztest.customized_output.assert_skip",
                    "execution_time":"0.00",
                    "status":"blocked",
                    "reason":"Unknown"
                },
                {
                    "identifier":"testing.ztest.customized_output",
                    "execution_time":"0.00",
                    "status":"passed"
                }
            ]
        }
    ]
}
```

**To Reproduce**
Steps to reproduce the behavior:
Using reel_board as an example, we can reproduce on frdm_k64f, Intel_CAVS... as well.
1. twister -p reel_board --device-testing --device-serial /dev/ttyACM0 -T tests/ztest/custom_output/ -vv
2. cd twister-out/
3. see twister.json or twister_report.xml

**Expected behavior**
We should have the same result in twister.log and twister.json(xml).

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain 0.14.1
 - Commit d78194546746

nashif, 2022-05-17:
This is an issue with the test, it does not report results that twister can parse (customized output). Like many broken tests in the tree, twister is exposing this so it can be fixed.

smrtos, 2022-05-31:
Should be fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/46052, https://github.com/zephyrproject-rtos/zephyr/pull/46132

LixinGuoX, 2022-05-31:
> Should be fixed by: #46052, #46132

With those patches, twister still can not parse that output, we don't have to change twister to fit this case, just customize the output a little bit more since it has been customized.

PR #46119 (2022-06-05): tests: cumstom_output: change output to fit twister
Since output can be cumstomized, so slightly change the output format,
so that twister can parse the output and collect the passrate correctly.

Fixes #45675
closed 2022-06-05 by carlescufi