zephyrproject-rtos/zephyr #45678: Devnonce

Andermutu, 2022-05-16:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

fabiobaltieri, 2022-05-17:
@Andermutu do you have the device LoRaWAN version configured correctly on the TTN console? I think that the current stack should use "LoRaWAN Specification 1.0.4", I have few devices set up with that with no explicit devnonce handling and they work fine. You may also want to look into either storing and setting a devnonce from the application either using https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/lorawan/lorawan.h#L94

or the NVM module 
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/lorawan/nvm/lorawan_nvm_settings.c

(though they may not be available in the current nRF Connect SDK Zephyr version).

Andermutu, 2022-05-18:
@fabiobaltieri @Mani-Sadhasivam, you are right, the problem is that the `class_a` sample must provide a unique `dev_nonce` to work with "LoRaWAN Specification 1.0.4". I have first tried not to use explicit devnonce handling. On TTN, my device was configured as 1.0.3. After changing to 1.0.4 I was getting "devnonce is too small" error. The solution was to provide a unique devnonce, that should be stored on the nvs.

The [9cc32af](https://github.com/zephyrproject-rtos/zephyr/commit/9cc32af99e7986e32e1fe82f3c8e5d30e1fc3997) changes worked well to provide that unique devnonce, after modifying `lorawan.h` and `lorawan.c` files.

ballardr, 2022-05-19:
I've had this same issue. I've added the below function which I call just before lorawan_join(). It's a bit hacky, but it works. For me it just means I don't need to muck round too much with the upstream libraries.

```
#include <random/rand32.h>
#include <LoRaMac.h>
int lorawan_setRandomDevNonce () {

	MibRequestConfirm_t mibReq;
	mibReq.Type = MIB_NVM_CTXS;
	LoRaMacMibGetRequestConfirm( &mibReq );
	union {
		uint16_t u16[2];
		uint32_t u32;
		uint8_t u8[4];
	} num;

	int ret = sys_csrand_get(num.u8,2);
	if (ret < 0) {
		num.u32 = sys_rand32_get();
	}
    LoRaMacNvmData_t* MacContexts = mibReq.Param.Contexts;

    MacContexts->Crypto.DevNonce = num.u16[0];

	return 0;
}
```

Note, to get #include <LoRaMac.h> to work in my application code, I had to add the below to cmake so to include the loramac-node directories as search paths for headers.
zephyr_library_include_directories($ENV{ZEPHYR_BASE}/../modules/lib/loramac-node/src/mac)
zephyr_library_include_directories($ENV{ZEPHYR_BASE}/../modules/lib/loramac-node/src/mac/region)
zephyr_library_include_directories($ENV{ZEPHYR_BASE}/../modules/lib/loramac-node/src/boards)
zephyr_library_include_directories($ENV{ZEPHYR_BASE}/../modules/lib/loramac-node/src/system)
zephyr_library_include_directories($ENV{ZEPHYR_BASE}/../modules/lib/loramac-node/src/radio)

Andermutu, 2022-05-19:
@ballardr Thank you for sharing your solution! I have been questioning whether using a random devnonce is a good solution in terms of reliability, and finally decided to make a counter that saves on flash memory. For people like me who have limited knowledge of how zephyr builds/versions work, it's tricky to figure out the problem without asking for help.

ballardr, 2022-05-19:
So LoRaWAN Specifications 1.0.3 (and similar for older versions) on line 974 specifically says "DevNonce is a random value".

However specifications 1.0.4 and 1.1 both say "DevNonce is a counter starting at 0 when the device is initially powered up and incremented with every Join-request. A DevNonce value SHALL NEVER be reused for a given JoinEUI value."

So a random devnonce is a good solution depending what version of LoRaWAN your targeting. In reality, it more so depends how strict the network server is (Chirpstack, TTN etc) as I don't think for instance that chirpstack actually cares if its random or incremented.

Mani-Sadhasivam, 2022-06-18:
Closing this issue since we already have a solution in mainline. Please reopen if you face any further issue, thanks!

closed 2022-06-18 by Mani-Sadhasivam

DivyaaMohann, 2023-11-03:
Hi, 
I am facing the same issue in LoRaWAN specification 1.0.4, DevNonce is too small. How do I manually rectify this in the code any source of reference would greatly help. I would also like to know the dependencies that the Zephyr prj.conf file requires to run the LoRaWAN sample in Zephyr.
Thank you.