zephyrproject-rtos/zephyr #45697: RING_BUF_DECLARE broken for C++

ottojo, 2022-05-16:
The `RING_BUF_DECLARE` macro (https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/ring_buffer.h#L76-L83) initializes the `ring_buf` (defined [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/ring_buffer.h#L34-L43)) by listing `size`, then `buffer`. This order was changed in #42845 [(here)](https://github.com/zephyrproject-rtos/zephyr/pull/42845/files#diff-caddaf4f58cb580dc17aa673b2347995552e02b5a438d9df68c9dc00c0684ca9L34-L58). C++ requires the order to match, resulting in compilation failure when using `RING_BUF_DECLARE`, which was possible before.

**Environment**
 - OS: Linux
 - Current master branch of zephyr, SDK 0.13.1
 - ARM GCC
 - CONFIG_STD_CPP17=y

stephanosio, 2022-05-16:
cc @npitre

ottojo, 2022-05-16:
@stephanosio just did, i was hesistant to change the ring_buf struct, fearing breaking something else, but i guess changing the macro would work even better!

closed 2022-05-18 by carlescufi

npitre, 2022-10-11:
Agreed. You may just adjust the macro with no impact on existing code.

PR #45698 (2022-05-18): fix ring_buf designated initializer order for RING_BUF_DECLARE
fixes #45697
commit a36995e2 (2022-05-18): ring_buffer: fix designated initializer order in RING_BUF_DECLARE

Change order of designated initializers for ring_buf struct in the
RING_BUF_DECLARE macro, to match the order in the struct definition.
This fixes compilation when using C++, which requires using the same
order as in the struct definition.
Fixes #45697