zephyrproject-rtos/zephyr #45702: Reboot instead of halting the system

OmarMorceli, 2022-05-16:
I have a Bluetooth Mesh application that is causing a halting system.

I would like to know if it is possible to reboot the MCU instead of stopping it ?

rodrigopex, 2022-05-19:
@OmarMorceli You could try using the [watchdog](https://docs.zephyrproject.org/latest/services/task_wdt/index.html?highlight=watchdog%20task) feature for that. Maybe this halt issue can see as a good thing. You'll notice the system is crashing at some point during the tests. If it reboots, you probably wouldn't know about the crash. I suggest you track the BLE Mesh bug down before accepting that and wait for seamless recovery by rebooting. Maybe the bug is at the mesh code.

OmarMorceli, 2022-05-23:
Thank you @rodrigopex
Regarding the [watchdog ](https://docs.zephyrproject.org/latest/services/task_wdt/index.html)feature, I have an application running on an nRF52840 Dongle that communicates with Bluetooth Mesh (using some vendor models) and sends/receives data over USB .

* How many channels should I use ?
* Where do i put the task_wdt_feed()  exactly ?
* And in what case enabling the [TASK_WDT_HW_FALLBACK ](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_TASK_WDT_HW_FALLBACK)is useful ?

rodrigopex, 2022-05-27:
@OmarMorceli sorry for the delay.  

> * How many channels should I use ?

You would add a channel for each thread you want to monitor. Take a look at the [sample](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.7.2/samples/subsys/task_wdt/src/main.c).

> * Where do i put the task_wdt_feed()  exactly ?

At a point of the code, you are sure will be reached before the timeout of the channel. 

> * And in what case enabling the [TASK_WDT_HW_FALLBACK ](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_TASK_WDT_HW_FALLBACK)is useful ?

The `task watchdog` is a regular thread, so it could get stuck as well. To avoid that you can reinforce the mechanism using the hardware watchdog. I should suggest using that. 

I hope it helps. Regards.

closed 2022-05-29 by OmarMorceli