zephyrproject-rtos/zephyr #45735: Eth W5500 Driver via SPI is deadlocking

lucasgcbkhomp, 2022-05-17:
**Describe the bug**
When utilizing the W5500 for Ethernet via SPI with [Zephyr's driver](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.9.1/zephyr/reference/devicetree/bindings/ethernet/wiznet,w5500.html?highlight=wiznet), it will eventually become unresponsive due to the module no longer triggering the GPIO callback, which in turn never gives the "int" semaphore back which prevents it from running its `w5500_isr` interrupt function.

I'm using:
- Nordic nRF52840
- nRF Connect SDK 1.9.1 (Zephyr 2.7.1)
- SEGGER Embedded Studio / nRF Connect for VSCode
- I've checked Discord, and at least one other person came across this
- I've tried dropping in the recent driver and it gave me the same result. Maybe I'm missing configuration and/or docs are missing additional info?
- I've attempted increasing timeout, thread priority, RX stack size to no avail
- I've noticed SPI API for Zephyr is very slow compared to baremetal from another application I've used

**To Reproduce**
Steps to reproduce the behavior:
0. Get a board with the W5500 Ethernet module driver
1. Configure DHCPv4 or static IP Address
2. Configure Ethernet with the W5500 module driver
3. Build and Flash the board with the IDE/West/Ninja commands of your choice
4. Ping the board in your local network 
5. Observe that the response time is roughly 1.8ms (very long for local)
6. Eventually (40 mins) the device's ethernet interface will lock up, making the address unreachable and isolating the device.

This lock up can be sped up by using more demanding examples, like MQTT Publisher.

**Expected behavior**
Ethernet interface W5500 to run indefinitely after being set up.
If the module locks up, it should probably detect the lock and reset just the module with an error log.

**Impact**
This is a showstopper because the board must be reboot manually for the application to work upon detecting this fault, which is not possible to work with.

**Logs and console output**
Only logs I have are custom traces that lead to showing the `int_sem` no longer is given up, locking up the driver with no countermeasure.

The following no longer calls back after some time:
https://github.com/zephyrproject-rtos/zephyr/blob/734c3454578d4db89b7a879b7339559b20c04c44/drivers/ethernet/eth_w5500.c#L492-L493
Which leads to the following never calling:
https://github.com/zephyrproject-rtos/zephyr/blob/734c3454578d4db89b7a879b7339559b20c04c44/drivers/ethernet/eth_w5500.c#L423-L431
Which locks here:
https://github.com/zephyrproject-rtos/zephyr/blob/734c3454578d4db89b7a879b7339559b20c04c44/drivers/ethernet/eth_w5500.c#L283-L292
**Environment (please complete the following information):**
 - Linux Ubuntu Focal Fossa flashing a nRF52840 with JLink
 - gcc-arm-none-eabi-9-2019-q4-major with nRF Connect SDK 1.9.1
 - Nordic ncs [2.7.99](https://github.com/nrfconnect/sdk-zephyr/tree/v2.7.99-ncs1-branch)

**Additional context**
My board is custom so the SPI is set up as follows:
```js
&spi0 {
	compatible = "nordic,nrf-spim";
	sck-pin = <32>;
	miso-pin = <25>;
	mosi-pin = <24>;
	cs-gpios = <&gpio0 22 GPIO_ACTIVE_LOW>;
	status = "okay";
	w5500: w5500@0 {
		compatible = "wiznet,w5500";
		label = "w5500";
		reg = <0>;
		spi-max-frequency = <8000000>;
		int-gpios = <&gpio0 14 GPIO_ACTIVE_LOW>;
		reset-gpios = <&gpio0 13 GPIO_ACTIVE_LOW>;
	};
};
```

Any tips on how to fix this? I might make a PR myself.
Currently I don't understand why the GPIO stops, but I'm assuming it's because the SPI is too slow because after I increased the max frequency the problem took longer to occur (it's still too slow compared to bare metal though).

lucasgcbkhomp, 2022-05-17:
@parthitce Any ideas on what could be happening here with this W5500 deadlock issue? I'd be happy to help checking it out

parthitce, 2022-05-17:
@jancoow @lucasgcbkhomp Let me test this from my end and write back. Probably related to https://github.com/zephyrproject-rtos/zephyr/pull/42241 ?

lucasgcbkhomp, 2022-05-17:
> @jancoow @lucasgcbkhomp Let me test this from my end and write back. Probably related to #42241 ?

That seems to be the very same issue; with one possible fix being a way to handle a missed beat from the module (as it is being done in that PR). 

Currently if a single GPIO interrupt from the module is missed, the driver will deadlock waiting for the previously missed read and the ethernet interface will "brick" until reboot. 

With my SPI being slow, it (thankfully) made it more obvious. However, even in cases where it already works fine, high traffic/load or interference can still conjure this situation so technically field applications may just suddenly stop and people would be none the wiser.

I'm testing [elcritch](https://github.com/elcritch)'s fix, and currently it seems to be handling everything I can throw at it for TCP:

https://github.com/zephyrproject-rtos/zephyr/blob/871117699cd11139ef18ebad526903b5330cfb68/drivers/ethernet/eth_w5500.c#L283-L303

jancoow, 2022-05-18:
I've made a pull request with a proper solution.

honestech74, 2022-05-18:
I've tested #42241 with the simple https server and it seems like it doesn't handle the deadlock perfectly.
Here is ping log to the app after about 5 hours.
```
64 bytes from 192.168.101.246: icmp_seq=18226 ttl=64 time=2.80 ms
64 bytes from 192.168.101.246: icmp_seq=18227 ttl=64 time=3.09 ms
64 bytes from 192.168.101.246: icmp_seq=18228 ttl=64 time=3.14 ms
64 bytes from 192.168.101.246: icmp_seq=18229 ttl=64 time=3.14 ms
64 bytes from 192.168.101.246: icmp_seq=18230 ttl=64 time=2.59 ms
From 192.168.101.244 icmp_seq=18238 Destination Host Unreachable
From 192.168.101.244 icmp_seq=18239 Destination Host Unreachable
From 192.168.101.244 icmp_seq=18240 Destination Host Unreachable
```

jancoow, 2022-05-18:
> I've tested #42241 with the simple https server and it seems like it doesn't handle the deadlock perfectly.
> Here is ping log to the app after about 5 hours.
> ```
> 64 bytes from 192.168.101.246: [...]

Can you test this fix? : https://github.com/zephyrproject-rtos/zephyr/pull/45765

honestech74, 2022-05-19:
> Can you test this fix? : https://github.com/zephyrproject-rtos/zephyr/pull/45765

@jancoow I am testing it and it's working for about 8 hours so far. I will update if there is any issue.

honestech74, 2022-05-20:
@jancoow I think that #45765 fixes this issue and tested it for about 32 hours.
@lucasgcbkhomp I checked #45765 is the right solution and you can close this.

closed 2022-05-25 by carlescufi