zephyrproject-rtos/zephyr #45750: tests-ci : kernel: timer: tickless test Failed

hakehuang, 2022-05-18:
**Describe the bug**
    tickless test is Failed on v3.1.0-rc1 on mimxrt1170_evk_cm7

    see logs for details

    **To Reproduce**
    1. 
    ```
    scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1170_evk_cm7      --sub-test kernel.timer
    ```
    2. See error

    **Expected behavior**
    test pass

    **Impact**

    **Logs and console output**
    ```
    START - test_sleep_abs

Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:809: test_sleep_abs: (late >= 0 && late < k_us_to_ticks_ceil32(250) is false)
expected wakeup at 14481, got 14483 (late 2)
FAIL - test_sleep_abs in 0.32 seconds

        
    ```

    **Environment (please complete the following information):**
     - OS: (e.g. Linux )
     - Toolchain (e.g Zephyr SDK)
     - Commit SHA or Version used: v3.1.0-rc1

hakehuang, 2022-05-18:
@dleach02 issue found on v3.1.0-rc1

mmahadevan108, 2022-05-18:
Unable to reproduce the failure on tag  `v3.1.0-rc1`. Below is the test result for test `tests\kernel\timer\timer_api`
```
*** Booting Zephyr OS build v3.1.0-rc1  ***
Running TESTSUITE timer_api
===================================================================
START - test_time_conversions
 PASS - test_time_conversions in 0.11 seconds
===================================================================
START - test_timer_duration_period
 PASS - test_timer_duration_period in 0.326 seconds
===================================================================
START - test_timer_restart
 PASS - test_timer_restart in 0.276 seconds
===================================================================
START - test_timer_period_0
 PASS - test_timer_period_0 in 0.202 seconds
===================================================================
START - test_timer_period_k_forever
 PASS - test_timer_period_k_forever in 0.202 seconds
===================================================================
START - test_timer_expirefn_null
 PASS - test_timer_expirefn_null in 0.326 seconds
===================================================================
START - test_timer_periodicity
 PASS - test_timer_periodicity in 0.252 seconds
===================================================================
START - test_timer_status_get
 PASS - test_timer_status_get in 0.1 seconds
===================================================================
START - test_timer_status_get_anytime
 PASS - test_timer_status_get_anytime in 0.326 seconds
===================================================================
START - test_timer_status_sync
 PASS - test_timer_status_sync in 0.352 seconds
===================================================================
START - test_timer_k_define
 PASS - test_timer_k_define in 0.702 seconds
===================================================================
START - test_timer_user_data
 PASS - test_timer_user_data in 0.301 seconds
===================================================================
START - test_timer_remaining
 PASS - test_timer_remaining in 0.51 seconds
===================================================================
START - test_timeout_abs
 PASS - test_timeout_abs in 0.2 seconds
===================================================================
START - test_sleep_abs
 PASS - test_sleep_abs in 0.13 seconds
===================================================================
TESTSUITE timer_api succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

hakehuang, 2022-05-19:
@mmahadevan108 , thanks for the check, just want to confirm do you build with below command line?
```
west build -b mimxrt1170_evk_cm7 -- -DCONF_FILE="prj_tickless.conf"
```

mmahadevan108, 2022-05-20:
Thank you. I am able to reproduce the failure.

mmahadevan108, 2022-05-20:
This is seen when when enable `CONFIG_PM`. MIMXRT1170 does not support Power Management yet. I have changed this to an enhancement.

danieldegrasse, 2022-05-23:
@mmahadevan108 PM should be supported on the RT1170, see #41171

PR #45918 (2022-05-25): dts: nxp: fix PM minimum residency and exit latency for rt11xx
Fix PM minimum residency time and exit latency for rt11xx to reflect
actual delays when entering and exiting low power modes.

Fixes #45750
commit 75513ba6 (2022-05-25): dts: nxp: fix PM minimum residency and exit latency for rt11xx

Fix PM minimum residency time and exit latency for rt11xx to reflect
actual delays when entering and exiting low power modes.

Fixes #45750
closed 2022-05-25 by carlescufi