zephyrproject-rtos/zephyr #45778: Unable to use thread aware debugging with STM32H743ZI

bdesterBE, 2022-05-18:
**Describe the bug**
When attempting to use the `CONFIG_THREAD_DEBUG_INFO` configuration option for thread aware debugging with the STM32H743ZI develpment board, after running `west debug` an output containing 'cant read "_TARGETNAME": no such variable', followed by "You can't do that when you target is `exec'" is received.

If the configuration option is not added to the project's prj.conf file, I am able to use `west debug` without issue.

I have attempted to add `$_TARGETNAME configure -rtos Zephyr` to zephyr/boards/arm/nucleo_h743zi/support/openocd.cfg per https://github.com/zephyrproject-rtos/zephyr/issues/37205, to no avail.

**To Reproduce**
The behavior can be seen on the blinky example.
Steps to reproduce the behavior:
1. add CONFIG_THREAD_DEBUG_INFO to zephyr/samples/basic/blinky/prj.conf
2. west build -p -b nucleo_h743zi zephyr/samples/basic/blinky
3. west debug

**Expected behavior**
The ability to run openocd debugging with thread awareness using `west debug` on the STM32H743ZI development board.

**Impact**
Currently have no other way of debugging with thread awareness in our system, making it difficult to see which thread could be causing various issue including watchdog timer triggers.

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Zephyr SDK 13.
 - Commit SHA: 140837ce3ac6ba6ac01d804ebd40c8aaf00a3011
 - Latest version: v3.0.0-rc1

erwango, 2022-05-19:
Correct flag name is `CONFIG_DEBUG_THREAD_INFO`

erwango, 2022-05-19:
Ok, here is the problem:

https://github.com/zephyrproject-rtos/zephyr/pull/37516 implemented integrated west runner support for openocd thread info configuration. For that it relied on the command `$_TARGETNAME configure -rtos Zephyr` which unfortunately is not generic enough to work with all openocd targets.
For instance, on stm32h743, it should use `$_CHIPNAME.cpu0 configure -rtos Zephyr`, which is getting the whole feature to work as expected.

We miss a way to overload this default generic command, either fully, either only overwriting the target handle (`_TARGETNAME` replaced by `_CHIPNAME.cpu0`). Note that this i required for all STM32H7 target, but likely other targets subject to multicore (for instance lpc4370 > `$_CHIPNAME.m4 `).

erwango, 2022-05-19:
Proposal to fix this is to add an openocd runner arg which would overwrite the default openocd script target handle `_TARGETNAME`. For instance:
`board_runner_args(openocd --target-script-handle "_CHIPNAME.cpu0")`

@mbolivar-nordic Would it be fine with you ?

mbolivar-nordic, 2022-05-19:
> @mbolivar-nordic Would it be fine with you ?

Yes

commit be217004 (2022-05-24): boards: stm32h7: Update openocd runner to specify target handle

Openocd scripts for STM32H7 SoCs use _CHIPNAME.cpu{0|1} as
target handle.
Specify this thanks to new openocd runner option '-target-handle'.

This is required to allow thread awareness debugging on these targets.

Fixes #45778
closed 2022-05-24 by mbolivar-nordic