zephyrproject-rtos/zephyr #45799: LED strip driver flips colors on stm32h7

endian-benjamin, 2022-05-19:
We're using the WS2812 RGB LED with the led_strip driver on two separate boards: one based on nrf52840 and one based on stm32h7. 

On the 840 everything works as expected. On the h7 however, it seems to flip the red and green channels! I.e. you ask it to shine `#ff0000` and it shines `#00ff00` instead. Note that `#0000ff` works the same on both SoCs.

My best guess currently is that the SPI controller operates in 16 bit mode and ends up byteswapping due to endianness... But this is kinda weird, since the `led_strip.c` at least TRIES to configure the controller in 8-bit mode:

```
#define SPI_FRAME_BITS 8

/*
 * SPI master configuration:
 *
 * - mode 0 (the default), 8 bit, MSB first (arbitrary), one-line SPI
 * - no shenanigans (don't hold CS, don't hold the device lock, this
 *   isn't an EEPROM)
 */
#define SPI_OPER (SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB | \
		  SPI_WORD_SET(SPI_FRAME_BITS))
```

This is easy to work around in software, but perhaps it's really low hanging fruit for someone more intimate with the STM32 HAL.

erwango, 2022-05-19:
@endian-benjamin Could it be linked to frame format (TI/Motorola) ? I know we had issue around that previoulsy with LED strip previoulsy. Would you mind having a try to switch that and see if fixes thing ?

See `frame-format` here: https://docs.zephyrproject.org/latest/build/dts/api/bindings/led_strip/worldsemi%2Cws2812-spi.html#dtbinding-worldsemi-ws2812-spi

Note that this doesn't explain NRF vs STM32 diff

endian-benjamin, 2022-05-20:
looks like a dead link, and I can't find a `frame-format` binding in latest main.

GeorgeCGV, 2022-05-20:
@endian-benjamin , the `frame-format` is a field of the `spi-device`, see https://github.com/zephyrproject-rtos/zephyr/blob/ce84c782dcd3cee6fad6ac7ec7642accdaa42d1f/dts/bindings/spi/spi-device.yaml#L33-L46.

endian-benjamin, 2022-05-20:
Does not help. I'll see if I can dig deeper into the SPI controller driver, I'm guessing the answer is there. It will take a while until I get som spare time tho, due to deadlines...

FRASTM, 2022-05-23:
I have check on my side the sample on the nucleo h743zi board
The spi1 requires the clock 
 The usual overlay and con

FRASTM, 2022-05-24:
When the SPI clock source is not explicetly declared by the DTS, 
the SPI clock source has the default value in the RCC register D2CCIP1R (0 --> pll1_q_clk) but the `clock_control_get_rate` does not know it and returns the value of the pll clock (96MHz) for the spi clock : which is not correct.
That's why the SPI clock source must be declared with
```
&spi1 {
	/delete-property/ clocks;
	clocks = <&rcc STM32_CLOCK_BUS_APB2 0x00001000>,
		 <&rcc STM32_SRC_PLL1_Q SPI123_SEL(0)>;
};
```
even if the default value is kept SPI123_SEL(0).

closed 2022-05-24 by mbolivar-nordic