zephyrproject-rtos/zephyr #45802: Some tests reported as PASSED (device) but they were only build

evgeniy-paltsev, 2022-05-19:
**Describe the bug**
Some tests reported at same time as PASSED (device) and as only build.

Here is log example of running 3 tests which were only build (according to this log message: `3 test configurations were only built.`) and passed, according to another log message:

```
- 3 test scenarios (3 configurations) selected, 0 configurations discarded due to filters.
- Adding tasks to the queue...
- Added initial list of jobs to queue
- 1/3 nsim_hs6x                 samples/hello_world/sample.basic.helloworld        PASSED (device)
- 2/3 nsim_hs6x                 samples/philosophers/sample.kernel.philosopher.stacks PASSED (device)
- 3/3 nsim_hs6x                 tests/kernel/common/kernel.common                  PASSED (device)

- 3 of 3 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 92.75 seconds
- In total 45 test cases were executed, 2 skipped on 1 out of total 456 platforms (0.22%)
- 0 test configurations executed on platforms, 3 test configurations were only built.
```

Twister options used:

```
./scripts/twister -v --ninja --test samples/hello_world/sample.basic.helloworld --test samples/philosophers/sample.kernel.philosopher.stacks --test tests/kernel/common/kernel.common --platform nsim_hs6x --west-runner=mdb-hw --device-testing --device-serial /dev/haps-serial --west-flash=--dig-device=JtagHs2 --pre-script /some-locationpre_script.sh --report-dir /some-location/out
```

Unfortunately this require HW board for reproducing.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK, 0.13.2
 - Commit SHA or Version used: 2dc866fa48

evgeniy-paltsev, 2022-05-19:
As I can see - tests are actually executed on device. However 
 * they are reported as a only built
 * there is no information about execution time - so it is marked as `PASSED (device)` instead of `PASSED (device 13.577s)`

mbolivar-nordic, 2022-05-24:
@evgeniy-paltsev did you bisect this one too?

hakehuang, 2022-05-26:
https://github.com/zephyrproject-rtos/zephyr/pull/45943 maybe this can fix your issue

mbolivar-nordic, 2022-05-26:
> #45943 maybe this can fix your issue

@evgeniy-paltsev if you can please confirm this and close this issue, the bug statistics would be grateful :)

smrtos, 2022-05-27:
Should be the same as https://github.com/zephyrproject-rtos/zephyr/issues/46033
Looking into it.

smrtos, 2022-05-30:
Should be fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/46104

smrtos, 2022-05-30:
https://github.com/zephyrproject-rtos/zephyr/pull/46104 merged. This one can be closed.

enjiamai, 2022-05-30:
Hi, @evgeniy-paltsev, Could you please help confirm if this issue got fixed or not?  @smrtos got this fixed.

evgeniy-paltsev, 2022-05-30:
Hi @enjiamai, I'll check this today or tomorrow, thanks!

rljordan, 2022-06-07:
@evgeniy-paltsev Has this been confirmed as fixed?

evgeniy-paltsev, 2022-06-07:
Yep, it's fixed.

I've just forgotten to close the issue, sorry for that :)

closed 2022-06-07 by evgeniy-paltsev