zephyrproject-rtos/zephyr #45807: CivetWeb doesn't build for CC3232SF

mkcol, 2022-05-19:
**Describe the bug**
The CivetWeb http server sample does not build for cc32325sf board. I am using the following commandline to build
```
$ west build -p auto -b cc3235sf_launchxl zephyr/samples/net/civetweb/http_server
```

I have added the below options to the prj.conf
```
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_WIFI=y
CONFIG_WIFI_SIMPLELINK=y
```

And the result is the below error
```
/home/dev/zephyr/samples/net/civetweb/http_server/../common/include/libc_extensions.h:40:5: error: expected identifier or '(' before 'int'
   40 | int ferror(FILE *stream);
      |     ^~~~~~
/home/dev/zephyr/samples/net/civetweb/http_server/../common/include/libc_extensions.h:40:5: error: expected ')' before '(' token
   40 | int ferror(FILE *stream);
      |     ^~~~~~
[6/209] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc32_sw.c.obj
ninja: build stopped: subcommand failed.
```

Can anybody help point me in the right direction to get this sample running on the cc3235sf

rlubos, 2022-05-20:
`WIFI_SIMPLELINK` requires full libc (Newlib by default), and this configuration is not compatible with with `libc_extensions.h` used by civetweb, as it seems to be written for minimal libc. You can remove or ifdef conflicting declarations in there, but this leads to another issue.

It seems that pthreads, used by civetweb, are also not compatible with Newlib:
```
In file included from /home/robert/repos/zephyrproject/zephyr/include/zephyr/posix/time.h:61,
                 from /home/robert/repos/zephyrproject/zephyr/include/zephyr/posix/pthread.h:12,
                 from /home/robert/repos/zephyrproject/modules/lib/civetweb/src/civetweb.c:170:
/home/robert/repos/zephyrproject/zephyr/include/zephyr/posix/posix_types.h:45:3: error: conflicting types for 'pthread_attr_t'
   45 | } pthread_attr_t;
      |   ^~~~~~~~~~~~~~
In file included from /home/robert/zephyr-sdk/zephyr-sdk-0.13.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:223,
                 from /home/robert/zephyr-sdk/zephyr-sdk-0.13.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/time.h:28,
                 from /home/robert/repos/zephyrproject/modules/lib/civetweb/src/civetweb.c:166:
/home/robert/zephyr-sdk/zephyr-sdk-0.13.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:75:3: note: previous declaration of 'pthread_attr_t' was here
   75 | } pthread_attr_t;
      |   ^~~~~~~~~~~~~~

```
@carlescufi Who's in charge of this `pthread` library for Zephyr?

gnfranco, 2022-05-21:
I'm with similar problem here.
But I can't build civetweb to any of my board. Even e when I try to build for "atmel-explained" it not build successful.

gnfranco, 2022-05-21:
I'm using Zephyr toolchain 0.14.1 and actually I'm able to build and flash successfully for a couple of projects and boards like cdc_acm sample and echo_bot for frdm_k64f, nrf52840 and ESP32 WROOM , wifi station with shell terminal enabled on ESP32 but I can't build civetweb http_server or webshocket_server samples.

I've tryed to build civetweb samples on zephyr but can't build with success for anyone of my boards. Both projects stucks when building processes. West starts building process but stops saying that can't found some sources...

Have any suggestion that I can use to build http_server and webshocket_server samples right way?

mbolivar-nordic, 2022-05-24:
We have been looking for a civetweb maintainer for a while with no success. I am going to propose that we remove support for this module and close this issue accordingly. If anyone else steps up to maintain it, we can reassign.

tgorochowik, 2022-05-26:
What @rlubos suggests is a valid workaround and should make civetweb build properly, there is a ways to disable posix APIs in Zephyr so there are no conflicts regarding pthreads, the problem here is that `WIFI_SIMPLELINK` requires both newlibc and Zephyr posix APIs (both provide pthreads) - it probably needs those APIs for other  reasons than actual pthreads but this is what causes the conflict and renders pthreads unusable when Simplelink is enabled - I think it should be investigated from this perspective.

> I'm with similar problem here. But I can't build civetweb to any of my board. Even e when I try to build for "atmel-explained" it not build successful.

It works just fine with current Zephyr when started like this:
```
west build -p auto -b sam_e70_xplained zephyr/samples/net/civetweb/http_server
```

If you are getting some errors building it, please file a separate issue and paste the logs you're getting.

fkokosinski, 2022-05-31:
This issue seems to be caused by `CONFIG_SIMPLELINK_HOST_DRIVER` y-selecting `CONFIG_REQUIRES_FULL_LIBC`. By removing this dependency and applying some changes to include paths (see attached diff), I was able to build `samples/net/civetweb/http_server` for `cc3235sf_launchxl`.

Diff: [cc3235sf_launchxl-civet.diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/8806683/cc3235sf_launchxl-civet.diff.txt)

gnfranco, 2022-05-31:
> What @rlubos suggests is a valid workaround and should make civetweb build properly, there is a ways to disable posix APIs in Zephyr so there are no conflicts regarding pthreads, the problem here is that `WIFI_SIMPLELINK` requires both newlibc and Zephyr posix APIs (both provide pthreads) - it probably needs those APIs for other reasons than actual pthreads but this is what causes the conflict and renders pthreads unusable when Simplelink is enabled - I think it should be investigated from this perspective.
> 
> > I'm with similar problem here. But I can't build civetweb to any of my board. Even e when I try to build for "atmel-explained" it not build successful.
> 
> It works just fine with current Zephyr when started like this:
> 
> ```
> west build -p auto -b sam_e70_xplained zephyr/samples/net/civetweb/http_server
> ```
> 
> If you are building for `sam4s_xplained` then it doesn't have any random generator and you need to enable `CONFIG_TEST_RANDOM_GENERATOR` to build it.
> 
> If you are getting some other errors building, please file a separate issue and paste the logs you're getting (and the commands you use).

You are right... It's build properly from the root of Zephyr installation directory.
It is an good point.. I'm reading and whating people on YouTube that brave as good practices to you build projects "out of tree". I'm just source zephyr-env.sh and all seens to be right when I try to build most projects samples since now. But I achieve my first success build just after following your tip here.

What will be better to do? I have to build projects out of tree or is a better option to make all my build from the root instalation folder?

And so... Why some projects can't build out of installing tree?

tgorochowik, 2022-06-01:
In general, samples are in tree because they are just that - samples - if you are starting your own project based on Zephyr it is indeed better to build out of tree - if you are struggling with that, please create a new issue and include any relevant details. Please refer to https://github.com/zephyrproject-rtos/example-application for out of tree examples for apps/drivers/boards etc.

mbolivar-nordic, 2022-06-22:
"Fixed" by the removal of civetweb. See  #46746 for more details. Thanks for taking the time to file an issue and sorry for the inconvenience.

PR #46746 (2022-06-22): civetweb: remove obsolete code
Retrying https://github.com/zephyrproject-rtos/zephyr/pull/45988 now that we have released v3.1 and nobody has stepped up to maintain civetweb. (I can't reopen the old PR due to GitHub restrictions.)

This code has gone unmaintained and bugs continue to be reported
against it. We do not have the resources as a project to maintain this
in "odd fixes" mode, and nobody has stepped up to maintain it [1], so
sadly this must be removed for now.

If anyone would like to see civetweb supported in upstream Zephyr
again, they are welcome to add it back, as long as they pro [... cut]
commit f49d11d3 (2022-06-22): civetweb: remove obsolete code

This code has gone unmaintained and bugs continue to be reported
against it. We do not have the resources as a project to maintain this
in "odd fixes" mode, and nobody has stepped up to maintain it [1], so
sadly this must be removed for now.

If anyone would like to see civetweb supported in upstream Zephyr
again, they are welcome to add it back, as long as they promise to
maintain it going forward.

Many thanks to everyone who has contributed to civetweb support in
Zephyr while it was here. So long and thanks for all the fish.

Fixes: #45807
Fixes: #43910
Fixes [... cut]
closed 2022-06-22 by mbolivar-nordic