zephyrproject-rtos/zephyr #45834: SMP Server Sample needs `-DDTC_OVERLAY_FILE=usb.overlay` for CDC_ACM

richesonk, 2022-05-20:
**Describe the bug**
In trying to build the SMP Server sample application for CDC_ACM backend on my nRF5340DK, I had to add `-DDTC_OVERLAY_FILE=usb.overlay` to the build command given in the example. This overlay was needed for the device to get correctly "installed" on Windows as an CDC_ACM device.

It's unclear to me if that file should have been picked up automatically by the build system (in which case this is an actual code/build bug) or if the documentation just needs to be updated to add that definition in CDC_ACM example build line.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the sample as described at https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html#building-the-sample-application (under the USB CDC_ACM tab).
2. Flash the application to the nRF5340DK
3. Plug the nRF5340DK USB port (the port actually routed to the nRF5340, not the one to the on-board Segger) into a Windows (Windows 10 in my case) machine
4. Notice that device enumerates, but no driver is installed. Look in Device Manager and see that the device is listed under "Other Devices" with a marking indicating that no driver has been installed.
 
**Expected behavior**
Device should enumerate and be identified and "installed" as a COM port (which does happen if the `-DDTC_OVERLAY_FILE=usb.overlay` definition is added to the build line)

**Impact**
This is a minor annoyance, but did cost a little bit of time to sort it out. Hopefully it's just a documentation issue

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Nordic nRF Connect SDK
 - Zephyr commit 038a2de883 included via nRF Connect SDK at v1.9.99-dev1

jfischer-no, 2022-05-20:
>Hopefully it's just a documentation issue

Thanks for the report, it is documentation issue.

mbolivar-nordic, 2022-05-24:
@de-nordic could you please send a docs patch?

de-nordic, 2022-05-25:
> @de-nordic could you please send a docs patch?

Will do.

PR #46004 (2022-05-25): samples/mgmt/mcumgr: Fix CDC ACM sample build information
The commit adds missing info on required usb.overlay.

Fixes: #45834
commit 1d6b275c (2022-05-25): samples/mgmt/mcumgr: Fix CDC ACM sample build information

The commit adds missing info on required usb.overlay.

Fixes: #45834
closed 2022-05-25 by mbolivar-nordic