zephyrproject-rtos/zephyr #45842: drivers: modem: uart_mux errors after second call to gsm_ppp_start

gordonklaus, 2022-05-20:
After calling gsm_ppp_start and successfully connecting to the network, then calling gsm_ppp_stop and disconnecting, then calling gsm_ppp_start again, uart_mux errors occur.  Powering up and down the modem via power callbacks.  Using a BG96 modem.

Logs:
```sh
*** Booting Zephyr OS build v3.1.0-rc1  ***
[00:00:00.004,000] <inf> ulmo_main: Starting application
[00:00:00.610,000] <inf> ulmo_bg9x: modem status: 0
[00:00:06.871,000] <inf> ulmo_bg9x: modem status changed: 1
[00:00:06.877,000] <inf> ulmo_main: modem powered on
[00:00:08.883,000] <inf> ulmo_main: modem powered on
[00:00:09.903,000] <inf> modem_gsm: PPP channel 2 connected to GSM_1
[00:00:10.175,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:10.236,000] <inf> modem_gsm: Model: BG96
[00:00:10.296,000] <inf> modem_gsm: Revision: BG96MAR02A07M1G
[00:00:10.358,000] <inf> modem_gsm: IMEI: # deleted
[00:00:10.369,000] <inf> modem_gsm: Network registration denied.
[00:00:11.381,000] <inf> modem_gsm: Network registration denied.
[00:00:12.393,000] <inf> modem_gsm: Network registered, roaming.
[00:00:12.413,000] <inf> net_ppp: Initializing PPP to use GSM_1
[00:00:12.426,000] <inf> modem_gsm: AT channel 1 connected to GSM_2
[00:00:15.570,000] <inf> ulmo_main: Network connected
[00:00:20.575,000] <inf> ulmo_main: Network disconnected
[00:00:20.588,000] <inf> modem_gsm: GSM network interface down
[00:00:23.538,000] <inf> ulmo_bg9x: modem status changed: 0
[00:00:23.544,000] <inf> ulmo_main: modem powered off
[00:00:29.815,000] <inf> ulmo_bg9x: modem status changed: 1
[00:00:29.821,000] <inf> ulmo_main: modem powered on
[00:00:31.827,000] <inf> ulmo_main: modem powered on
[00:00:32.847,000] <inf> modem_gsm: PPP channel 2 connected to GSM_1
[00:00:33.122,000] <inf> modem_gsm: Network registration denied.
[00:00:34.134,000] <inf> modem_gsm: Network registered, roaming.
[00:00:34.152,000] <inf> net_ppp: Initializing PPP to use GSM_1
[00:00:37.199,000] <err> uart_mux: Ring buffer full, drop 1 bytes
[00:00:37.205,000] <err> uart_mux: Ring buffer full, drop 1 bytes
[00:00:38.208,000] <err> uart_mux: Ring buffer full, drop 1 bytes
[00:00:38.215,000] <err> uart_mux: Ring buffer full, drop 1 bytes
[00:00:39.219,000] <err> uart_mux: Ring buffer full, drop 1 bytes
[00:00:39.226,000] <err> uart_mux: Ring buffer full, drop 1 bytes
[00:00:40.230,000] <err> uart_mux: Ring buffer full, drop 1 bytes
[00:00:40.236,000] <err> uart_mux: Ring buffer full, drop 1 bytes
# same errors for another 17 seconds, then silence
```

My blinking LED stops blinking around the same time the errors appear.  Maybe the system halts, or just the system workqueue?

This is using v3.1.0-rc1.

bbilas, 2022-05-20:
Have you tried to use `quectel-bg9x` driver that should be more suitable for this modem?

gordonklaus, 2022-05-22:
@bbilas I haven't.  Thanks for the suggestion.  I'll give it a try.

Shouldn't the `gsm_ppp` driver also be suitable for the BG96?  There's `CONFIG_MODEM_GSM_QUECTEL` which suggests it should.  And also the fact that it works great after the first call to `gsm_ppp_start` :)

Is the main difference between these drivers the fact that `quectel-bg9x` offloads socket operations to the modem, while `gsm_ppp` works in PPP mode?  Are there other important differences?

bbilas, 2022-05-23:
> @bbilas I haven't. Thanks for the suggestion. I'll give it a try.
> 
> Shouldn't the `gsm_ppp` driver also be suitable for the BG96? There's `CONFIG_MODEM_GSM_QUECTEL` which suggests it should. And also the fact that it works great after the first call to `gsm_ppp_start` :)

I haven't got a chance to use BG96 so I don't know, to be honest. I've been using the `gsm_ppp` driver with the EC21 and EG21 modules only as it's noted here https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/Kconfig.gsm#L35=.

> Is the main difference between these drivers the fact that `quectel-bg9x` offloads socket operations to the modem, while `gsm_ppp` works in PPP mode? Are there other important differences?

I've never been using `quectel-bg9x` driver so I can't help here.

closed 2022-08-07 by github-actions[bot]