zephyrproject-rtos/zephyr #45844: Not all bytes are download with http request

AndreyDodonov-EH, 2022-05-20:
**Describe the bug**
Downloading considerable amount of data fails.
Getting fewer bytes than expected, without any error indication from networking stack.

Using http_client, socket_tls and default TCP stack (no socket offloading).

Download amount ca. 1 MB. Chunks are 1 KB.

In general, use case is very close to `big_http_download` sample, but:
- in the setup where resends are needed (medium-quality Wi-Fi and slow target).
- without explicitly setting any networking sizes -` CONFIG_NET_TCP_MAX_RECV_WINDOW_SIZE, CONFIG_NET_TCP_MAX_SEND_WINDOW_SIZE, CONFIG_NET_PKT_TX_COUNT, CONFIG_NET_PKT_RX_COUNT, CONFIG_NET_BUF_TX_COUNT or CONFIG_NET_BUF_RX_COUNT`

**Expected behavior**
All bytes are downloaded

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain NRF with SDK 1.9.99 (based on Zephyr 3.0) or Zephyr main. With Zephyr 2.7 all works as expected.

rlubos, 2022-05-26:
I was able to reproduce the issue with `qemu_x86`, after modifying `slip` driver to randomly drop packets. However from the observations I've made, I'm not really convinced it's a Zephyr issue, let me elaborate.

With 10% received packet drop the download seemed still stable, it was sluggish due to retransmissions, it did not fail though. After increasing the packet drop to 20% I've observed two possible scenarios for the download failure:
1. The file server increases the retransmission timeout exponentially on each unaclowneged packets. I've noticed that  in case of a few consecutive packet drops however it appears to give up, there's no RST message or connection close, the connection just stalls. I've set a RX timeout on a socket to 10 minutes, after that the connection is closed but this is initiated client-side. The socket reports EAGAIN in such case (RX timeout).
2. The file server closes the TCP connection abruptly, after getting an ACK for retransmitted packet. The reason for this event is unknown to me, but again I don't see any blame on the Zephyr side. The connection was closed by the peer, and that's how it's notfied at the socket layer (by returning 0 from recv()).

Fortunately I was able to catch both failures in a single PCAP file, the connection stall can be observed at frame number 789 (after which we close the connection from the client side), the TCP connection close can be observed at grame number 131. I'm attaching the PCAP file, perhaps someone could explain this behaviour better than me:
[TLS_retransmit_issues.zip](https://github.com/zephyrproject-rtos/zephyr/files/8779492/TLS_retransmit_issues.zip)

The only difference I've spotted between Zephyr 2.7 and 3.0 is the RX window size, for 3.0 it's larger by default. With smaller (1280 as fixed in 2.7) window size I could only reproduce scenario 1 with 20% packet drop. Same effect with 3.0 and window size reduced to 1280 bytes.

@AndreyDodonov-EH Any chance to provide a wireshark log for your case, so we could investigate together what's happening on your side, i. e. whether you hit the same issue?

carlescufi, 2022-06-01:
@AndreyDodonov-EH could you please reply to the comment above?

AndreyDodonov-EH, 2022-06-01:
> @AndreyDodonov-EH could you please reply to the comment above?

@carlescufi  I'll provide the Wireshark traces and some additional info in 2-3 days, currently busy on another project.
@rlubos  I  really appreciate you looking into this even without me providing sufficient information.

AndreyDodonov-EH, 2022-06-09:
@carlescufi @rlubos 
Here are the logs in happy case (2.7, "No_MissingBytes.pcapng") and unhappy case (main, "MissingBytes.pcapng")

I could see that Herou sets RST flag, but for some reason it's doing that only with newer version of Zephyr (didn't dig deeeper).

[Traces.zip](https://github.com/zephyrproject-rtos/zephyr/files/8870535/Traces.zip)

Also some basic logs in unhappy case:

```
[00:00:15.299,743] <inf> socket_manager: DNS resolved successfully
[00:00:15.299,835] <inf> socket_manager: IP address of netilion-ehlp-public-file-api.herokuapp.com is 54.73.53.134
[00:00:15.299,865] <inf> socket_manager: Trying to access 54.73.53.134 on port 443
[00:00:15.300,659] <inf> socket_manager: Set TLS socketopt successful
[00:00:19.195,312] <inf> netilion_cloud_client: HTTP GET request on netilion-ehlp-public-file-api.herokuapp.com/file/QWX43SEANDR
[00:00:25.807,556] <inf> netilion_cloud_client: Downloading: 9999/1048576
[00:00:31.747,558] <inf> netilion_cloud_client: Downloading: 20239/1048576
[00:00:31.818,359] <inf> netilion_cloud_client: Downloading: 30479/1048576
[00:00:36.479,156] <inf> netilion_cloud_client: Downloading: 40719/1048576
[00:00:36.550,231] <inf> netilion_cloud_client: Downloading: 50959/1048576
[00:00:41.211,486] <inf> netilion_cloud_client: Downloading: 61199/1048576
[00:00:41.316,925] <inf> netilion_cloud_client: Downloading: 71439/1048576
[00:00:45.945,831] <inf> netilion_cloud_client: Downloading: 81679/1048576
[00:00:50.399,383] <inf> netilion_cloud_client: Downloading: 91919/1048576
[00:00:50.474,609] <inf> netilion_cloud_client: Downloading: 102159/1048576
[00:00:55.130,371] <inf> netilion_cloud_client: Downloading: 112399/1048576
[00:00:55.200,439] <inf> netilion_cloud_client: Downloading: 122639/1048576
[00:00:59.864,685] <inf> netilion_cloud_client: Downloading: 132879/1048576
[00:01:04.534,484] <inf> netilion_cloud_client: Downloading: 143119/1048576
[00:01:04.605,438] <inf> netilion_cloud_client: Downloading: 153359/1048576
[00:01:09.050,842] <inf> netilion_cloud_client: Downloading: 163599/1048576
[00:01:09.125,274] <inf> netilion_cloud_client: Downloading: 173839/1048576
[00:01:13.886,291] <inf> netilion_cloud_client: Downloading: 184079/1048576
[00:01:18.545,532] <inf> netilion_cloud_client: Downloading: 194319/1048576
[00:01:18.621,429] <inf> netilion_cloud_client: Downloading: 204559/1048576
[00:01:23.290,191] <inf> netilion_cloud_client: Downloading: 214799/1048576
[00:01:23.364,227] <inf> netilion_cloud_client: Downloading: 225039/1048576
[00:01:27.809,173] <inf> netilion_cloud_client: Downloading: 235279/1048576
[00:01:28.113,586] <inf> netilion_cloud_client: Downloading: 245519/1048576
[00:01:32.539,703] <inf> netilion_cloud_client: Downloading: 255759/1048576
[00:01:37.199,676] <inf> netilion_cloud_client: Downloading: 265999/1048576
[00:01:37.275,115] <inf> netilion_cloud_client: Downloading: 276239/1048576
[00:01:41.930,145] <inf> netilion_cloud_client: Downloading: 286479/1048576
[00:01:42.008,422] <inf> netilion_cloud_client: Downloading: 296719/1048576
[00:01:46.664,001] <inf> netilion_cloud_client: Downloading: 306959/1048576
[00:01:51.321,197] <inf> netilion_cloud_client: Downloading: 317199/1048576
[00:01:51.397,735] <inf> netilion_cloud_client: Downloading: 327439/1048576
[00:01:56.057,098] <inf> netilion_cloud_client: Downloading: 337679/1048576
[00:01:56.134,094] <inf> netilion_cloud_client: Downloading: 347919/1048576
[00:02:00.584,442] <inf> netilion_cloud_client: Downloading: 358159/1048576
[00:02:05.240,295] <inf> netilion_cloud_client: Downloading: 368399/1048576
[00:02:05.310,394] <inf> netilion_cloud_client: Downloading: 378639/1048576
[00:02:09.979,858] <inf> netilion_cloud_client: Downloading: 388879/1048576
[00:02:10.052,795] <inf> netilion_cloud_client: Downloading: 399119/1048576
[00:02:14.703,887] <inf> netilion_cloud_client: Downloading: 409359/1048576
[00:02:14.808,258] <inf> netilion_cloud_client: Downloading: 419599/1048576
[00:02:19.234,313] <inf> netilion_cloud_client: Downloading: 429839/1048576
[00:02:23.892,150] <inf> netilion_cloud_client: Downloading: 440079/1048576
[00:02:23.964,935] <inf> netilion_cloud_client: Downloading: 450319/1048576
[00:02:28.627,136] <inf> netilion_cloud_client: Downloading: 460559/1048576
[00:02:28.700,622] <inf> netilion_cloud_client: Downloading: 470799/1048576
[00:02:33.357,391] <inf> netilion_cloud_client: Downloading: 481039/1048576
[00:02:37.811,859] <inf> netilion_cloud_client: Downloading: 491279/1048576
[00:02:37.888,549] <inf> netilion_cloud_client: Downloading: 501519/1048576
[00:02:42.551,452] <inf> netilion_cloud_client: Downloading: 511759/1048576
[00:02:42.626,525] <inf> netilion_cloud_client: Downloading: 521999/1048576
[00:02:45.625,122] <inf> netilion_cloud_client: Request finished. Sent 98, received (524047/1048576).
```

rlubos, 2022-06-09:
The only difference I see is the window size, after we implemented a proper RX window handling, we report smaller window in the ACK. But this doesn't explain why does the server decide to abort the connection out of blue, I see nothing suspicious about hte ACK whcih seemingly has triggered the RST message...

Anyway, did you have commits https://github.com/zephyrproject-rtos/zephyr/pull/45070 included when testing the bad scenario? It could be the case that such an event went unnoticed w/o those commits, the connection abort from the peer side would be reported as a graceful shutdown. But it should be now reported properly as an error condition (ECONNRESET AFAIR).

AndreyDodonov-EH, 2022-06-09:
Yes, Those commits are included. 

Still, I can try to update networking to newest state (e.g. NRF SDK 2.0) and check the behavior.

AndreyDodonov-EH, 2022-06-09:
Tried networking from NRF SDK 2.0, same issue.

However, when I explicitly made RCV window bigger, download works:
`CONFIG_NET_TCP_MAX_RECV_WINDOW_SIZE=2560`

[No_MissingBytes_Because_bigger_window.zip](https://github.com/zephyrproject-rtos/zephyr/files/8871614/No_MissingBytes_Because_bigger_window.zip)

rlubos, 2022-06-09:
> However, when I explicitly made RCV window bigger, download works

Would be really great to understand what is so exceptional about the larger window size that makes the server to retain the TCP connection...

AndreyDodonov-EH, 2022-06-10:
> > However, when I explicitly made RCV window bigger, download works
> 
> Would be really great to understand what is so exceptional about the larger window size that makes the server to retain the TCP connection...

I will try to test the same scenario with other servers than Heroku.

AndreyDodonov-EH, 2022-06-10:
I now tested with a different environment:

1. Local Wi-Fi network
2. Simple Node JS express server
3. HTTP (not HTTPS as before)

Results are... strange.
Both versions 2.7.99 and 3.0 could download the file completely, but in case of 2.7.99 it took 15 seconds, while in case of 3.0 it took almost one hour.

Wireshark logs attached.
[LocalHTTP.zip](https://github.com/zephyrproject-rtos/zephyr/files/8878519/LocalHTTP.zip)

rlubos, 2022-06-10:
Wow, that's indeed strange. If you look closely into the wireshark, you could see a repetitive pattern:
1. The initial RX window size (Zephyr) is 1536
2. Server sends a data packet, full MTU size (1460 app data bytes)
3. Zephyr sends ACK with window size reduced by the received amount of data (window size 76)
4. Now, for whatever reason server takes 5 second break (which is the root cause of the slow download). After that, it sends the maximum possible amout of data according to the window size (76 bytes).
5. As the previous data was already consumed by Zephyr, it replies with the actual window size, which is 1460.
6. Go to point 2 and repeat.

Now this makes me wonder, it looks as if the server was waiting for Zephyr to send a consecutive ACK with an updated window size. I see that Wireshark sometimes tags an acknowledment as "TCP Window Update". I'm not sure, is there an algorithm defined on when to send those? Perhaps we should implement some kind of proactive window notification in Zephyr in case window size is increased above certain treshold? CC @ssharks Perhaps you have some insight here?

ssharks, 2022-06-10:
Hmmm... somewhat unexpected indeed. I'm rolling in after a long discussion. Just blondly looking at your application I see no obvious reasons for the server to abort the download. I'm using the TCP stack in a similar way without issues. Most issues I had where in the uploading direction, not in download.
A thing I have bumped into quite often is that some service providers have minimum throughput rules of X bytes in Y seconds. Otherwise the connection is terminated.
One observation: The succeeding download takes roughly the same amount of time, the broken did.... is this coincidence?

With the default settings I was where getting trouble. So I have increased the number of NET_BUF_RX/TX_COUNT to 32 or higher and decreased the MTU. Lowing the MTU does add some overhead, but offers you to fit more packets into the same memory, avoiding trouble with re-transmit.
The throughput is direct proportional to the receive window and inverse proportional to the round trip time. Most of the time the receive window is the limiting factor for TCP download.

When fiddling with the MTU, make sure you take the fixes of https://github.com/zephyrproject-rtos/zephyr/pull/45890 into account.
A nasty one I also bumped into was: https://github.com/zephyrproject-rtos/zephyr/pull/45890/commits/f8e3acc0b33d5fc411de177e356d1fa698ba0526

Hope it was helpfull

AndreyDodonov-EH, 2022-06-10:
@ssharks As for the time required, in the first case I download from remote HTTPS server (heroku) without any configs. In this case timing was close, yes, for both failing and succeeding cases (2.7.99 and 3.0)

Then increased RCV window size and everything worked.

In the second case, I tried HTTP and local server without setting any sizes via configs. It worked, but took 15 seconds and almost 50 minutes (with 2.7.99 and 3.0, respectively).

AndreyDodonov-EH, 2022-06-10:
@ssharks I think default for buffers is already 36, no? 
https://docs.zephyrproject.org/3.0.0/reference/kconfig/CONFIG_NET_BUF_RX_COUNT.html

Also, how do you low MTU? I didn't see the config flag for that

ssharks, 2022-06-10:
The MTU is gathered from the interface, in my case I use CONFIG_NET_PPP_MTU_MRU=600.

ssharks, 2022-06-10:
> @ssharks As for the time required, in the first case I download from remote HTTPS server (heroku) without any configs. In this case timing was close, yes, for both failing and succeeding cases [...]

So as I read you correctly. The stack is not functionally broken, but the default settings need to be improved to make sure you get at least a reasonably functional system out of the box. Only if you have NET_L2_ETHERNET enabled you have 36, otherwise 16, which barely fits a single 1460 byte MSS.

In my case to be functional (using PPP), I need the CONFIG_NET_TCP_MAX_SEND_WINDOW_SIZE + 2 x MTU to make the TX work. In that sense it would be handy if the MSS would be limited not only by the interface, but also by the number of buffers.

AndreyDodonov-EH, 2022-06-10:
In case of 2.7.99 I had CONFIG_NET_TCP_MAX_SEND_WINDOW_SIZE set to 1152 even, which is lower than Ethernet MTU (1500). And it was working reasonably well

ssharks, 2022-06-12:
After thinking a bit more. You see a similar in the results of @rlubos: https://github.com/zephyrproject-rtos/zephyr/issues/23302#issuecomment-1142138602. 

What I think is happening: When the window full is returned by Zephyr, the host stops sending, waits for a time before probing again. This causes a significant slowdown. The Zephyr stack does never send a TCP window update message. If it would do, or implement a different method of avoiding ending up in a window full situation, the throughput would greatly improve, especially with small receive windows. The exact behavior is largely scheduling depending, which could explain the difference you see between versions.

ssharks, 2022-06-13:
To validate this hypothesis I made a basic implementation of the TCP window update. 

See:
https://github.com/ssharks/zephyr/tree/net/tcp_window_update

Using the tests in tests/net/socket/tcp I do see a considerable reduction in test time. Previously there used to be a considerable difference between the cooperative and preemptive scheduling. With this fix, this difference is gone.

@AndreyDodonov-EH: If you have an opportunity, could you give it a spin on real HW?

Note: It is a premature fix, which needs some work to be upstream able. Feel free pick it up to make it into a real fix.

AndreyDodonov-EH, 2022-06-13:
@ssharks Thanks a lot, I'll give it a try on a real hardware, both local http and remote https cases.

AndreyDodonov-EH, 2022-06-13:
@ssharks Unfortunately I don't see any speed improvement, server still sleeps 5 seconds for whatever reason. 
Also I am quite confused with my traces, see https://github.com/zephyrproject-rtos/zephyr/issues/45844#issuecomment-1152276707

Here it looks as if server sets TCP window full, which I find hard to believe.
I guess I'm misreading the traces...

rlubos, 2022-06-13:
> What I think is happening: When the window full is returned by Zephyr, the host stops sending, waits for a time before probing again.

The thing with those traces here is that Zephyr does not report a Zero Window - it reports a RX window of size 76 bytes, so the server doesn't really send a Zero Window Probe, just a regular data packet, but delayed by 5 seconds. With Zero Window scenario I've observed that servers typically send a probe after ~200ms.

 > To validate this hypothesis I made a basic implementation of the TCP window update.

Thanks, that's more or less what I was thinking we should implement. The algorithm you mention in the comment, is it covered by some official RFC? Or at least do you have any link describing the algorithm?

rlubos, 2022-06-13:
> Unfortunately I don't see any speed improvement, server still sleeps 5 seconds for whatever reason.

@AndreyDodonov-EH Would you be able to share a pcap with @ssharks window update implementation applied? I'm curious what's happening in there. Maybe the threshold is set too low?

> Here it looks as if server sets TCP window full, which I find hard to believe.

Are you referring to `[TCP Window Full]` tag next to the data packet sent by the server? If so, it's not the server setting any Window Full flag, it's just wireshark tagging the packet, that it filled out the current RX window (Zephyr side).

ssharks, 2022-06-13:
> The thing with those traces here is that Zephyr does not report a Zero Window - it reports a RX window of size 76 bytes, so the server doesn't really send a Zero Window Probe, just a regular data [...]

The thing I found:
https://en.wikipedia.org/wiki/Silly_window_syndrome

I took the following side for inspiration:
https://www.geeksforgeeks.org/silly-window-syndrome/#:~:text=Silly%20Window%20Syndrome%20is%20a,shrink%20to%20a%20silly%20value.

Maybe it is good to have a good read of:
https://datatracker.ietf.org/doc/html/rfc813
At the moment, I do not have time to dive deep at this issue.

rlubos, 2022-06-13:
@ssharks Thank you for the links. It's clear now that our implementation suffers from this "Silly Window Syndrome".

If I understood the document correctly, we could mitigate the problem by adding ACK timer (delayed work) to our TCP implementation.
1. In case the reported RX window would be small, postpone the ACK by some small delay (RFC suggests 200 ms),
2. Otherwise send ACK right away. Optionally, we could reuse the delayed work and schedule it with `K_NO_WAIT`  - that way, if more TCP packets are already queued in the RX thread, they should be processed first (in the cooperative mode), and then ACK would be sent only once.
3. When the RX window size is increased (after consuming the data at the application level) **and** the ACK timer is running, if the window size is increased above the threshold, reschedule it to `K_NO_WAIT` - that way the actual delay that the timer was originally scheduled with would rarely apply.

That doesn't sound too complex to implement, I might give it a shot and see how would it actually work.

ssharks, 2022-06-13:
You are totally right. Delayed ACK is I think a very good and efficient way to solve the problem. The only thing is that the timeout should at least be less than the re-transmission timeout, which is default at 200 ms in Zephyr. So 100 ms would be better suited.
Under normal circumstances the receive window will be cleared much sooner.
A point of attention: I would suggest to only apply delayed ACK when the window becomes too small. Default we should react immediately to avoid conflicting with the Nagle algorithm.

A second addition is that the actually advertised receive window can be advertised to be less then the actual available amount  to avoid hosts sending small frames. A suggestion could be to send out a 0 window when the window is less then a threshold, like the MSS.

It would be wonderful if you could give it a go, I think the benefits can be huge.

rlubos, 2022-06-15:
@AndreyDodonov-EH Would be good if you could give it one more try with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/46584

I've tested the throughput with small window and I must say the performance is much better. Would like to see how does that server of yours resond to that :)

AndreyDodonov-EH, 2022-06-18:
> @AndreyDodonov-EH Would be good if you could give it one more try with this PR: #46584
> 
> I've tested the throughput with small window and I must say the performance is much better. Would like to see how does that server of yours respond to that :)

@rlubos 

I tried your PR https://github.com/zephyrproject-rtos/zephyr/pull/46584, and with both local HTTP and remote HTTPS everything works nicely!
Downloading 1MB from local host HTTP in chunks of 1KB takes ca. 9 seconds now!
Also no missing bytes in download from remote HTTPS server. I also observed ca. 20% speed-up compared to 2.7.99.

closed 2022-06-23 by carlescufi