zephyrproject-rtos/zephyr #45845: tests: The failure test case number increase significantly in CMSIS DSP tests on ARM boards.

enjiamai, 2022-05-20:
**Describe the bug**
After the twister overhaul PR https://github.com/zephyrproject-rtos/zephyr/pull/44094, it reveal some incorrect testcase numbers in the twister report. 
Take frdm_k64f as an example, following tests are affected most:  
tests/lib/cmsis_dsp/martix  => 5940   failure
tests/lib/cmsis_dsp/transform  => 1100  failure
tests/lib/cmsis_dsp/filtering => 2172 failure

Because that testcases cannot be parsed well when twister to scan the test case from source code. We did not see this issue because
these incorrect testcases was judge as "skipped" instead of "failure" before this PR, but issues existed before.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephry-env.sh
2. twister -p frdm_k64f --device-testing --device-serial=/dev/ttyACM0  -T tests/lib/cmsis_dsp/ -v

**Expected behavior**
The failure testcases should be very less.

**Impact**
The failure testcases increase significantly in the test report.

**Logs and console output**
```
<testsuites><testsuite name="frdm_k64f" timestamp="2022-05-18T01:48:09.248298" time="115.16999999999999" tests="15000" failures="9282" errors="0" skipped="154"><properties>
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK 0.14.2)
 - Commit SHA: d78194546746

enjiamai, 2022-05-20:
A workaround for this: https://github.com/zephyrproject-rtos/zephyr/pull/45832

enjiamai, 2022-05-28:
Brief summary of this issue:
The problem happens due to 2 factors:

(1) There are some testcases show in testplan but not a expect testcases need to run. This issue was existed previously.
  
This is due to twister cannot parse the scanned source code well for some testcases was written by self-defined macro. It's un-parsable unless we can consider the #ifdef factor, that means need more complicated preprocessor mechanism to deal with that.   

(2) After twister overhaul, these incorrect testcases were revealed and were reported status(blocked) reason(unknown) so they were marked as failed.
These testcases were judged as skip previously. 

There might be some solutions for this issue: 
(1) Change the way these testcases defined.
(2) Improve the way of getting testcases in twister.  #46052 is one of this way to solve it.

smrtos, 2022-05-30:
Should be fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/46052

PR #46278 (2022-06-07): twister: fix setting status on unprocessed tests
- twister: do not mark tests as blocked if a testsuite has passed
- twister: do not set case status on flash timeout

Fixes #45845
commit bcf4e235 (2022-06-07): twister: do not set case status on flash timeout

Status was set for testcases after a timeout on flash, which does not
abort the process and continues with test execution and eventually
passes the test.

Setting status on the testcases was done too early and before we have
executed the tests.

Also increase the flash timeout to 60s.

Fixes #45845
commit eeffdc37 (2022-06-07): twister: do not mark tests as blocked if a testsuite has passed

Do not set status as blocked on testsuites which passed. The sub cases
with no status will get a processed later.

Fixes #45845
closed 2022-06-07 by nashif