zephyrproject-rtos/zephyr #45846: New ZTEST API for noisily skipping a test based dependency failures

aaronemassey, 2022-05-20:
**Is your feature request related to a problem? Please describe.**

When conducting a larger integration test a system may depend on numerous
drivers that the integration test is only indirectly verifying -- i.e. we're
testing the system that relies on the driver, not the driver itself.
It's entirely possible that a regression in one particular feature for
a single driver causes a large number of assertion failures in such an
integration test.

Another such example would be testing a basic feature of a complex system first and
then testing numerous other more advanced features that rely on that basic
feature.
If the basic feature fails, they all fail.

With 10 or fewer assertion failures this may not be such a problem, but with
increased assertions within an integration test it becomes increasingly difficult
to pin-point the suspect cause of the failures.
This is especially compounded by the fact a test failure investigator may not
be familiar with the system under test.

I understand this is more of a future problem, but one I think we *want to have*
it'd be a lot easier to have a solution for this as we write these
integration tests instead of coming back to them to understand the functional
dependencies in the system-under-test.

**Describe the solution you'd like**

A macro similar to `zassert` in the new ZTEST_API that noisily skips a test
(still failing the suite) with a `Assumption failed` not to be confused with
`Assertion failed`.
One obvious name for such a macro is `zassume`.
However, @jrosenth suggested a different name for this function to more
obviously indicate its difference from `zassert`.

Essentially, tell the test-runner that this test couldn't be run because the
functionality it relies on is broken or missing.

Ideas:

* `zassume` - `Assumption failed`
* `zsuppose` - `Supposition failed`
* `zpresuppose` - `Presupposition failed`
* `zpresume` - `Presumption failed`
* `zpresuppose` - `Presupposition failed`
* `zrequire` - `Requirement failed`

There exists pretext for such functionality within a test framework in the
[JUnit API](https://junit.org/junit4/javadoc/4.12/org/junit/Assume.html). 

I'm also more than happy to implement this if the community is on
board. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Status Quo:

Pros:
* Nothing to be done!
* No one has to learn an added feature of the test API.

Cons:
* The problem I mentioned clearly still exists.
* It'll be difficult to course-correct existing integration tests.

**Additional context**

Example:

```c
// Located in some test setup
zrequire_ok(my_driver_dependency_init());
...
...
...
// Located in the actual test
zassert_ok(my_complex_system_feature());
```

**What I'm looking for**

I'd like some comments about if this interesting to the greater community and
what would be the appropriate name. I'll also be bringing this up in the testing
work group next week.

aaronemassey, 2022-05-23:
There is a very real downside to using a name other than `zassume` and that's because the `assume` name is pretty standard for this kind of thing given its implementation in the JUnit4 API.

PerMac, 2022-05-24:
Interesting idea. Just trying to clarify if I got it correctly: at the suite level (i.e related to the whole built application) we would get an error, but a particular testcase with failing `zassume` will be skipped from the report or get a skipped status? I am asking in terms of junit type report and its limited available statuses.  
 If skipped from report, I am worried about total number of test cases in a run, since test cases won't always add to the same number. I know it is already not always the case (e.g a build failed and not all/none test cases were populated) but we are trying to solve this. 
I am asking about junit xmls since we use jenkins a lot in my company and use its support for test management based on those xmls. I haven't seen "assumes" yet, so I am wondering how are they reported.

aaronemassey, 2022-05-25:
> Interesting idea. Just trying to clarify if I got it correctly: at the suite level (i.e related to the whole built application) we would get an error, but a particular testcase with failing [...]

So it's been a while since I've worked that in detail with JUnit reports. After a bit of hunting I did find [https://llg.cubic.org/docs/junit/](https://llg.cubic.org/docs/junit/), which describes how a skip would be reported as a tag, similarly to a failure tag.

That being said, for exact details on how this will be reported in an xml file from twister I'll link a PR that fixes this issue here and we can continue discussion there when its ready. This is still in the early phases at the moment but something I'd like to get done soon after the Zephyr Dev Summit.

How do you feel about the name? Do you think zassume is too similar to zassert?