zephyrproject-rtos/zephyr #45851: For native_posix programs, k_yield doesn't yield to k_msleep threads

corbett-cappon, 2022-05-21:
When compiling programs for native_posix, if a higher priority thread calls k_msleep with a small timeout, then the lower priority thread calls k_yield, control is not returned to the higher priority thread, even if the small timeout has expired.

**To Reproduce**

Attached is a simple program to demonstrate this. Unzip in zephyr/samples, compile via "west build -b native_posix", and run it via "cd build/native_posix/test_yield_native_posix/zephyr" then "./zephyr.exe".

[test_yield_native_posix.zip](https://github.com/zephyrproject-rtos/zephyr/files/8747065/test_yield_native_posix.zip)

**Expected behavior**

Program should print:
"Beginning High Priority Thread"
the character "H" zero or more times
Beginning Low Priority Thread
the characters "H" and "L" interleaved one or more times
"Exiting Main Thread"
then the program should exit

**Actual behavior**

Instead we get the following output:

Beginning High Priority Thread
Beginning Low Priority Thread
H
L
L
then the character "L" printed continuously, and the program never exits. This is because the low priority thread never gives control to either the sleeping main thread, or the sleeping high priority thread.

**Impact**

This prevents the uart unit tests from working with my interrupt uart driver that I wrote, unless I modify the zephyr unit tests to use "sleep" instead of "yield". It might also cause other bugs, due to my threads in my interrupt uart driver being starved, or other various threads in native_posix being starved.

**Logs and console output**

Here is a console log from running my test program

[test_yield_native_posix_console_output.txt](https://github.com/zephyrproject-rtos/zephyr/files/8747075/test_yield_native_posix_console_output.txt)

**Environment (please complete the following information):**

OS: Ubuntu 20.04.
Toolchain: Zephyr SDK
Version used: Zephyr 3.1.0-rc1

**Additional context**

Can't think of any needed additional context

aescolar, 2022-05-25:
Hi @corbett-cappon,
it seems what you are facing is a relatively elaborate infinite loop in the low priority thread. Note that in native_posix code itself takes no simulated time to execute, so the code executes and yields in zero time, and the kernel switches back to it immediately as the higher priority thread is not yet ready. Please take a loop here:
https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#important-limitations 
You can work around it by having your low priority thread take a bit of time when compiled for native_posix, by adding a conditionally compiled call to [k_busy_wait()](https://docs.zephyrproject.org/latest/kernel/services/threads/index.html#c.k_busy_wait) 
https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#working-around-these-limitations

aescolar, 2022-05-25:
@mbolivar-nordic I removed the bug label, as this is not really a bug, but a known limitation. But I do not know what label we can apply nowadays for this.

mbolivar-nordic, 2022-05-25:
@aescolar if this is a documented limitation, I think we can close this.

@corbett-cappon if you want to open a separate enhancement issue to get this fixed, feel free, but since this is not a bug, I'd like to close out the issue for now.

closed 2022-05-25 by mbolivar-nordic