zephyrproject-rtos/zephyr #45855: Runtime fault when running with CONFIG_NO_OPTIMIZATIONS=y

aloksethi, 2022-05-21:
**Describe the bug**
I have two threads in the code which are just printing a string on the uart. If i compile with CONFIG_NO_OPTIMIZATIONS=y, I see two prints on the uart and then nothing. If i attach a debugger (Ozone), I see some usage fault (varying from memfault to usage fault).

I am using arduino_nano_33_ble board. I have connected a FTDI serial to USB board to the two serial  lines and observing the output on minicom. For debugging, I have connected the 5 lines to the nordic PCA10040 board (nrf52DK i think)

The issue seems very similar to issue  **#12820**, however, even increasing the size of the stacks  by 10 times didn't help. Further, I do not see any traces related to fault on the uart. I have even increased the PRIVILEGED_STACK_SIZE from 1024 to 4096 but that didn't help either.

below are the contents of my prj.conf
```
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y

#CONFIG_STACK_SENTINEL=y
CONFIG_NO_OPTIMIZATIONS=y

#CONFIG_TRACING=y
#CONFIG_ASSERT=y

CONFIG_PRIVILEGED_STACK_SIZE=4096
CONFIG_THREAD_STACK_INFO=y
CONFIG_THREAD_MONITOR=y
CONFIG_KERNEL_SHELL=y
CONFIG_INIT_STACKS=y
CONFIG_THREAD_NAME=y
```
contents of my code file
```
#define UART_DEVICE_NODE DT_CHOSEN(zephyr_shell_uart)
const struct device *g_dbg_uart_dev = DEVICE_DT_GET(UART_DEVICE_NODE);

#define DBGS_HDLR_STACK_SIZE	1024
#define DBGS_HDLR_PRIORITY	7/*(CONFIG_NUM_PREEMPT_PRIORITIES - 1)*/

static void dbgs_hdlr2(void *p1, void *p2, void *p3)
{
	ARG_UNUSED(p1);
	ARG_UNUSED(p2);
	ARG_UNUSED(p3);

	while(1)
	{
		print_uart("alive2\r\n");
		k_msleep(100);
		//k_sleep(K_FOREVER);
	}
}
static void dbgs_hdlr(void *p1, void *p2, void *p3)
{
	ARG_UNUSED(p1);
	ARG_UNUSED(p2);
	ARG_UNUSED(p3);

	dbg_ser_init();

	while(1)
	{
		print_uart("alive\r\n");
		k_msleep(300);
		//k_sleep(K_FOREVER);
	}
}
/*
 * callback for interrupt handler
 */
static void dbgs_int_cb(const struct device *uart_dev, void *user_data)
{
	uint8_t c;

	if (!uart_irq_update(uart_dev)) {
		return;
[... 126 lines cut ...]
[36/53] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
[37/53] Linking C static library zephyr/drivers/pinctrl/libdrivers__pinctrl.a
[38/53] Linking C static library zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a
[39/53] Linking C static library zephyr/kernel/libkernel.a
[40/53] Building C object zephyr/arch/arch/arm/core/aarch32/mpu/CMakeFiles/arch__arm__core__aarch32__mpu.dir/arm_core_mpu.c.obj
[41/53] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/malloc.c.obj
[42/53] Linking C static library zephyr/lib/libc/minimal/liblib__libc__minimal.a
[43/53] Building C object zephyr/arch/arch/arm/core/aarch32/mpu/CMakeFiles/arch__arm__core__aarch32__mpu.dir/arm_mpu.c.obj
[44/53] Linking C static library zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a
[45/53] Linking C executable zephyr/zephyr_pre0.elf

[46/53] Generating dev_handles.c
[47/53] Building C object zephyr/CMakeFiles/zephyr_pre1.dir/dev_handles.c.obj
[48/53] Linking C executable zephyr/zephyr_pre1.elf

[49/53] Generating linker.cmd
[50/53] Generating isr_tables.c, isrList.bin
[51/53] Building C object zephyr/CMakeFiles/zephyr_final.dir/dev_handles.c.obj
[52/53] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[53/53] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       61760 B       928 KB      6.50%
            SRAM:       25536 B       256 KB      9.74%
        IDT_LIST:          0 GB         2 KB      0.00%
```

carlescufi, 2022-05-24:
@aloksethi could you please show us  your `prj.conf` with **all** stack sizes increased?

aloksethi, 2022-05-24:
@carlescufi  I used CONFIG_PRIVLEGED_STACK_SIZE=4096 in my prj.conf
For increasing the stack size of threads I just passed the increased size to K_THREAD_DEFINE.

Is there any other size that should have been increased?

closed 2022-05-26 by anangl

aloksethi, 2022-05-27:
@anangl thanks. i will check it.

aloksethi, 2022-05-27:
@anangl btw how did you figure it out that the culprit is idle thread? basically is there any good thread aware debugging tool?