zephyrproject-rtos/zephyr #45866: drivers/entropy: stm32: non-compliant RNG configuration on some MCUs

taltenbach, 2022-05-22:
**Describe the bug**
On some STM32 MCUs, such as the STM32H723 or STM32L5, the RNG peripheral must be configured according to specific values provided in the reference manual to obtain best latency and to be compliant with NIST. This includes:
* Writing appropriate values to CONFIG1, CONFIG2 and CONFIG3 fields in RNG_CR register.
* Writing a proper value in RNG_HTCR register. 

However, it seems that the RNG driver:
* Does not write CONFIGx fields, although their reset value does not correspond the recommended configuration.
* Does not properly follow the process specified in the reference manual to write to RNG_HTCR. Indeed, CONDRST is not set before writing to RNG_HTCR, even though the [RM0468](https://www.st.com/resource/en/reference_manual/rm0468-stm32h723733-stm32h725735-and-stm32h730-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) states (p. 1440):
> Writing in RNG_HTCR is taken into account only if CONDRST bit is set, and CONFIGLOCK bit is cleared in RNG_CR. Writing to this register is ignored if CONFIGLOCK=1.

**To Reproduce**
N/A

**Expected behavior**
The STM32 RNG peripheral should be properly configured. 

**Impact**
RNG configuration is not compliant with NIST and entropy generation latency might not be ideal. 

**Environment**
 - Commit e9d3791

**Additional context**
Note that I unfortunately have no STM32 MCU with a RNG peripheral having such configuration fields at my disposal, so this issue is only based on my understanding of the reference manual. In case the issue turns out to be real, I am willing to create a PR but I won't be able to test the fix on real hardware.

taltenbach, 2022-05-24:
Precisely! I expect to be able to lay hands next week on a B-U85I-IOT01 board that includes an RNG peripheral with RNG_HTCR register and CONFIGx fields. Therefore, if that's alright with you I can fix the issue next week. Otherwise, if you prefer, I should be able to do the PR earlier but in that case it would be preferable that someone ensures that the entropy driver is still working properly on real hardware

FRASTM, 2022-05-25:
In order to apply the RNG configuration for the NIST certification, the conditions are about the rng_clock source, the RNG_CR register to write and loop , then immediately writing the RNG_HTCR magic nb then immediately writing the RNG_HTCR config nb. 
Because the RNG NIST config can depend on the mcu, a suggestion would be to introduce a new property in the RNG DTS, named `nist_config`

closed 2022-08-22 by fabiobaltieri