zephyrproject-rtos/zephyr #45893: MCUboot authentication failure with RSA-3072 key on i.MX RT 1160 EVK

danie007, 2022-05-23:
**Cannot use RSA-3072 keys with Zephyr & MCUboot**
Whenever I try signing the zephyr application with `RSA-3072` key (_both example & genrated one_) the application loading fails at bootloader stage.

- I'm using NXP's i.MX RT 1160 EVK
- I've tried printing the keyfile path on both MCUBoot build and zephyr application build and it points correctly to the desired file.

**To Reproduce**
Steps to reproduce the behavior on Ubuntu 22.04 PC:

```bash
west build -b mimxrt1160_evk_cm7 -s bootloader/mcuboot/boot/zephyr -d build-mcuboot -- -DCONFIG_BOOT_SIGNATURE_KEY_FILE=\"root-rsa-3072.pem\"
west build -b mimxrt1160_evk_cm7 -s zephyr/samples/hello_world -d build-hello-signed -- -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-3072.pem\"
west flash -d build-mcuboot
west flash -d build-hello-signed
```

**Expected behavior**
Should print `Hello world` text.

**Impact**
`showstopper` This issue does not allow the applicaiton to be loaded and renders the board unusable.

**Logs and console output**

1. MCUBoot build log: **Note** that the key file is correctly set as `/home/ux/zephyrproject/bootloader/mcuboot/root-rsa-3072.pem`

   ```log
   $ west build -b mimxrt1160_evk_cm7 -s bootloader/mcuboot/boot/zephyr -d build-mcuboot -- -DCONFIG_BOOT_SIGNATURE_KEY_FILE=\"root-rsa-3072.pem\"
   -- west build: generating a build system
   Loading Zephyr default modules (Zephyr base (cached)).
   -- Application: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr
   -- Cache files will be written to: /home/ux/.cache/zephyr
   -- Zephyr version: 3.1.0-rc2 (/home/ux/zephyrproject/zephyr)
   -- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
   -- Board: mimxrt1160_evk_cm7
   -- Found host-tools: zephyr 0.14.1 (/home/ux/zephyr-sdk-0.14.1)
   -- Found dtc: /home/ux/zephyr-sdk-0.14.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
   -- Found toolchain: zephyr 0.14.1 (/home/ux/zephyr-sdk-0.14.1)
   -- Found BOARD.dts: /home/ux/zephyrproject/zephyr/boards/arm/mimxrt1160_evk/mimxrt1160_evk_cm7.dts
   -- Found devicetree overlay: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
   -- Found devicetree overlay: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
   -- Found devicetree overlay: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
   -- Found devicetree overlay: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
   -- Found devicetree overlay: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
   -- Found devicetree overlay: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
   -- Found devicetree overlay: /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
   unit address and first address in 'reg' (0x20000) don't match for /soc/spi@400cc000/is25wp128@0/partitions/partition@10000
   -- Generated zephyr.dts: /home/ux/zephyrproject/build-mcuboot/zephyr/zephyr.dts
   -- Generated devicetree_unfixed.h: /home/ux/zephyrproject/build-mcuboot/zephyr/include/generated/devicetree_unfixed.h
   -- Generated device_extern.h: /home/ux/zephyrproject/build-mcuboot/zephyr/include/generated/device_extern.h
   -- Including generated dts.cmake file: /home/ux/zephyrproject/build-mcuboot/zephyr/dts.cmake
   Parsing /home/ux/zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig
   Loaded configuration '/home/ux/zephyrproject/build-mcuboot/zephyr/.config'
   No change to configuration in '/home/ux/zephyrproject/build-mcuboot/zephyr/.config'
   No change to Kconfig header in '/home/ux/zephyrproject/build-mcuboot/zephyr/include/generated/autoconf.h'
   Load components for MIMXRT1166:
   driver_common component is included.
   driver_reset component is included.
   device_CMSIS component is included.
   CMSIS_Include_core_cm component is included.
   device_system component is included.
   driver_lpuart component is included.
   driver_gpt component is included.
   driver_igpio component is included.
   driver_flexspi component is included.
   driver_cache_armv7_m7 component is included.
   driver_xip_board_evkmimxrt1160 component is included.
   MCUBoot bootloader key file: /home/ux/zephyrproject/bootloader/mcuboot/root-rsa-3072.pem
   -- Configuring done
   -- Generating done
   -- Build files have been written to: /home/ux/zephyrproject/build-mcuboot
   -- west build: building application
   [54/62] Linking C executable zephyr/zephyr_pre0.elf

   [57/62] Linking C executable zephyr/zephyr_pre1.elf

   [62/62] Linking C executable zephyr/zephyr.elf
   Memory region         Used Size  Region Size  %age Used
           OCRAM:          0 GB       256 KB      0.00%
           FLASH:       65500 B       128 KB     49.97%
               SRAM:       37760 B        64 MB      0.06%
               ITCM:        4040 B       256 KB      1.54%
               DTCM:          0 GB       256 KB      0.00%
           IDT_LIST:          0 GB         2 KB      0.00%

   ```

2. Zephyr application build log: **Note** that the key file is correctly set as `bootloader/mcuboot/root-rsa-3072.pem`

   ```log
   $ west build -b mimxrt1160_evk_cm7 -s zephyr/samples/hello_world -d build-hello-signed -- -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-3072.pem\"
   -- west build: generating a build system
   Loading Zephyr default modules (Zephyr base (cached)).
   -- Application: /home/ux/zephyrproject/zephyr/samples/hello_world
   -- Cache files will be written to: /home/ux/.cache/zephyr
   -- Zephyr version: 3.1.0-rc2 (/home/ux/zephyrproject/zephyr)
   -- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
   -- Board: mimxrt1160_evk_cm7
   -- Found host-tools: zephyr 0.14.1 (/home/ux/zephyr-sdk-0.14.1)
   -- Found dtc: /home/ux/zephyr-sdk-0.14.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
   -- Found toolchain: zephyr 0.14.1 (/home/ux/zephyr-sdk-0.14.1)
   -- Found BOARD.dts: /home/ux/zephyrproject/zephyr/boards/arm/mimxrt1160_evk/mimxrt1160_evk_cm7.dts
   unit address and first address in 'reg' (0x20000) don't match for /soc/spi@400cc000/is25wp128@0/partitions/partition@10000
   -- Generated zephyr.dts: /home/ux/zephyrproject/build-hello-signed/zephyr/zephyr.dts
   -- Generated devicetree_unfixed.h: /home/ux/zephyrproject/build-hello-signed/zephyr/include/generated/devicetree_unfixed.h
   -- Generated device_extern.h: /home/ux/zephyrproject/build-hello-signed/zephyr/include/generated/device_extern.h
   -- Including generated dts.cmake file: /home/ux/zephyrproject/build-hello-signed/zephyr/dts.cmake
   Parsing /home/ux/zephyrproject/zephyr/Kconfig
   Loaded configuration '/home/ux/zephyrproject/build-hello-signed/zephyr/.config'
   No change to configuration in '/home/ux/zephyrproject/build-hello-signed/zephyr/.config'
   No change to Kconfig header in '/home/ux/zephyrproject/build-hello-signed/zephyr/include/generated/autoconf.h'
   Load components for MIMXRT1166:
   driver_common component is included.
   driver_reset component is included.
   device_CMSIS component is included.
   CMSIS_Include_core_cm component is included.
   device_system component is included.
   driver_lpuart component is included.
   driver_gpt component is included.
   driver_igpio component is included.
   driver_cache_armv7_m7 component is included.
   -- Signature key file: bootloader/mcuboot/root-rsa-3072.pem
   -- Configuring done
   -- Generating done
   -- Build files have been written to: /home/ux/zephyrproject/build-hello-signed
   -- west build: building application
   [39/47] Linking C executable zephyr/zephyr_pre0.elf

   [42/47] Linking C executable zephyr/zephyr_pre1.elf

   [47/47] Linking C executable zephyr/zephyr.elf
   Memory region         Used Size  Region Size  %age Used
           OCRAM:          0 GB       256 KB      0.00%
           FLASH:       31148 B     301000 B     10.35%
               SRAM:        4224 B        64 MB      0.01%
               ITCM:          0 GB       256 KB      0.00%
               DTCM:          0 GB       256 KB      0.00%
           IDT_LIST:          0 GB         2 KB      0.00%
   ```

3. Serial Console log:
   ```log
    *** Booting Zephyr OS build v3.1.0-rc2-4-gfa84da588dbf  ***
    I: Starting bootloader
    I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
    I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
    I: Boot source: none
    I: Swap type: none
    E: Image in the primary slot is not valid!
    E: Unable to find bootable image
   ```

**Environment:**

- OS: `Ubuntu 22.04`
- Toolchain: `zephyr-sdk-0.14.1`
- Commit `4cc5c56d8c4a4727bcfb6a49cdc2d6ffba39295f`

---

CC @tkmozhi @vishnusudhan

danieldegrasse, 2022-05-25:
@danie007 try building with `-DCONFIG_BOOT_SIGNATURE_TYPE_RSA_LEN=3072`. I reproduced this issue on an RT1060, and that Kconfig option allowed authentication to proceed for me

danie007, 2022-05-26:
Thanks @danieldegrasse we're able to successfully build and authenticate with the `CONFIG_BOOT_SIGNATURE_TYPE_RSA_LEN=3072` line added in `prj.conf` file.

closed 2022-05-26 by danie007